micro-ROS/micro_ros_setup #210: Build fails with missing ros-foxy-performance-test-fixture

AndyBlightLeeds, 2020-10-09:
I have been building the firmware for the ESP32 using instructions on this post: <https://discourse.ros.org/t/micro-ros-porting-to-esp32/16101>.  When I ran the command

```bash
ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32
```
the build failed with the error message:

```bash
...
Successfully installed catkin-pkg-0.4.23 docutils-0.16 empy-3.3.4 lark-parser-0.10.0 python-dateutil-2.8.1
executing command [sudo -H apt-get install -y ros-foxy-performance-test-fixture]
Reading package lists... Done
Building dependency tree  
Reading state information... Done
E: Unable to locate package ros-foxy-performance-test-fixture
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-foxy-performance-test-fixture] failed
```

Running the command `sudo -H apt-get install -y ros-foxy-performance-test-fixture` manually showed the same problem.  The zephyr host build also fails in the same way.  I guess that the package `ros-foxy-performance-test-fixture` has been removed from the ROS servers for some reason. 

As I just want to work on my ESP32 board, and I guess the package `ros-foxy-performance-test-fixture` is not a 'must have', is there someway to temporarily remove the dependency on the package `ros-foxy-performance-test-fixture`?

pablogs9, 2020-10-09:
Please check if you are updated with this PR: https://github.com/micro-ROS/micro_ros_setup/pull/208

In this PR you can see how to SKIP the installation of these packages because I guess that this can happen with some other.

An explanation for this situation can be found here: https://github.com/ament/ament_cmake/issues/285

If you don't mind I'm closing since this a ROS 2 dependency, please reopen if you can't solve the issue.

closed 2020-10-09 by pablogs9

AndyBlightLeeds, 2020-10-09:
Thanks for the super quick reply :-)

Pulling the micro_ros_setup repo to get the new change didn't fix the problem so I stopped and deleted the docker container I'm using.  When I restarted the container and built again, the problem was solved.  There is probably a neater way to do this using `ros deps` but I'm able to get on with testing code on my ESP32.  Thanks again.

pablogs9, 2020-10-09:
BTW, if you are targetting ESP32, I suggest you trying also the standalone IDF component for micro-ROS: https://github.com/micro-ROS/micro_ros_espidf_component