micro-ROS/micro_ros_setup #212: build problem

DorBenHarush, 2020-10-09:
Hi, A build error was rasied due to this function:  RCUTILS_CAN_RETURN_WITH_ERROR_OF(false);
inside `microros_ws/firmware/mcu_ws/ros2/rosidl/rosidl_runtime_c/src/string_functions.c`
It appears that the funcion is undefined, I think that this `#include "rcutils/macros.h"` file contained the definiton prior to the problem and it has been moved to another file without updating the include. For now I simplly put the function in a comment and it works.
 (esp32 but I dont know if relates) 

Best regards,
Dor.

arunlee77, 2020-10-12:
Building for olimex-stm32-e407

AR:   stm32_boot.o stm32_bringup.o stm32_spi.o libatomic.o stm32_autoleds.o stm32_buttons.o stm32_appinit.o stm32_timer.o stm32_usb.o stm32_usb.o stm32_ostest.o
make[2]: Leaving directory '/root/uros_ws/firmware/NuttX/configs/olimex-stm32-e407/src'
LD: nuttx
arm-none-eabi-ld --entry=__start -nostartfiles -nodefaultlibs -g -T/root/uros_ws/firmware/NuttX/configs/olimex-stm32-e407/scripts/ld.script -L"/root/uros_ws/firmware/NuttX/staging" -L"/root/uros_ws/firmware/NuttX/arch/arm/src/board" -L "/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard" \
        -o "/root/uros_ws/firmware/NuttX/nuttx"   \
        --start-group -lsched -ldrivers -lconfigs -lc -lmm -larch -lxx -lapps -lnet -lfs -lbinfmt -lxx -lboard -lsupc++ "/usr/lib/gcc/arm-none-eabi/9.2.1/thumb/v7e-m+fp/hard/libgcc.a" --end-group
arm-none-eabi-ld: /root/uros_ws/firmware/NuttX/staging/libapps.a(librosidl_runtime_c_string_functions.c.obj): in function `rosidl_runtime_c__String__init':
/root/uros_ws/firmware/mcu_ws/ros2/rosidl/rosidl_runtime_c/src/string_functions.c:27: undefined reference to `RCUTILS_CAN_RETURN_WITH_ERROR_OF'
arm-none-eabi-ld: /root/uros_ws/firmware/NuttX/staging/libapps.a(librosidl_runtime_c_string_functions.c.obj): in function `rosidl_runtime_c__String__Sequence__init':
/root/uros_ws/firmware/mcu_ws/ros2/rosidl/rosidl_runtime_c/src/string_functions.c:120: undefined reference to `RCUTILS_CAN_RETURN_WITH_ERROR_OF'
make[1]: *** [Makefile:185: nuttx] Error 1
make[1]: Leaving directory '/root/uros_ws/firmware/NuttX/arch/arm/src'
make: *** [tools/Makefile.unix:422: pass2] Error 2

pablogs9, 2020-10-13:
Related to Foxy sync and with https://github.com/micro-ROS/micro_ros_setup/pull/209

Will be solved during this morning.

pablogs9, 2020-10-13:
Upgraded to new version, PR closed w/ CI passed. 

I'm closing this issue, but please reopen if the problem is not solved for you with a clean workspace of micro-ROS.

closed 2020-10-13 by pablogs9