micro-ROS/micro_ros_setup #215: Custom message test

MARLARCHAN, 2020-10-13:
Dear @pablogs9,

I would like to know the subscriber part from ping_pong example, where can I find fake ping code? 
When I test my custom message with the command ros2 topic echo /microROS/publisher, Publisher start run.
However, when I test for subsciber, ros2 topic echo /microROS/subscriber, it stop and wait like fake ping start run in ping_pong example. I attach my test in the following:

And, I would like to ask
What's the meaning of command for fake ping?
ros2 topic pub --once /microROS/ping std_msgs/msg/Header '{frame_id: "fake_ping"}'

Which code is related with this part?

And how can I test for custom message for subscriber run?
Best Regards,
Marlar  

[image]

pablogs9, 2020-10-13:
Please read carefully the explanation of the ping pong demo. In this demo, every node sends pings and waits for pongs, and also when a node receive a ping, it sends a pong.

So, if you open a single node, you can listen to its pings with the ROS2 CLI tool:

```bash
ros2 topic echo /microROS/ping
```

And if you send a ping with the CLI tools, the node will answer it with a pong:

```bash
ros2 topic pub --once /microROS/ping std_msgs/msg/Header '{frame_id: "fake_ping"}'
```

You have the code here: https://github.com/micro-ROS/micro-ROS-demos/blob/foxy/rclc/ping_pong/main.c

pablogs9, 2020-11-16:
Closing since not answered. Reopen if the problem still exists.

closed 2020-11-16 by pablogs9