micro-ROS/micro_ros_setup #216: Configuring micro-ROS firmware

vibnwis, 2020-10-13:
Hi Pablo and team,
1. Would you sharing a bit what this, "Configuring micro-ROS firmware" is about, and how would it impact the execution when the CAN-bus transport is ready.?

2. If one did not run this, the configuring command, can he still run the corresponding application at later stage? and change one to the other?

3. When the CAN-bus transport is ready, How would the command be effected? I presume, it will not change anything for me, as I am  interpreting udp as CAN-bus. In the case I need udp, I can use ip and port to differentiate upd from CAN-bus.

4. The line "--dev or -d: agent string descriptor in a serial-like transport", is this only for serial-like transport?

5. When compiling the uros firmware for Olimex stm32-e407, I was assuming, the transport is udp, and the ips must of netmask 255.255.255.0, correct?

pablogs9, 2020-10-13:
Hi @vibnwis 

> Would you sharing a bit what this, "Configuring micro-ROS firmware" is about, and how would it impact the execution when the CAN-bus transport is ready.?

Configuring the micro-ROS firmware does basically 2 things: 
1. Selects the app that is going to be built. Remember that each app has a `colcon.meta` that allows a fine-grain configuration for each layer, we usually are specially interested in RMW and middleware where it is possible to tune the memory usage of the micro-ROS build. [Check this](https://micro-ros.github.io/docs/tutorials/core/microxrcedds_rmw_configuration/).
2. Selects the transport that is going to be use and the parameters: serial port and the serial device or UDP and agent IP/port.

This can be modified by changing [these scripts](https://github.com/micro-ROS/micro_ros_setup/blob/foxy/config/freertos/olimex-stm32-e407/configure.sh), that basically are modifying the global `colcon.meta`

> If one did not run this, the configuring command, can he still run the corresponding application at later stage? and change one to the other?

If you don't run the configuration command the system won't know which app it should build. Please note that in the case of [micro-ROS for Linux](https://micro-ros.github.io/docs/tutorials/core/first_application_linux/) you don't select any app because [all examples](https://github.com/micro-ROS/micro-ROS-demos) are being built and the transport will be always UDP.

> When the CAN-bus transport is ready, How would the command be effected? I presume, it will not change anything for me, as I am interpreting udp as CAN-bus. In the case I need udp, I can use ip and port to differentiate upd from CAN-bus.

IMO, the best solution is to add in your [configure script](https://github.com/micro-ROS/micro_ros_setup/blob/foxy/config/freertos/olimex-stm32-e407/configure.sh) an option for CAN and inside it, you will copy your transports, similar to what is done [here](https://github.com/micro-ROS/micro_ros_setup/blob/2f59d8704ff1ecec2f53d6e4aa514766d6b26773/config/freertos/olimex-stm32-e407/configure.sh#L35)

Some questions:
- Which is your hardware board?
- Which RTOS are you going to use? I guess that FreeRTOS right?
- Which parameters do you need in order to init the CAN bus? You will need an address? Ports? Maybe we need to update something in the RMW if you need multiple parameters, but don't mind, we can update it quickly.

> The line "--dev or -d: agent string descriptor in a serial-like transport", is this only for serial-like transport?

Yes, when using serial-like transports, the middleware will be inited with the framing module enabled and it will pass this `-d` parameter to your init function.

When using udp-like transports, the middleware will suppose that you are able to send whole messages and the parameters passed with `-i` and `-p` will be passed to your init function as strings. The inside your init function you can handle them as you want.

> When compiling the uros firmware for Olimex stm32-e407, I was assuming, the transport is udp, and the ips must of netmask 255.255.255.0, correct?

Olimex STM32-E407 can be build for Nuttx, FreeRTOS and Zephyr. Each one supports different transports, [check this table](https://github.com/micro-ROS/micro_ros_setup#configuring-micro-ros-firmware). In the case of FreeRTOS, you can use UART or UDP as transports.

In the case of UDP, if I'm not wrong it will try to retrieve an IP using DHCP, [check init task for this port](https://github.com/micro-ROS/freertos_apps/blob/34656bb8e1b5f80a2ef4f05e26da371f7538da55/microros_olimex_e407_extensions/Src/main.c#L417)

vibnwis, 2020-10-14:
The answers to the questions:
Which is your hardware board?
Olimex STM32-E407.

Which RTOS are you going to use? I guess that FreeRTOS right?
Yes, it is FreeRTOS.

Which parameters do you need in order to init the CAN bus? You will need an
address? Ports? Maybe we need to update something in the RMW if you need
multiple parameters, but don't mind, we can update it quickly.

CAN-bus frames are identified by its frame-ID which also acts the frame's
priority levels. At the moment, the sender sets the frame-ID as part of the
frame, and receiver sets the hardware filters for the frame-ID and receives
the intended frames. As I don't know how the future applications would end
up, I don't know which parameters I will use for now. You can interpret it
as, there aren't any parameters needed for the moment.

The template's IP and Port are not usable for the time being.

Appreciate your assistance in willing to update the RMW. It is my pleasure
to do that when the time comes.

Very grateful for all the assistance you and your team provided, i.e., your
answers and the info in web-pages.

pablogs9, 2020-10-14:
Nice @vibnwis you are really welcome.
I'm closing this. Don't hesitate to reopen if you have more questions!

Let me know when you start integration the CAN integration, so I can give you support.

closed 2020-10-14 by pablogs9