micro-ROS/micro_ros_setup #219: Build fails

AndyBlightLeeds, 2020-10-29:
Hi,

Up until now, I have been using the https://github.com/micro-ROS/micro_ros_espidf_component repo to program my ESP32 but decided it was time to use custom ROS messages instead of bit packing standard int32 messages. Unfortunately, the docker I was using, https://github.com/micro-ROS/micro_ros_espidf_component/tree/foxy/docker, does not have ROS Foxy so I can't use that to generate messages.

I started with a Foxy Focal docker and followed the instructions in the [first tutorial](https://micro-ros.github.io/docs/tutorials/core/first_application_linux/) to create the workspace.  I then deleted the firmware directory and setup  the ESP32 firmware using these commands:

```bash
sudo apt update
ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32
# Configure the test app.
ros2 run micro_ros_setup configure_firmware.sh int32_publisher -t udp -i 192.168.2.28 -p 8888
# Add Wi-Fi credentials using this command.
ros2 run micro_ros_setup build_firmware.sh menuconfig
# Build the code.
ros2 run micro_ros_setup build_firmware.sh
# Flash the code.
ros2 run micro_ros_setup flash_firmware.sh
```
I then tested the built application with the pre-built agent using this command:

```bash
docker run -it --rm --net=host microros/micro-ros-agent:foxy udp4 --port 8888
```
and it worked as expected.  I then added the custom message following [this tutorial](https://micro-ros.github.io/docs/tutorials/core/create_new_type/) and built using the command:

```bash
    ros2 run micro_ros_setup build_firmware.sh
```

When I checked the `build` directory for the new message package I could not find it. So I then tried using `colcon build` but there was still no sign of my new message package.  I then remembered that new package was in the `firmware/mcu_ws` so I cd'd into that directory and built using `colcon build --merge-install`.  This resulted in the error message:

```bash
    Finished <<< rosidl_adapter [0.67s]
    Starting >>> rosidl_parser                                                                                                                                                                                                             
    --- stderr: rosidl_runtime_cpp                                                                                                                                                                                                          
    CMake Error at CMakeLists.txt:19 (find_package):
      By not providing "Findperformance_test_fixture.cmake" in CMAKE_MODULE_PATH
      this project has asked CMake to find a package configuration file provided
      by "performance_test_fixture", but CMake did not find one.
    
      Could not find a package configuration file provided by
      "performance_test_fixture" with any of the following names:
    
        performance_test_fixtureConfig.cmake
        performance_test_fixture-config.cmake
    
      Add the installation prefix of "performance_test_fixture" to
      CMAKE_PREFIX_PATH or set "performance_test_fixture_DIR" to a directory
      containing one of the above files.  If "performance_test_fixture" provides
      a separate development package or SDK, be sure it has been installed.
    
    
    ---
    Failed   <<< rosidl_runtime_cpp [1.35s, exited with code 1]
    Finished <<< test_interface_files [1.32s]
    Finished <<< microcdr [1.39s]
    Finished <<< tracetools [1.35s]
    Aborted  <<< tracetools_read [0.85s]
    Aborted  <<< rosidl_parser [0.75s]
    Aborted  <<< tracetools_trace [0.90s]
    Aborted  <<< rcutils [0.81s]
    
    Summary: 7 packages finished [2.18s]
      1 package failed: rosidl_runtime_cpp
      4 packages aborted: rcutils rosidl_parser tracetools_read tracetools_trace
      1 package had stderr output: rosidl_runtime_cpp
      52 packages not processed
```

As the `rosidl_default_runtime` package is critical to using custom ROS messages, I'm stuck. This might be related to an issue I raised #210 that also depended on the `performance_test_fixture` package that is no longer available or I might just be doing the wrong thing!

pablogs9, 2020-10-29:
Hello @AndyBlightLeeds, so I understand that your micro-ROS app is publishing correctly the custom messages right?

Let me know what is your main problem:
- Integrating your own custom type in micro-ROS client side, or
- Building the custom message along with the agent so you can `ros2 topic echo` it?

AndyBlightLeeds, 2020-10-29:
The problem is integrating the custom type in the micro-ROS build system.  My problems are:
1. Am I building the my message package correctly?  Using the command `ros2 run micro_ros_setup build_firmware.sh` didn't pick up my new message package.  Can you please confirm that I have the right set up?
2. If I am building it correctly, then the missing `performance_test_fixture` package is causing the build to fail, so I need a work around.  

The message file I'm using is shown below:
```
# Motor speeds: 0 stop, -ve backwards, +ve forwards, 100 maximum speed.
int8 left
int8 right
# Motor operation duration in ms.
# uint16 used to mpose maximum duration of 65 seconds.
uint16 duration_ms
```

pablogs9, 2020-10-29:
Two points:

Regarding the external msg creation in ESP32 here you have a complete video of the procedure:

https://asciinema.org/a/ERulP9hCXpkzPgpS0eKXPcozz

The session is quite long but you can find:
- Message creation [5:40](https://asciinema.org/a/ERulP9hCXpkzPgpS0eKXPcozz?t=5:40)
- Header check [11:28](https://asciinema.org/a/ERulP9hCXpkzPgpS0eKXPcozz?t=11:28)
- Library .a check [12:08](https://asciinema.org/a/ERulP9hCXpkzPgpS0eKXPcozz?t=12:08)

Regarding the `performance_test_fixture` it should not being built because all the `*_cpp` are keep out of the micro-ROS build, precisely in the here you can see the exact moment in the `micro_ros_setup` when colcon tools is call:

https://github.com/micro-ROS/freertos_apps/blob/1c3b88be3405906847d25b4d33d02aad6ab48215/microros_esp32_extensions/libmicroros.mk#L36

AndyBlightLeeds, 2020-10-29:
Thanks for the video.  I'll follow it and let you know how I get on.

AndyBlightLeeds, 2020-10-30:
@pablogs9 Thanks for the excellent video.  I copied your commands on my PC and it worked.  I then added my own message package and that worked as well :-)

I guess my problem was that I tried to add the ESP32 to an existing build of the agent and something went wrong.  I'll keep the agent and ESP32 code in separate workspaces from now on.

Thanks again for your prompt replies.

closed 2020-10-30 by AndyBlightLeeds