micro-ROS/micro_ros_setup #226: esp32 installation error

RajasJ99, 2020-11-12:
[image]

when i run the command :ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32
it starts properly but stops at the exact same step as mentioned in image and gives error of permission denied
checked updates but everything is set for display only and no process is currently running as seen in photo still the error
ROS2 version - foxy

Thanks in advance

pablogs9, 2020-11-12:
In which step do you have this error? This is related to a missing `sudo`. Tell me details about your process.

pablogs9, 2020-11-12:
Anyway, meanwhile we debug and fix this permission issue, if you are targeting micro-ROS for ESP32 check this port: https://github.com/micro-ROS/micro_ros_espidf_component

RajasJ99, 2020-11-13:
Thanks for the instant reply, and will check the esp-idf component method,
And yeah it looks like an sudo error. My process is simple : 

1. git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
2.rosdep update && rosdep install --from-path src --ignore-src -y
3.colcon build
4.source install/local_setup.bash
5.ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32

I have tried this probably like 3 times now , but it stops at same step eveytime

pablogs9, 2020-11-13:
I see that the error is in the `apt` of [this line](https://github.com/micro-ROS/micro_ros_setup/blob/0cac790119a8aaec0bfe1dcbca43d9b70ea46fe5/config/freertos/esp32/create.sh#L17).

This was added https://github.com/micro-ROS/micro_ros_setup/pull/220.

Should we find another way of installing `pip3` and `virtualenv`? Maybe we can just check if the packages are installed and raise an error telling the user that they are needed, because of course we can't assume root permission in this creation script. 

Thoughts @jamoralp?

RajasJ99, 2020-11-13:
Well i just deleted the two lines: apt install -y python3-pip ,pip3 install virtualenv from create.sh
And then installed them manually. 
Later when i ran : ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32 , there was no error.

jamoralp, 2020-11-13:
The error is clearly coming from there. I will think about a better way to solve this. Probably warning the user about which packages are required to be installed in case they don't already exist in the system is the better solution.

flynneva, 2020-11-16:
> Well i just deleted the two lines: apt install -y python3-pip ,pip3 install virtualenv from create.sh
> And then installed them manually.
> Later when i ran : ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32 , there was no error.

for anyone else - dont forget to re-run `colcon build` after making these changes ^^^.
i forgot too :smile:

jamoralp, 2020-11-16:
Please see https://github.com/micro-ROS/micro_ros_setup/pull/231

pablogs9, 2020-11-16:
Closing since #231 merged. Please reopen if problems related to this issue.

closed 2020-11-16 by pablogs9