micro-ROS/micro_ros_setup #227: Build agent errors

RajasJ99, 2020-11-13:
when ran command in workspace: ros2 run micro_ros_setup create_agent_ws.sh , got the following error:
[image]

Also when tried colcon build found another error : 
[image]
 
The two errors are probably not related because the first one is due to :rosdep install and the later is due to micr_ros_agent directory

FilipStarkenberg, 2020-11-13:
I have the exact same error. Tested on two machines running Ubuntu 20.04.

pablogs9, 2020-11-13:
Did you install the prerequisites? 

`sudo apt install python3-rosdep`

[image]

FilipStarkenberg, 2020-11-13:
Yes I did and I have a fresh installation of foxy. I think there is some sort of problem with fastrtps.

RajasJ99, 2020-11-13:
The first error has gone now , forgot to install python3-rosdep ,but the second error remains, tried completely rebuilding my workspace but the error is still there

pablogs9, 2020-11-16:
Hello everyone. Sorry for this error.

I have reset the `foxy` branch to in Micro XRCE-Agent  to a34ad20a6dcc457cf34a3df97cacb87b00a2ff9d, so it should be working again.

I'm closing since solved, please reopen if any other issue.

CC: @jamoralp

closed 2020-11-16 by pablogs9