micro-ROS/micro_ros_setup #228: Rtt

Aerotic, 2020-11-14:


servetcoskun, 2020-11-15:
Can confirm, I have the same issue following "[First micro-ROS Application on Linux](https://micro-ros.github.io/docs/tutorials/core/first_application_linux/)"

peratle, 2020-11-15:
Hi,

I also experienced this, but found a way to solve it:

In file ~/microros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/CMakeLists.txt, add dependency to fastrtps three places (find_package, target_link_libraries, amend_export_dependencies)

Here you can see parts of the file after change:

```
...
find_package(ament_cmake REQUIRED)
find_package(microxrcedds_agent REQUIRED)
find_package(fastrtps REQUIRED)

add_executable(${PROJECT_NAME} src/main.cpp)

target_link_libraries(${PROJECT_NAME}
  PRIVATE
    microxrcedds_agent
    fastrtps
    $<$<BOOL:$<PLATFORM_ID:Linux>>:rt>
    $<$<BOOL:$<PLATFORM_ID:Linux>>:dl>
  )

set_target_properties(${PROJECT_NAME} PROPERTIES
  CXX_STANDARD
    11
  CXX_STANDARD_REQUIRED
    YES
  )

target_compile_options(${PROJECT_NAME}
  PRIVATE
    $<$<OR:$<CXX_COMPILER_ID:GNU>,$<CXX_COMPILER_ID:Clang>>:-Wall>
    $<$<OR:$<CXX_COMPILER_ID:GNU>,$<CXX_COMPILER_ID:Clang>>:-Wextra>
    $<$<OR:$<CXX_COMPILER_ID:GNU>,$<CXX_COMPILER_ID:Clang>>:-Wpedantic>
  )

ament_export_dependencies(microxrcedds_agent fastrtps)
...
```

FilipStarkenberg, 2020-11-15:
Thank you. It worked for me as well. Good work!

pablogs9, 2020-11-16:
Hello everyone. Sorry for this error.

I have reset the `foxy` branch to in Micro XRCE-Agent  to a34ad20a6dcc457cf34a3df97cacb87b00a2ff9d, so it should be working again.

@peratle thanks a lot for this fix, we are going to check internally why this error is introduced in the latest commit.

I'm closing since solved, please reopen if any other issue.

CC: @jamoralp

closed 2020-11-16 by pablogs9

Hubery-wl, 2023-05-06:
I found that all of my content loaded, but still report this error, I've read a lot of comments but still can't fix it.