micro-ROS/micro_ros_setup #229: pass-through build arguments don't work if there is no preceding argument

RobertWilbrandt, 2020-11-15:
I recently contributed a feature to pass through build arguments for zephyr builds using a syntax like ```micro_ros_setup build_firmware.sh -f -- -DCMAKE_EXPORT_COMPILE_COMMANDS=on```. As i wanted to implement a similar feature for the flash step (enabling e.g. ```--runner jlink``` for west, allowing the use of a jlink programmer) i realized this doesn't work correctly when there is no argument preceding the ```--```.

An example:
- Using ```ros2 run micro_ros_setup build_firmware.sh -v -- -h``` works fine (kind of). It starts with verbsoe build information and then starts west, which runs cmake. cmake then sees the ```-h``` and prints its help message.
- ```ros2 run micro_ros_setup build_firmware.sh -- -h``` does not work: It "skips" over the ```-h``` and just prints the help message for the ```build_firmware.sh``` script.

This seems to be caused by the actual ```ros2 run``` command, as far as i can tell the ```--``` just gets stripped away if it is the first argument. I am willing to fix this, but have troubles understanding if this is the correct behavior (meaning my change cannot work) or where exactly in ```ros2cli``` i should start to look. Does anybody have insights into this? As far as i can tell this is not mandated behavior from [the design document](https://design.ros2.org/articles/ros_command_line_arguments.html), but i could be wrong.

Any insight/help in this topic would be appreciated.

pablogs9, 2020-11-16:
Hello @RobertWilbrandt, I'm not really into `ros2cli` so I'm sorry I can't help. Maybe open an issue in any of their repos?

Let us know if you have news about this.

pablogs9, 2022-01-28:
Any update on this?

Acuadros95, 2022-07-26:
Closing as issue moved to other repo.

closed 2022-07-26 by Acuadros95