micro-ROS/micro_ros_setup #230: flash error with stlink v2

ncnynl, 2020-11-16:
~/microros_ws$ ros2 run micro_ros_setup flash_firmware.sh
Flashing firmware for freertos platform olimex-stm32-e407
Flashing firmware for freertos platform olimex-stm32-e407
Error. Unsuported OpenOCD USB programmer

ncnynl, 2020-11-16:
I have fixed

~/microros_ws/src/micro_ros_setup/config/freertos/olimex-stm32-e407/flash.sh

add as below： 
elif lsusb -d 0483:3748; then
   PROGRAMMER=interface/stlink-v2.cfg

cd ~/microros_ws/
colcon build 
ros2 run micro_ros_setup flash_firmware.sh

pablogs9, 2020-11-16:
Closing since solved. Please consider creating a Pull Request with your changes/improvements in order to contribute them to the community.

closed 2020-11-16 by pablogs9

rajesh3699, 2023-06-15:
I am using a docker running. I could create firmware using the repo. But while flashing it throws this error:

/microros_ws/install/micro_ros_setup/config/freertos/nucleo_f767zi/flash.sh: line 10: [false: command not found
Error. Unsuported OpenOCD USB programmer
[ros2run]: Process exited with failure 1

The solution above did not work.

This is flash.sh:

EXTENSIONS_DIR=$FW_TARGETDIR/freertos_apps/microros_nucleo_f767zi_extensions

USE_STFLASH=false

pushd $EXTENSIONS_DIR > /dev/null

  if [ -f build/micro-ROS.bin ]; then

    echo "Flashing firmware for $RTOS platform $PLATFORM"
    if [$USE_STFLASH = true]; then
      st-flash --reset write build/micro-ROS.bin 0x8000000
    else
      if lsusb -d 0483:374b; then
        ST_INTERFACE=interface/stlink-v2-1.cfg
      elif lsusb -d 0483:3748; then
        PROGRAMMER=interface/stlink-v2.cfg
        ST_INTERFACE=interface/stlink-v2.cfg
      else
      # TODO: add stlink v3, should it be stlink.cfg ?
        echo "Error. Unsuported OpenOCD USB programmer"
        exit 1
      fi
      openocd -f $ST_INTERFACE -f target/stm32f7x.cfg -c init -c "reset halt" -c "flash write_image erase build/micro-ROS.bin 0x08000000" -c "reset" -c "exit"
    fi
  else
    echo "build/micro-ROS.bin not found: please compile before flashing."
  fi

popd > /dev/null

Any solution for this?