micro-ROS/micro_ros_setup #234: not seeing topics over wifi after following esp32devkitc instructions

flynneva, 2020-11-18:
i think ive [followed all the instructions](https://micro-ros.github.io/blog/2020/08/27/esp32/) correctly. the one area where I wasnt sure if I messed up was inputting my specific IP address into the build command.

after running the micro-ros agent with the ESP32 device powered on I was expecting to be able to do a `ros2 topic list` and see some sort of topic being published from the device.

any tips on how to troubleshoot/fix m situation? I know i didnt provide a lot of info but im not sure what else to put down here.

pablogs9, 2020-11-18:
Hello @flynneva, let see what's happening here.

First, you have to create the firmware:

```bash
ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32
```

Then let's configure it with the Agent IP (your computer IP), mine is 192.168.1.100:

```bash
ros2 run micro_ros_setup configure_firmware.sh int32_publisher -t udp -i 192.168.1.100 -p 8888
```

In a separate terminal, let open the agent:

```bash
docker run -it --rm --net=host microros/micro-ros-agent:foxy udp4 --port 8888 -v6
```

Now, we are going to use a special micro-ROS for ESP32 function to configure the wifi, this should open a menu and there, under the micro-ROS setting you can set your wifi SSID and password:

```bash
ros2 run micro_ros_setup build_firmware.sh menuconfig
```

Configuration done, now connect the divide using USB and build and flash:

```bash
ros2 run micro_ros_setup build_firmware.sh
ros2 run micro_ros_setup flash_firmware.sh
```

We can inspect the ESP32 output using this command, here you will see if the device is connecting to wifi, to agent, etc...

```bash
ros2 run micro_ros_setup build_firmware.sh monitor
```

---

If this does not work please copy here the micro-ROS Agent output and the `build_firmware.sh monitor` output.

flynneva, 2020-11-18:
> ros2 run micro_ros_setup build_firmware.sh monitor

this monitor step wasnt in the docs but it looks like is the missing step to be able to "see" the topics coming from the esp32 device. might be good to add that to the instructions so that others dont run into the same issue.

thanks for the tips!

closed 2020-11-18 by flynneva

pablogs9, 2020-11-18:
The monitor step just opens a serial port to see the output of the ESP32 firmware, not related to the working of the embedded code. You should be able to run the same sample without opening the monitor.

Anyhow, nice that you can use micro-ROS!

flynneva, 2020-11-18:
hmm interesting. ok ill keep debugging, maybe i missed something else

flynneva, 2020-11-18:
weird. i mustve messed up the ip address for my computer somehow. just reran the steps you outlined and it works great!

thanks again for your help!

flynneva, 2020-11-18:
kjhAS

simonjcarr, 2021-01-08:
I am having a similar problem. After writing the firmware to the microcontroller, the Docker agent does not show any messages.
I am also confused / missing something. IF we are using ROS2 why do we need to provide an Agent IP Address? Is Micro-ROS using ROS rather than ROS2?
Does anyone know of a step by step tutorial for ESP32 and Ubuntu 20 or even better, Youtube video?

jamoralp, 2021-01-11:
> I am having a similar problem. After writing the firmware to the microcontroller, the Docker agent does not show any messages.
> I am also confused / missing something. IF we are using ROS2 why do we [...]

Good morning, answering to your question: the IP address must be provided on the client side, so the socket opened by the client knows how to call `getaddrinfo` in order to be able to connect to the socket opened by the agent.

Regarding the problems with the Docker, I think you might have to be more specific. It could be that you are mismatching the IP configuration, also maybe you have not properly set up the ESP32 WiFi credentials. Have you tried looking into ESP32 log while running the application? Did you launch the Agent's docker with the `--net=host` option?

Also, it might be better if you just open a new issue, since this one has been closed for two months already. Thanks!

carloscastmar, 2021-10-02:
I had the same problem with the wifi connection and I solved it thanks to these comments. However, once I connect the board to the wifi and the agent, no topic appears after using the ros2 topic list command. Does anyone know what may be the cause of this problem?

RioKJ21, 2022-12-12:
> Hello @flynneva, let see what's happening here.
> 
> First, you have to create the firmware:
> 
> ```shell
> ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32
> ```
> 
> Then let's configure it with the Agent IP (your computer IP), mine is 192.168.1.100:
> 
> ```shell
> ros2 run micro_ros_setup configure_firmware.sh int32_publisher -t udp -i 192.168.1.100 -p 8888
> ```
> 
> In a separate terminal, let open the agent:
> 
> ```shell
> docker run -it --rm --net=host microros/micro-ros-agent:foxy udp4 --port 8888 -v6
> ```
> 
> Now, we are going to use a special micro-ROS for ESP32 function to configure the wifi, this should open a menu and there, under the micro-ROS setting you can set your wifi SSID and password:
> 
> ```shell
> ros2 run micro_ros_setup build_firmware.sh menuconfig
> ```
> 
> Configuration done, now connect the divide using USB and build and flash:
> 
> ```shell
> ros2 run micro_ros_setup build_firmware.sh
> ros2 run micro_ros_setup flash_firmware.sh
> ```
> 
> We can inspect the ESP32 output using this command, here you will see if the device is connecting to wifi, to agent, etc...
> 
> ```shell
> ros2 run micro_ros_setup build_firmware.sh monitor
> ```
> 
> At this point, if everything has gone ok you should be able to see your topic using:
> 
> ```
> ros2 topic list
> ```
> 
> _Be careful about the Domain ID, some users change it and micro-ROS by defaults uses Domain ID 0_
> 
> If this does not work please copy here the micro-ROS Agent output and the `build_firmware.sh monitor` output.

Hey, I already followed your steps and it does not work. At first it worked, i can display the topic. However, now I want to try it again, it does not show that int32_publisher topic. 

Here is the output
 
[image]