micro-ROS/micro_ros_setup #236: cannot subscribe to best effort publisher

cocodmdr, 2020-11-23:
Hello,

I am using microROS since two months now and I noticed some strange behaviour.

I am running the ping_pong example with freeRTOS on olimex board.
Everything runs fines when publishing fake ping I can see a pong coming.

However I cannot subscribe to /microROS/pong messages in ROS.

- I can see the pong with the built-in ROS2 tool : **ros2 topic echo /microROS/pong**
- But I cannot subscribe to this topic with a simple subscriber like this one : https://index.ros.org/doc/ros2/Tutorials/Writing-A-Simple-Cpp-Publisher-And-Subscriber/

I noticed that it is related to this function  **rclc_publisher_init_best_effort()**

Changing this function with **rclc_publisher_init_default()** everything works well and I can subscribe to the topic.

cocodmdr, 2020-11-23:
student@student-W65-W67RZ1:~/microros_test_ws$ ros2 topic info /microROS/pong -v
Type: std_msgs/msg/Header

Publisher count: 1

Node name: _CREATED_BY_BARE_DDS_APP_
Node namespace: _CREATED_BY_BARE_DDS_APP_
Topic type: std_msgs/msg/Header
Endpoint type: PUBLISHER
GID: 01.0f.1f.19.30.83.06.00.01.00.00.00.00.00.02.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_BEST_EFFORT
  Durability: RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL
  Lifespan: 2147483651294967295 nanoseconds
  Deadline: 2147483651294967295 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 2147483651294967295 nanoseconds

Subscription count: 3

Node name: _CREATED_BY_BARE_DDS_APP_
Node namespace: _CREATED_BY_BARE_DDS_APP_
Topic type: std_msgs/msg/Header
Endpoint type: SUBSCRIPTION
GID: 01.0f.1f.19.30.83.06.00.01.00.00.00.00.00.04.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_BEST_EFFORT
  Durability: RMW_QOS_POLICY_DURABILITY_VOLATILE
  Lifespan: 2147483651294967295 nanoseconds
  Deadline: 2147483651294967295 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 2147483651294967295 nanoseconds

Node name: _ros2cli_378472
Node namespace: /
Topic type: std_msgs/msg/Header
Endpoint type: SUBSCRIPTION
GID: 01.0f.1f.19.68.c6.05.00.01.00.00.00.00.00.05.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_BEST_EFFORT
  Durability: RMW_QOS_POLICY_DURABILITY_VOLATILE
  Lifespan: 2147483651294967295 nanoseconds
  Deadline: 2147483651294967295 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 2147483651294967295 nanoseconds

Node name: minimal_subscriber
Node namespace: /
Topic type: std_msgs/msg/Header
Endpoint type: SUBSCRIPTION
GID: 01.0f.1f.19.cd.c5.05.00.01.00.00.00.00.00.12.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_RELIABLE
  Durability: RMW_QOS_POLICY_DURABILITY_VOLATILE
  Lifespan: 2147483651294967295 nanoseconds
  Deadline: 2147483651294967295 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 2147483651294967295 nanoseconds

pablogs9, 2020-11-23:
Could you share the C++ code in order to test it?

cocodmdr, 2020-11-23:
`// Copyright 2016 Open Source Robotics Foundation, Inc.
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

#include <functional>
#include <memory>

#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/header.hpp"

using std::placeholders::_1;

class MinimalSubscriber : public rclcpp::Node
{
public:
  MinimalSubscriber()
  : Node("minimal_subscriber")
  {
    subscription_ = this->create_subscription<std_msgs::msg::Header>(
      "/microROS/pong", 10, std::bind(&MinimalSubscriber::topic_callback, this, _1));
  }

private:
  void topic_callback(const std_msgs::msg::Header::SharedPtr msg) const
  {
    RCLCPP_INFO(this->get_logger(), "I heard: ");
  }
  rclcpp::Subscription<std_msgs::msg::Header>::SharedPtr subscription_;
};

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<MinimalSubscriber>());
  rclcpp::shutdown();
  return 0;
}
`

pablogs9, 2020-11-23:
But see, the C++ subscriber is being created in reliable mode `RMW_QOS_POLICY_RELIABILITY_RELIABLE` chech the last subscriber.

You have to use the RCLCPP `create_subscription` to create a best effort subscriber.

pablogs9, 2020-11-23:
Here you have an example: https://github.com/ros2/demos/blob/master/demo_nodes_cpp/src/topics/listener_best_effort.cpp

cocodmdr, 2020-11-23:
Than you very much !

pablogs9, 2020-11-23:
Please close the issue if you achieve subscribing to the topic.

cocodmdr, 2020-11-23:
ok I am trying to adapt the code and I will close when I succeed.

cocodmdr, 2020-11-23:
I changed QoS of the subscriber witch was 10 to rclcpp::SensorDataQoS() and it works !
Thank you for your help ! Now I understand Qos a bit better !

`#include <functional>
#include <memory>

#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/header.hpp"

using std::placeholders::_1;

class MinimalSubscriber : public rclcpp::Node
{
public:
  MinimalSubscriber()
  : Node("minimal_subscriber")
  {
    subscription_ = this->create_subscription<std_msgs::msg::Header>("/microROS/pong", rclcpp::SensorDataQoS(), std::bind(&MinimalSubscriber::topic_callback, this, _1));
  }

private:
  void topic_callback(const std_msgs::msg::Header::SharedPtr msg) const
  {
    RCLCPP_INFO(this->get_logger(), "I heard: ");
  }
  rclcpp::Subscription<std_msgs::msg::Header>::SharedPtr subscription_;
};

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<MinimalSubscriber>());
  rclcpp::shutdown();
  return 0;
}`

closed 2020-11-23 by cocodmdr

21coolkarni, 2024-01-30:
how can we visualize best effort publisher data on rqt it shows unknown when I subscribe