micro-ROS/micro_ros_setup #237: Exception in buiild_agent.sh

NielsKorschinsky, 2020-11-25:
Hello,

I'm currently trying to setup micro ros with zephyr. I've already installed zephyr, foxy, colcon & ros2.
I'm working down the (First Steps)[https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/zephyr/] for both Linux & With Zephyr. Everything works, but building the agent fails both times.  

I've already flashed my board. It's the `disco_l475_iot1`. Building the agent fails for both this & the generic board from the Linux setup.

If requested, i can provide more information. I'm on Ubunut LTS 20.04.1

```
ros2 run micro_ros_setup create_agent_ws.sh
Repo-file ros2.repos already present, overwriting!
...
=== ./drive_base (git) ===

Bereits auf 'master'
Ihr Branch ist auf demselben Stand wie 'origin/master'.
=== ./eProsima/Micro-XRCE-DDS-Agent (git) ===

Bereits auf 'foxy'
Ihr Branch ist auf demselben Stand wie 'origin/foxy'.
=== ./uros/micro-ROS-Agent (git) ===

Bereits auf 'foxy'
Ihr Branch ist auf demselben Stand wie 'origin/foxy'.
#All required rosdeps installed successfully
niels@niels-Lenovo:~/microros_ws$ ros2 run micro_ros_setup build_agent.sh
Building micro-ROS Agent
Starting >>> microxrcedds_agent
--- stderr: microxrcedds_agent                               
/home/niels/microros_ws/src/eProsima/Micro-XRCE-DDS-Agent/src/cpp/utils/ArgumentParser.cpp:24:21: error: ‘bool eprosima::uxr::agent::parser::utils::usage(const char*, bool)’ should have been declared inside ‘eprosima::uxr::agent::parser::utils’
   24 |         bool no_help)
      |                     ^
make[2]: *** [CMakeFiles/microxrcedds_agent.dir/build.make:310: CMakeFiles/microxrcedds_agent.dir/src/cpp/utils/ArgumentParser.cpp.o] Fehler 1
make[2]: *** Auf noch nicht beendete Prozesse wird gewartet …
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/microxrcedds_agent.dir/all] Fehler 2
make: *** [Makefile:130: all] Fehler 2
---
Failed   <<< microxrcedds_agent [10.7s, exited with code 2]

Summary: 0 packages finished [10.9s]
  1 package failed: microxrcedds_agent
  1 package had stderr output: microxrcedds_agent
  1 package not processed
```

jamoralp, 2020-11-25:
Good morning, yes, I noticed this just earlier this morning. This pull request https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/200 fixes it :slightly_smiling_face:

pablogs9, 2020-11-25:
Closing since https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/200 merged. 

Please retry to build the micro-ROS Agent. Sorry for the inconvenience.

closed 2020-11-25 by pablogs9

NielsKorschinsky, 2020-11-25:
Hello,
thanks for this very quick response!
I'm fairly new, could you please tell me where I can find the folder to pull this change?
I'm building it in `~/microros_ws$`
```
 ros2 run micro_ros_setup create_agent_ws.sh
Repo-file ros2.repos already present, overwriting!
...
=== ./drive_base (git) ===

Bereits auf 'master'
Ihr Branch ist auf demselben Stand wie 'origin/master'.
=== ./eProsima/Micro-XRCE-DDS-Agent (git) ===

Bereits auf 'foxy'
Ihr Branch ist 1 Commit hinter 'origin/foxy', und kann vorgespult werden.
  (benutzen Sie "git pull", um Ihren lokalen Branch zu aktualisieren)
=== ./uros/micro-ROS-Agent (git) ===

Bereits auf 'foxy'
Ihr Branch ist auf demselben Stand wie 'origin/foxy'.
#All required rosdeps installed successfully
```

pablogs9, 2020-11-25:
You have a couple of options. One is creating a new micro-ROS workspace and running the create_agent.sh again.

Also, you can pull the repo in `src/eProsima/Micro-XRCE-DDS-Agent` to the last commit.