micro-ROS/micro_ros_setup #245: Create Firmware error FreeRTOS Nucleo_F446ZE

RTrioux, 2020-12-09:
Hi, I know I'm not really on the most canonic setup for ROS/Micro-Ros but ... 
ArchLinux / zsh / ROS2 Foxy

I have followed instruction from the [micro-ros tutorial](https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/freertos/)

Despite my rosdep is normally correctly configured on Arch, I don't get why I get these errors during the first step
The create firmware command:
`ros2 run micro_ros_setup create_firmware_ws.sh freertos nucleo_f446ze`

Give me the following error:
`
...
Downloading ARM compiler, this may take a while
=== ./freertos_apps (git) ===
Clonage dans '.'...
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:

rosidl_default_runtime: No definition of [rosidl_typesupport_introspection_cpp] for OS [arch]
rosidl_typesupport_c: No definition of [mimick_vendor] for OS [arch]
libyaml_vendor: No definition of [rcpputils] for OS [arch]
rclc: No definition of [osrf_testing_tools_cpp] for OS [arch]
rcl_lifecycle: No definition of [osrf_testing_tools_cpp] for OS [arch]
rosidl_typesupport_cpp: No definition of [rcpputils] for OS [arch]
rosidl_default_generators: No definition of [rosidl_typesupport_introspection_cpp] for OS [arch]
rcl_action: No definition of [osrf_testing_tools_cpp] for OS [arch]
tracetools_test: No definition of [launch_ros] for OS [arch]
rclc_lifecycle: No definition of [osrf_testing_tools_cpp] for OS [arch]
tracetools_launch: No definition of [launch_ros] for OS [arch]
test_rmw_implementation: No definition of [rmw_dds_common] for OS [arch]
ros2trace: No definition of [ros2cli] for OS [arch]
rmw_implementation: No definition of [rcpputils] for OS [arch]
rcutils: No definition of [osrf_testing_tools_cpp] for OS [arch]
rcl: No definition of [rcpputils] for OS [arch]
rcl_logging_noop: No definition of [launch_testing] for OS [arch]
rmw: No definition of [osrf_testing_tools_cpp] for OS [arch]
`
How can I fix this ?
Thanks for considering this issue.

pablogs9, 2020-12-10:
Hello, maybe you can find a solution in a ros infrastructure repo like https://github.com/ros/rosdistro. I guess that this is not really related to micro-ROS build system.

RTrioux, 2020-12-11:
Hi,
Thanks a lot for your quick answer.
It seems you're right, I'll continue to investigate on rosdep side, I'm not used to ROS environment yet.
I'm closing this.
Thanks.

closed 2020-12-11 by RTrioux

pablogs9, 2020-12-11:
Cool, thanks!