micro-ROS/micro_ros_setup #246: Using generate_lib problems

mrbhjv, 2020-12-12:
Hi

I'm trying to use the generate_lib configuration to later use it to link to my project. I'm using FreeRTOS on an STM32 F7 chip, so I have my custom pipe based on a CubeMx generated code. I'll be also using FreeRTOS+TCP and FreeRTOS+POSIX (Which I added to use micro-ros).

My intention is to build micro-ros for my target, and then have it as an artifact to link to my project during the build process. Thus, only needing to build micro-ros when adding new messages. We'll be using Docker to have a stable image to build micro-ros.

However, I haven't found much documentation on how to use the `generate_lib` configuration. So, I based myself on how the setup works. *While doing this I also noticed that micro-ros examples use CMakeForceCompiler, which is deprecated.
I have my CMakeLists.txt that has all the flags and definitions needed and sets the compiler to my arm gcc/g++. It also adds the include directories for the FreeRTOS POSIX libraries, as it's needed for all the time functions.

I also set `#define configUSE_POSIX_ERRNO 1` on freertos.h

However, when I'm trying to generate the static library I run into

`firmware/mcu_ws/uros/rcutils/src/filesystem.c:248:20: error: 'errno' undeclared (first use in this function)`

If I'm not wrong, that means `libc` is not linking correctly.

Please notice that I don't want to be forced to build my own app at this stage, I'd like to be able to have a static lib to later use. Most of the setups sort of force the user to build their app at the same time. Maybe more documentation on how it's actually trying to link would be nice, as I'm guessing I'm forgetting something.

This is my toolchain. I've tried with and without the lines that have a comment that I don't normally use

```
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

# I normally don't use this. Added to test with micro-ros examples
set(CMAKE_CROSSCOMPILING 1)
set(CMAKE_SYSTEM_PROCESSOR arm)

set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_SYSTEM_VERSION 1)
set(CMAKE_C_COMPILER_WORKS 1)
set(CMAKE_CXX_COMPILER_WORKS 1)
set(CMAKE_AR arm-none-eabi-ar)
message("-- TOOLCHAIN AR ${CMAKE_AR}")
set(CMAKE_C_COMPILER arm-none-eabi-gcc)
message("-- TOOLCHAIN C COMPILER ${CMAKE_C_COMPILER}")
set(CMAKE_CXX_COMPILER arm-none-eabi-g++)
message("-- TOOLCHAIN CXX COMPILER ${CMAKE_CXX_COMPILER}")
set(CMAKE_ASM_COMPILER arm-none-eabi-gcc)
message("-- TOOLCHAIN ASM COMPILER ${CMAKE_ASM_COMPILER}")
set(CMAKE_OBJCOPY arm-none-eabi-objcopy)
message("-- TOOLCHAIN OBJCOPY ${CMAKE_OBJCOPY}")
set(CMAKE_OBJDUMP arm-none-eabi-objdump)
message("-- TOOLCHAIN OBJDUMP ${CMAKE_OBJDUMP}")
set(CMAKE_SIZE arm-none-eabi-size)
message("-- TOOLCHAIN SIZE ${CMAKE_SIZE}")

# I normally don't use this. Added to test with micro-ros examples
set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)

# Flags
set(C_COMMON_FLAGS "${C_COMMON_FLAGS} -mcpu=cortex-m7 -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb")

# generates a separate ELF section for each function in the source file
set(C_COMMON_FLAGS "${C_COMMON_FLAGS} -ffunction-sections")

# Enables or disables the generation of one ELF section for each variable in the source file
set(C_COMMON_FLAGS "${C_COMMON_FLAGS} -fdata-sections")

set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} ${C_COMMON_FLAGS}")

# Enables or disables the generation of code needed to support C++ exceptions.
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fno-exceptions")

# Disable generation of information about every class with virtual functions for use by the C++ runtime
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fno-rtti")

# Do not emit the extra code to use the routines specified in the C++ ABI for thread-safe initialization of local statics.
# Reduce bin size
# TODO Do we have code that could have thread safety problem?
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fno-threadsafe-statics")

# Register destructors for objects with static storage duration with the __cxa_ atexit function rather than the atexit function
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fno-use-cxa-atexit")

# Add common C language flags
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${C_COMMON_FLAGS}")

add_definitions("-DUSE_HAL_DRIVER -DSTM32F746xx")

include_directories(SYSTEM
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/FreeRTOS-Plus-POSIX/include
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/FreeRTOS-Plus-POSIX/include/portable/st
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/include
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/include/private
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/include/FreeRTOS_POSIX
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/include/FreeRTOS_POSIX/sys
${CMAKE_CURRENT_LIST_DIR}/f7-drivers/STM32F7xx_HAL_Driver/Inc
${CMAKE_CURRENT_LIST_DIR}/f7-drivers/CMSIS/Device/ST/STM32F7xx/Include
${CMAKE_CURRENT_LIST_DIR}/f7-drivers/CMSIS/Include
${CMAKE_CURRENT_LIST_DIR}/f7-middlewares/Third_Party/FreeRTOS/Source/include
${CMAKE_CURRENT_LIST_DIR}/f7-middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2
${CMAKE_CURRENT_LIST_DIR}/f7-middlewares/FreeRTOS/Source/portable/GCC/ARM_CM7/r0p1
)
# I normally don't use this. Added to test with micro-ros examples
set(__BIG_ENDIAN__ 0)
```

pablogs9, 2020-12-14:
Hello, I think that micro-ROS for Arduino is a great example of how to use the `generate lib` functionality: https://github.com/micro-ROS/micro_ros_arduino. In this case, we provide a standalone precompiled micro-ROS library for a bunch of platforms and I guess that this is what you need.

Please check [this script](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/extras/library_generation/library_generation.sh) to see how to use the generate lib.

Please let me know if you have any other questions.

mrbhjv, 2020-12-14:
Hi
Thanks for the reference, it did help. Seems what I was missing was to add `-DCLOCK_MONOTONIC=0 -D'__attribute__(x)='` to my toolchain, at least now rcutils passes, and I removed all the includes from include_directories. I'm not sure what `__attribute_(x)=` does and if I actually need it.

I've also modified the colcon.meta to be
```
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        },
        "rcl": {
            "cmake-args": [
                "-DBUILD_TESTING=OFF",
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
            ]
        }, 
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON"
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=ON",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=10",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=4",
                "-DRMW_UXRCE_TRANSPORT=udp",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=10.0.0.3",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=8888",
            ]
        }
    }
}
```
However, I now get two errors which I guess are related

The first one is about the package not being found, which is weird
```
CMake Warning at CMakeLists.txt:183 (find_package):
  By not providing "Findrosidl_typesupport_microxrcedds_cpp.cmake" in
  CMAKE_MODULE_PATH this project has asked CMake to find a package
  configuration file provided by "rosidl_typesupport_microxrcedds_cpp", but
  CMake did not find one.
```

The second one is a bunch `uxr` types not found
```
In file included from /microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/utils.h:20,
                 from /microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_client.c:15:
/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./types.h:83:3: error: unknown type name 'uxrUDPPlatform'
   83 |   uxrUDPPlatform udp_platform;
```

My wild guess is that I'm missing some flag or like.

pablogs9, 2020-12-14:
Regarding the cpp: [this line](https://github.com/micro-ROS/micro_ros_setup/blob/0f514e77af8ad8e4258d672b7d65674d09665a8f/config/generate_lib/generic/build.sh#L28) should be skipping all the cpp components of the ROS2 package set. In which packet are you getting this error?

Regarding your colcon.meta, you have a couple of commas malforming the JSON try removing last elements commas:

```json
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=ON",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF" <--- here
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=10",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=4",
                "-DRMW_UXRCE_TRANSPORT=udp",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=10.0.0.3",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=8888"   <--- here
            ]
```

Which transport will you use? I guess that UDP

mrbhjv, 2020-12-14:
I'll use UDP, could also try TCP, but definitely not serial

A little verbose, but here's the trace.
```
--- stderr: rmw_microxrcedds
-- TOOLCHAIN AR arm-none-eabi-ar
-- TOOLCHAIN C COMPILER arm-none-eabi-gcc
-- TOOLCHAIN CXX COMPILER arm-none-eabi-g++
-- TOOLCHAIN ASM COMPILER arm-none-eabi-gcc
-- TOOLCHAIN OBJCOPY arm-none-eabi-objcopy
-- TOOLCHAIN OBJDUMP arm-none-eabi-objdump
-- TOOLCHAIN SIZE arm-none-eabi-size
-- TOOLCHAIN AR arm-none-eabi-ar
-- TOOLCHAIN C COMPILER arm-none-eabi-gcc
-- TOOLCHAIN CXX COMPILER arm-none-eabi-g++
-- TOOLCHAIN ASM COMPILER arm-none-eabi-gcc
-- TOOLCHAIN OBJCOPY arm-none-eabi-objcopy
-- TOOLCHAIN OBJDUMP arm-none-eabi-objdump
-- TOOLCHAIN SIZE arm-none-eabi-size
-- TOOLCHAIN AR arm-none-eabi-ar
-- TOOLCHAIN C COMPILER arm-none-eabi-gcc
-- TOOLCHAIN CXX COMPILER arm-none-eabi-g++
-- TOOLCHAIN ASM COMPILER arm-none-eabi-gcc
-- TOOLCHAIN OBJCOPY arm-none-eabi-objcopy
-- TOOLCHAIN OBJDUMP arm-none-eabi-objdump
-- TOOLCHAIN SIZE arm-none-eabi-size
CMake Warning at /microros_ws/firmware/mcu_ws/install/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rcutils' exports library 'dl' which couldn't be found
Call Stack (most recent call first):
  /microros_ws/firmware/mcu_ws/install/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  CMakeLists.txt:27 (find_package)

CMake Warning at CMakeLists.txt:183 (find_package):
  By not providing "Findrosidl_typesupport_microxrcedds_cpp.cmake" in
  CMAKE_MODULE_PATH this project has asked CMake to find a package
  configuration file provided by "rosidl_typesupport_microxrcedds_cpp", but
  CMake did not find one.

  Could not find a package configuration file provided by
  "rosidl_typesupport_microxrcedds_cpp" with any of the following names:

    rosidl_typesupport_microxrcedds_cppConfig.cmake
    rosidl_typesupport_microxrcedds_cpp-config.cmake

  Add the installation prefix of "rosidl_typesupport_microxrcedds_cpp" to
  CMAKE_PREFIX_PATH or set "rosidl_typesupport_microxrcedds_cpp_DIR" to a
  directory containing one of the above files.  If
  "rosidl_typesupport_microxrcedds_cpp" provides a separate development
  package or SDK, be sure it has been installed.

In file included from /microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_count.c:22:
/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./types.h:83:3: error: unknown type name 'uxrUDPPlatform'
   83 |   uxrUDPPlatform udp_platform;
      |   ^~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/rmw_microxrcedds.dir/build.make:118: CMakeFiles/rmw_microxrcedds.dir/src/rmw_count.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/utils.h:20,
                 from /microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_client.c:15:
/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./types.h:83:3: error: unknown type name 'uxrUDPPlatform'
   83 |   uxrUDPPlatform udp_platform;
      |   ^~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/rmw_microxrcedds.dir/build.make:92: CMakeFiles/rmw_microxrcedds.dir/src/rmw_client.c.obj] Error 1
In file included from /microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_get_gid_for_publisher.c:15:
/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/types.h:83:3: error: unknown type name 'uxrUDPPlatform'
   83 |   uxrUDPPlatform udp_platform;
      |   ^~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/rmw_microxrcedds.dir/build.make:144: CMakeFiles/rmw_microxrcedds.dir/src/rmw_get_gid_for_publisher.c.obj] Error 1
In file included from /microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/utils.h:20,
                 from /microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_get_implementation_identifier.c:16:
/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./types.h:83:3: error: unknown type name 'uxrUDPPlatform'
   83 |   uxrUDPPlatform udp_platform;
      |   ^~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/rmw_microxrcedds.dir/build.make:157: CMakeFiles/rmw_microxrcedds.dir/src/rmw_get_implementation_identifier.c.obj] Error 1
In file included from /microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/utils.h:20,
                 from /microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_get_serialization_format.c:16:
/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./types.h:83:3: error: unknown type name 'uxrUDPPlatform'
   83 |   uxrUDPPlatform udp_platform;
      |   ^~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/rmw_microxrcedds.dir/build.make:170: CMakeFiles/rmw_microxrcedds.dir/src/rmw_get_serialization_format.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:81: CMakeFiles/rmw_microxrcedds.dir/all] Error 2
make: *** [Makefile:144: all] Error 2
```

pablogs9, 2020-12-14:
Could you share your `firmware/include/uxr/client/config.h` ? I guess that CMake is not configuring correctly the XRCE-DDS middleware

mrbhjv, 2020-12-14:
I don't have anything on the firmware folder, except `COLCON_IGNORE  PLATFORM  dev_ws  mcu_ws`

pablogs9, 2020-12-14:
sorry, inside `firmware/mcu_ws/install/microxrcedds_client/include/uxr/client/config.h`

mrbhjv, 2020-12-14:
I don't have exactly the path you said, but I have `firmware/mcu_ws/install/include/uxr/client/config.h` which I guess is the one you meant.
```
//    Copyright 2017 Proyectos y Sistemas de Mantenimiento SL (eProsima).
//
//    Licensed under the Apache License, Version 2.0 (the "License");
//    you may not use this file except in compliance with the License.
//    You may obtain a copy of the License at
//
//        http://www.apache.org/licenses/LICENSE-2.0
//
//    Unless required by applicable law or agreed to in writing, software
//    distributed under the License is distributed on an "AS IS" BASIS,
//    WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
//    See the License for the specific language governing permissions and
//    limitations under the License.

#ifndef _UXR_CLIENT_CONFIG_H_
#define _UXR_CLIENT_CONFIG_H_

#define UXR_CLIENT_VERSION_MAJOR 1
#define UXR_CLIENT_VERSION_MINOR 2
#define UXR_CLIENT_VERSION_MICRO 5
#define UXR_CLIENT_VERSION_STR "1.2.5"

/* #undef UCLIENT_PROFILE_DISCOVERY */

#define UCLIENT_PROFILE_UDP
/* #undef UCLIENT_PROFILE_TCP */
/* #undef UCLIENT_PROFILE_SERIAL */

/* #undef UCLIENT_PLATFORM_POSIX */
/* #undef UCLIENT_PLATFORM_POSIX_NOPOLL */
/* #undef UCLIENT_PLATFORM_WINDOWS */
/* #undef UCLIENT_PLATFORM_FREERTOS_PLUS_TCP */

/* #undef UCLIENT_EXTERNAL_TCP */
/* #undef UCLIENT_EXTERNAL_UDP */
/* #undef UCLIENT_EXTERNAL_SERIAL */

#define UXR_CONFIG_MAX_OUTPUT_BEST_EFFORT_STREAMS     1
#define UXR_CONFIG_MAX_OUTPUT_RELIABLE_STREAMS        1
#define UXR_CONFIG_MAX_INPUT_BEST_EFFORT_STREAMS      1
#define UXR_CONFIG_MAX_INPUT_RELIABLE_STREAMS         1

#define UXR_CONFIG_MAX_SESSION_CONNECTION_ATTEMPTS    10
#define UXR_CONFIG_MIN_SESSION_CONNECTION_INTERVAL    25
#define UXR_CONFIG_MIN_HEARTBEAT_TIME_INTERVAL        1

#ifdef UCLIENT_PROFILE_UDP
#define UXR_CONFIG_UDP_TRANSPORT_MTU                  512
#endif
#ifdef UCLIENT_PROFILE_TCP
#define UXR_CONFIG_TCP_TRANSPORT_MTU                  512
#endif
#ifdef UCLIENT_PROFILE_SERIAL
#define UXR_CONFIG_SERIAL_TRANSPORT_MTU               512
#endif

#endif // _UXR_CLIENT_CONFIG_H_
```

pablogs9, 2020-12-14:
We forgot to add the cmake platform name that allows [UXCE-DDS library to add the transports](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/ecf03ef0579387e6df8f53cdfe494e7b849fc1f4/CMakeLists.txt#L124). So, if I use this [meta](https://pastebin.ubuntu.com/p/tBvzMPzBYg/) and this [toolchain](https://pastebin.ubuntu.com/p/CbR7TQNGt7/) I fix this error. Anyways you will need to add the FreeRTOS includes to the toolchain.

The point here is that the middleware XRCE has some included transports (POSIX Serial,  POSIX UDP, FreeRTOS+TCP...) and it allows to use external transports, in the case of Arduino we are using a uart external transport check [here](https://github.com/micro-ROS/micro_ros_arduino/blob/afed440f4fc00bc8d3aada468471cd16c4ebcff7/extras/library_generation/library_generation.sh#L35) and [here](https://github.com/micro-ROS/micro_ros_arduino/blob/afed440f4fc00bc8d3aada468471cd16c4ebcff7/extras/library_generation/colcon.meta#L36). 

So you can do a couple of things:
1. Include FreeRTOS+TCP headers in this micro-ROS library generation and build the library linked against the XRCE included FreeRTOS+TCP transport. **Be aware that micro-ROS only has support for UDP in its rwm, anyways it should be easy to add TCP support**
2. Create a "dummy network" transport that uses functions that are defined in your application level in order to have the library independent from the transport. This is the approach used in Arduino, [here](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/extras/library_generation/arduino_xrce_transports/serial_transport_external.c) you have the dummy functions calls that are included in the library and [here](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/src/default_transport.cpp) that actual functions that are built at application build time and linked against the library.

Hope that this helps.

mrbhjv, 2020-12-14:
AFAIK, FreeRTOS+TCP has UDP support. As I'm creating UDP sockets with it.

I'll give it another spin and comment back.

mrbhjv, 2020-12-14:
So, I'm going with 1 for now. If needed, I'll go with 2.

I'm using the meta you suggested and the toolchain with the platform. After adding the platform, it started to ask for all the FreeRTOS headers.

Anyways, I'm now back to getting the following error
```
/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c:41:1: note: 'errno' is defined in header '<errno.h>'; did you forget to '#include <errno.h>'?
   40 | #include "rcutils/error_handling.h"
  +++ |+#include <errno.h>
   41 |
```

It does not make much sense, as the meta says it should skip the filesystem on rcutils.

```
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF",
            ]
        },
        "rcl": {
            "cmake-args": [
                "-DBUILD_TESTING=OFF",
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
            ]
        }, 
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON",
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=ON",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=10",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=4",
                "-DRMW_UXRCE_TRANSPORT=udp",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=10.0.0.3",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=8888",
            ]
        }
    }
}
```

pablogs9, 2020-12-14:
Since FreeRTOS does not have a complete errno POSIX support we tweak it like this in our ports: https://github.com/micro-ROS/freertos_apps/blob/34656bb8e1b5f80a2ef4f05e26da371f7538da55/microros_crazyflie21_extensions/include/FreeRTOS_POSIX/errno.h#L215

mrbhjv, 2020-12-14:
Well, in my FreeRTOS posix errno.h I have
```
/**
 * @name System Variable
 *
 * @brief Define FreeRTOS+POSIX errno, if enabled.
 * Set configUSE_POSIX_ERRNO to enable, and clear to disable. See FreeRTOS.h.
 *
 * @{
 */
#if ( configUSE_POSIX_ERRNO == 1 )
    extern int FreeRTOS_errno;
    #define errno    FreeRTOS_errno
#endif
```

And I have defined configUSE_POSIX_ERRNO to 1 in my freertos.h
```
#ifndef configUSE_POSIX_ERRNO
	#define configUSE_POSIX_ERRNO 1
#endif
```
But for some reason, it's not taking it...which makes me think

Well, I spent some time debugging it, it does take the posix errno when building `microxrcedds_client`

The problem is on rcutils, as it takes stricly the posix errno, but it does not go through the freertos.h before that.

I've tried adding `"-DconfigUSE_POSIX_ERRNO=1"` to the meta, but was unsuccessful, I'd like to enable the errno without having to modify the code.

I've done some digging, to analyze the verbose output of rcutils, and I've found some things that are worrysom, first, I haven't found any of the cmake definitions declared on `colcon.meta`. Is this correct?

Second, I've found that the C flags are added three times...Which is oddly weird
```
/gcc-arm-none-eabi/bin/arm-none-eabi-gcc -DRCUTILS_BUILDING_DLL -DROS_PACKAGE_NAME=\"rcutils\" -I/microros_ws/firmware/mcu_ws/build/rcutils/include -I/microros_ws/firmware/mcu_ws/uros/rcutils/include -I/microros_ws/firmware/mcu_ws/install/share/tinydir_vendor/cmake/../../../include -isystem /microros_ws/local/code -isystem /microros_ws/local/f7-drivers/STM32F7xx_HAL_Driver/Inc -isystem /microros_ws/local/f7-drivers/CMSIS/Device/ST/STM32F7xx/Include -isystem /microros_ws/local/f7-drivers/CMSIS/Include -isystem /microros_ws/local/f7-middlewares/Third_Party/FreeRTOS/Source/include -isystem /microros_ws/local/f7-middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 -isystem /microros_ws/local/f7-middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM7/r0p1 -isystem /microros_ws/local/freertos-plus-posix/FreeRTOS-Plus-POSIX/include -isystem /microros_ws/local/freertos-plus-posix/FreeRTOS-Plus-POSIX/include/portable -isystem /microros_ws/local/freertos-plus-posix/FreeRTOS-Plus-POSIX/include/portable/st -isystem /microros_ws/local/freertos-plus-posix/include -isystem /microros_ws/local/freertos-plus-posix/include/private -isystem /microros_ws/local/freertos-plus-posix/include/FreeRTOS_POSIX -isystem /microros_ws/local/freertos-plus-posix/include/FreeRTOS_POSIX/sys -isystem /microros_ws/local/freertos-plus-tcp/include -isystem /microros_ws/local/freertos-plus-tcp/portable/Compiler/GCC -isystem /microros_ws/local/freertos-plus-tcp/portable/NetworkInterface/include -isystem /microros_ws/local/freertos-plus-tcp/portable/NetworkInterface/STM32F7xx  -mcpu=cortex-m7 -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -DCLOCK_MONOTONIC=0 -D'__attribute__(x)=' -ffunction-sections -fdata-sections  -mcpu=cortex-m7 -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -DCLOCK_MONOTONIC=0 -D'__attribute__(x)=' -ffunction-sections -fdata-sections -mcpu=cortex-m7 -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -DCLOCK_MONOTONIC=0 -D'__attribute__(x)=' -ffunction-sections -fdata-sections -O3 -DNDEBUG   -DUSE_HAL_DRIVER -DSTM32F746xx -DUSE_HAL_DRIVER -DSTM32F746xx -Wall -Wextra -Wconversion -Wno-sign-conversion -Wpedantic -std=gnu11 -o CMakeFiles/rcutils.dir/src/cmdline_parser.c.obj   -c /microros_ws/firmware/mcu_ws/uros/rcutils/src/cmdline_parser.c
```

pablogs9, 2020-12-14:
The RCutils used in micro-ROS is this fork: https://github.com/micro-ROS/rcutils
A slightly modified version of the ROS 2 rcutils that included options such as no filesystem or 64-bit atomic library.

Regarding C flags are added three times I'm not sure if this is related to the micro-ROS build system or to `colcon`.

In any case, I would use the `-DconfigUSE_POSIX_ERRNO=1` not in the colcon.meta but in the toolchain in the same way we fake the `-DCLOCK_MONOTONIC=0 -D'__attribute__(x)='`

mrbhjv, 2020-12-14:
Okay!! that has seemed to build. I guess I would only need to statically link that to my project and it should be fine. Quite amazing.

However, I'd like to know one more thing, which version of the heap is being used for FreeRTOS? On our projects, we use `heap_useNewLib.c` from http://www.nadler.com/embedded/newlibAndFreeRTOS.html as the STM version is not good.

Is it used at all on micro-ROS or I shouldn't worry about it?

pablogs9, 2020-12-14:
In ROS you have configurable allocators so I guess that there is no problem in setting your own allocator in your micro-ROS projects. Let me show you how we configure them:

- [Here](https://github.com/micro-ROS/freertos_apps/blob/foxy/microros_olimex_e407_extensions/Src/allocators.c) we define a bunch of custom functions that calls the system ones.
- [Here](https://github.com/micro-ROS/freertos_apps/blob/a5b51d8fd5c4a58f02be17cf651a4cf92483b6fd/microros_olimex_e407_extensions/Src/main.c#L451) we set them as the interl micro-ROS memory allocation functions
- And finally, in the application level [this call](https://github.com/micro-ROS/freertos_apps/blob/a5b51d8fd5c4a58f02be17cf651a4cf92483b6fd/apps/int32_publisher/app.c#L33) will return the previously defined functions.

Of course, points 2 and 3 can be in the same functions. This will mock all the memory management functions along the micro-ROS stack, however I think that is important to notice that micro-ROS tries to minimize the dynamic memory allocations by using static memory pools (configured with the `"-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",` and so on). It just uses dynamic memory in the creation of the entities and no during the node operation.

You can find more information about memory management here: https://eprosima.com/index.php/resources-all/performance/micro-ros-client-memory-profiling.

Finally, could you tell us more about your use case? We are collecting users use cases and opinions, so if you want to share we will be happy.

mrbhjv, 2020-12-14:
Thanks, that makes sense. We're currently working with no dynamic allocations purely static allocations. I'll have to take a look at how it's being done, as we wouldn't want any dynamic memory if possible.

I was asking specifically about the heap as it's something that's usually preconfigured with freertos.h and in our case, don't build the typical `heap_4.c` but the other mentioned one instead.

Regarding our use case,
We're updating our site, https://capra.ooo/ but you can look for Capra Robotics on LinkedIn to see what we're currently working on :)

We have been working on all embedded systems until now and we are now moving upwards on the stack. And we want to use micro-ROS to link our embedded system with the high-level computer where the agent will be.

closed 2020-12-14 by mrbhjv

FranFin, 2020-12-14:
Hi @mrbhjv! Thanks for sharing info about your current projects. Here at micro-ROS, we'd like to know more about your use-case. If interested to have a chat with us someday soon, don't hesitate to contact me at francescafinocchiaro@eprosima.com!