micro-ROS/micro_ros_setup #250: Not able to Use esp32s2

FrGe2016, 2020-12-18:
Hi 

I followed the steps all the way up to Flashing micro-ROS firmware for the following situation
      i used the Foxy docker proposed in the First micro-ROS Application on FreeRTOS tutorial

       ros2 run micro_ros_setup create_firmware_ws.sh freertos  esp32

       ros2 run micro_ros_setup configure_firmware.sh ping_pong --transport udp

        ros2 run micro_ros_setup configure_firmware.sh menuconfig
             then updated my SSID and passworld
 
       ros2 run micro_ros_setup build_firmware.sh
            had 100% good compilation
      
       ros2 run micro_ros_setup flash_firmware.sh
            **At that point i had a lots of errors related to the fact that built and compilation have been done for the esp32 and not the    esp32s2**

       I tried two thing that i normaly do in a stand alone ESP-IDF
              1) idf.py set-target esp32s2
                        i did it fromthe directory  /microros_ws/firmware/freertos_apps/apps/ping_pong
                        **Then i had bash: idf.py: command not found**

              2) I Have tried to set the target to  esp32s2
                          export IDF_TARGET=esp32s2
 
                          then ros2 run micro_ros_setup build_firmware.sh and ros2 run micro_ros_setup flash_firmware.sh
                          **It did gave me the same errors  invalids packets...**

                          I have tried to redo the work from the step before
                                    ros2 run micro_ros_setup configure_firmware.sh ping_pong --transport udp

                                    the a new set of error
                                     .... -- Found Git: /usr/bin/git (found version "2.25.1") 
                                          CMake Error at /microros_ws/firmware/toolchain/esp-idf/tools/cmake/targets.cmake:57 (message):
                                         Unable to resolve 'esp32s2' for setting toolchain file.
                                         Call Stack (most recent call first):
                                           /microros_ws/firmware/toolchain/esp-idf/tools/cmake/project.cmake:295 (__target_set_toolchain)
                                           CMakeLists.txt:12 (project)

                                          -- Configuring incomplete, errors occurred!

         The board is ok i am able to flash with the esp2s2 target in a stand alone IDF installation in my home directory but not with       
         this setup

FrGe2016, 2020-12-18:
I have also tried via https://github.com/micro-ROS/micro_ros_espidf_component 2 times

1) Using the proposed Docker
       It did not work ( but i do not remembre the problem )
 2) With a fresh install of ESP-IDF in a clean in a new Foxy docker
       Here i have a long log of  and a fatal error at the end of the build operation ( I have saved the log if required )

pablogs9, 2021-01-15:
Hello @FrGe2016 here you have a screencast with instructions: https://asciinema.org/a/1sF0xeWvqd9bxIAVXfp696fKW

As you can see I'm not able to flash it because I don't have an ESP32S2, but it should work for you.

Let me know if you have any problem.

shic15, 2021-04-11:
Hey, recently I had also tried to run micro ros example on ESP32, after some figuring out I was able to make it work.
I see there are two methods of using micro ros with esp32 first is [installing and creating a esp env inside microros_ws](https://discourse.ros.org/t/micro-ros-porting-to-esp32/16101), second to use [micro ros as a component in espidf](https://github.com/micro-ROS/micro_ros_espidf_component). Both of the methods work for me, but I like the second one better as I'm already familiar with espidf commands also I haven't explored micro ros much, yet.

For your issue, I would suggest going with the second method (micro ros as a espidf component) cause I had some errors when I sourced both the ros2 and espidf environment variable path in the same terminal.

> I have also tried via https://github.com/micro-ROS/micro_ros_espidf_component 2 times
> 
> 1. Using the proposed Docker
>    It did not work ( but i do not remembre the problem )

After setting up the espidf environment variables and the present directory being the project folder, execute
```
idf.py set-target esp32s2
idf.py build 
idf.py -p (port) flash
```
It should compile and flash.

I would like to mention one more thing when I used the dockerized version of micro ros agent 
`docker run -it --rm --net=host microros/micro-ros-agent:foxy udp4 --port 8888 -v6
`
ROS2 could not detect the topic.
But the topic is available when I used the [snap package of micro ros agent](https://snapcraft.io/micro-ros-agent).

I believe I need to add something in the docker command for it to insinuate the ros2 on the local machine about the topic.

pablogs9, 2021-04-12:
If you are using the dockerized version of the docker using `-v /dev/shm:/dev/shm` should resolve the missing topic issue.

pablogs9, 2021-12-17:
Closing due to inactivity. Reopen if needed.

closed 2021-12-17 by pablogs9