micro-ROS/micro_ros_setup #252: Issue connecting with Agent

mrbhjv, 2021-01-11:
Hello,

I've been working with the self-contained library, and I've been running into different issues. I'm working with FreeRTOS+Posix  and FreeRTOS+TCP (it has UDP). The first issue came when having two tasks that required the IP stack, and I didn't know that the stack should be initiated before the FreeRTOS scheduler. The second one I'm currently having is that I see my program fails on https://github.com/micro-ROS/rmw-microxrcedds/blob/7f6ca2995e4e6eea5b6398b3cf152daefbbc00ea/rmw_microxrcedds_c/src/rmw_init.c#L312 when trying to connect. With some Wireshark inspection, I see some traffic coming out of my board, with the origin address being correct, but the destination is 0.0.0.0. AFAIK, 0.0.0.0 is an IP that indicates bad behavior.

My colcon.meta looks like
```
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF",
            ]
        },
        "rcl": {
            "cmake-args": [
                "-DBUILD_TESTING=OFF",
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
            ]
        }, 
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON",
                "-DconfigUSE_POSIX_ERRNO=1",
                "-DCMAKE_VERBOSE_MAKEFILE:BOOL=ON",
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=ON",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=10",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=4",
                "-DRMW_UXRCE_TRANSPORT=udp",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=10.46.28.3",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=8888",
            ]
        }
    }
}
```

I have another task that's simply creating a socket and sending a hello world to 10.0.0.3 at port 4444 just to verify it works, and while that one works fine and I can create listeners to that traffic, nothing seems to happen with the other traffic.
I'm assuming there's something that's not configured correctly.

Here's a snapshot from wireshark, I can see that in part of the body of that frame, there's an XRCE text.
[image]

pablogs9, 2021-01-12:
Can you provide steps to replicate your procedure? Are you using the static library generation?

Can you share you CMake toolchain? We need to ensure that the middleware is using this flag: https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/ecf03ef0579387e6df8f53cdfe494e7b849fc1f4/CMakeLists.txt#L124

mrbhjv, 2021-01-12:
Yes, I'm using the static library generation route, and I do have that flag added.
Let me also state that the library builds correctly and I have no issues compiling.
Here's my toolchain discussed on https://github.com/micro-ROS/micro_ros_setup/issues/252

```
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

set(PLATFORM_NAME "FreeRTOS_Plus_TCP")
set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)
set(CMAKE_CROSSCOMPILING 1)
set(CMAKE_SYSTEM_NAME Generic)
#set(CMAKE_SYSTEM_VERSION 1)
set(CMAKE_C_COMPILER_WORKS 1)
set(CMAKE_CXX_COMPILER_WORKS 1)
#set(CMAKE_SYSTEM_PROCESSOR arm)
set(CMAKE_AR arm-none-eabi-ar)
message("-- TOOLCHAIN AR ${CMAKE_AR}")
set(CMAKE_C_COMPILER arm-none-eabi-gcc)
message("-- TOOLCHAIN C COMPILER ${CMAKE_C_COMPILER}")
set(CMAKE_CXX_COMPILER arm-none-eabi-g++)
message("-- TOOLCHAIN CXX COMPILER ${CMAKE_CXX_COMPILER}")
set(CMAKE_ASM_COMPILER arm-none-eabi-gcc)
message("-- TOOLCHAIN ASM COMPILER ${CMAKE_ASM_COMPILER}")
set(CMAKE_OBJCOPY arm-none-eabi-objcopy)
message("-- TOOLCHAIN OBJCOPY ${CMAKE_OBJCOPY}")
set(CMAKE_OBJDUMP arm-none-eabi-objdump)
message("-- TOOLCHAIN OBJDUMP ${CMAKE_OBJDUMP}")
set(CMAKE_SIZE arm-none-eabi-size)
message("-- TOOLCHAIN SIZE ${CMAKE_SIZE}")

#set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)

set(C_COMMON_FLAGS "${C_COMMON_FLAGS} -fdebug-prefix-map=/microros_ws/firmware/mcu_ws=/workspaces/hircus-control-board/ThirdParty/micro-ros/sources -mcpu=cortex-m7 -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -DCLOCK_MONOTONIC=0 -D'__attribute__(x)=' -DconfigUSE_POSIX_ERRNO=1 -DipconfigZERO_COPY_TX_DRIVER=1 -DUSE_HAL_DRIVER -DSTM32F746xx")

# generates a separate ELF section for each function in the source file
set(C_COMMON_FLAGS "${C_COMMON_FLAGS} -ffunction-sections")

# Enables or disables the generation of one ELF section for each variable in the source file
set(C_COMMON_FLAGS "${C_COMMON_FLAGS} -fdata-sections")

set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} ${C_COMMON_FLAGS}")

# Enables or disables the generation of code needed to support C++ exceptions.
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fno-exceptions")

# Disable generation of information about every class with virtual functions for use by the C++ runtime
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fno-rtti")

# Do not emit the extra code to use the routines specified in the C++ ABI for thread-safe initialization of local statics.
# Reduce bin size
# TODO Do we have code that could have thread safety problem?
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fno-threadsafe-statics")

# Register destructors for objects with static storage duration with the __cxa_ atexit function rather than the atexit function
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fno-use-cxa-atexit")

# Add common C language flags
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${C_COMMON_FLAGS}")

#add_definitions("-DUSE_HAL_DRIVER -DSTM32F746xx")

set(CMAKE_BUILD_TYPE DEBUG)

include_directories(SYSTEM
${CMAKE_CURRENT_LIST_DIR}/code
${CMAKE_CURRENT_LIST_DIR}/f7-drivers/STM32F7xx_HAL_Driver/Inc
${CMAKE_CURRENT_LIST_DIR}/f7-drivers/CMSIS/Device/ST/STM32F7xx/Include
${CMAKE_CURRENT_LIST_DIR}/f7-drivers/CMSIS/Include
${CMAKE_CURRENT_LIST_DIR}/f7-middlewares/Third_Party/FreeRTOS/Source/include
${CMAKE_CURRENT_LIST_DIR}/f7-middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2
${CMAKE_CURRENT_LIST_DIR}/f7-middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM7/r0p1
# ==================
# = FreeRTOS+POSIX =
# ==================
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/FreeRTOS-Plus-POSIX/include
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/FreeRTOS-Plus-POSIX/include/portable
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/FreeRTOS-Plus-POSIX/include/portable/st
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/include
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/include/private
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/include/FreeRTOS_POSIX
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/include/FreeRTOS_POSIX/sys
# ================
# = FreeRTOS+TCP =
# ================
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-tcp/include/
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-tcp/portable/Compiler/GCC/
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-tcp/portable/NetworkInterface/include/
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-tcp/portable/NetworkInterface/STM32F7xx/
)
set(__BIG_ENDIAN__ 0)
```

I can't share the code I'm working with, however, I can briefly explain.

The steps are
* call FreeRTOS_IPInit
* start scheduler with two tasks that use the ethernet stack.
* One task simply opens a socket and sends a package to 10.46.28.3:4444 while the other task is the uros one.

pablogs9, 2021-01-12:
Is the other task sending correctly the data?

Regarding the micro-ROS task, FreeRTOS+TCP was a community-contributed transport for XRCE (micro-ROS middleware), you can take a look here: https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/161

Maybe it is interesting to analyze if the function to open/close/send/recv the XRCE stream data implemented [here](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/foxy/src/c/profile/transport/ip/udp/udp_transport_freertos_plus_tcp.c) are correctly behaving.

Do you have any way to debug these functions?

mrbhjv, 2021-01-12:
The other task is sending data correctly (a simple "Hello world"), and I can see that in the body of the outgoing message to port 8.8.8.8, there are the letters XRCE. The only big difference between the two tasks is that one will send a message to the destination IP, while the other one will do to 0.0.0.0.

I'll take a look at the code and analyze it, to be honest, it's the first time we work with FreeRTOS+TCP, so not much more idea than going through the code right now and debug it.

pablogs9, 2021-01-12:
What happens if micro-ROS is the only task using networking? Maybe the network stack has a limit for simultaneous connections.

mrbhjv, 2021-01-12:
The strange thing is that the socket is correctly created on the board (at least don't get any error) and I see a message incoming through the Ethernet directly connected to the board.

I'll try it out and let you know.

mrbhjv, 2021-01-13:
Running just the micro-ros task didn't change the behavior,
[image]

I still see the same type of messages, where this is the body of the first packages, with Len 24
[image]

And this is from the latter with Len 12
[image]

I do have a micro-ros agent running, and I'm sure I can create a basic script to listen to ports, but I simply see nothing when it comes to this one, it's like the destination address messes it up, that's why I created the issue here, as it might be some miss configuration.

pablogs9, 2021-01-13:
Can you open the micro-ROS Agent with the `-v6` flag and paste here the output?

I would like to see if the agent is receiving these messages. BTW the 24B message seems to be a correct XRCE Create Session message.

mrbhjv, 2021-01-13:
There's nothing, which is the weird part, and I can't even create a simple python script that'll listen to port 8888. The socket is created but nothing comes in, and the only difference when I run the one to 4444 is the destination IP.

Running it with `ros2 run micro_ros_agent micro_ros_agent udp4 -p 8888 -v6`
[image]

pablogs9, 2021-01-13:
Ok, so the packets are not arriving at your computer... Could you share how are you creating this other socket that does work?

I guess that there is something happening in the Micro XRCE-DDS Transport layer for FreeRTOS+TCP

mrbhjv, 2021-01-13:
Here's the code, I use python3

```
import socket

UDP_IP = ""
UDP_PORT = 4444

sock = socket.socket(socket.AF_INET, # Internet
                     socket.SOCK_DGRAM) # UDP
sock.bind((UDP_IP, UDP_PORT))

while True:
    data, addr = sock.recvfrom(1024) # buffer size is 1024 bytes
    print("received message: %s" % data)
```

Here's the output listening to port 4444
[image]

Heres' the same trace of wireshark
[image]

And here's the body
[image]

pablogs9, 2021-01-13:
I meant the FreeRTOS code

mrbhjv, 2021-01-13:
```
void board::NetworkTask::tick()
{
  auto socket = FreeRTOS_socket(FREERTOS_AF_INET, FREERTOS_SOCK_DGRAM, FREERTOS_IPPROTO_UDP);

  struct freertos_sockaddr dest;
  dest.sin_family = FREERTOS_AF_INET;
  dest.sin_addr = FreeRTOS_inet_addr_quick(10, 46, 28, 3);
  dest.sin_port = FreeRTOS_htons(4444);
  dest.sin_len = sizeof(freertos_sockaddr);
  if (FREERTOS_INVALID_SOCKET != socket)
  {
    FreeRTOS_sendto(socket, "Hello, world!", 13, 0, &dest, 0);
  }

  HAL_GPIO_TogglePin(LD1_GPIO_Port, LD1_Pin);
  osDelay(500);
}
```

pablogs9, 2021-01-13:
Ok, I guess that I know what is happening here.

We are providing the IP as a `const char *` type, so we are using `FreeRTOS_inet_addr` to convert it to a octet representation. Check [here](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/ecf03ef0579387e6df8f53cdfe494e7b849fc1f4/src/c/profile/transport/ip/udp/udp_transport_freertos_plus_tcp.c#L39).

In the [FreeRTOS+TCP documentation](https://www.freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_TCP/API/inet_addr.html) it says that this function has to be enabled using the flag `ipconfigINCLUDE_FULL_INET_ADDR`. Please check the documentation of this flag: http://www.openrtos.net/FreeRTOS-Plus/FreeRTOS_Plus_TCP/TCP_IP_Configuration.html#ipconfigINCLUDE_FULL_INET_ADDR.

Let me know if this is a solution for this weird "0.0.0.0" IPs...

mrbhjv, 2021-01-13:
So, I quickly debugged a bit what it does, and it seems to be calling the "send to" with sin_addr "8888" and port 8888 which is weird.

And I do have that flag enabled. It seems to be an error on what Ip address is being given to the sendto function.

mrbhjv, 2021-01-13:
This is what's being set on rmw_init
[image]

And this is the value that `FreeRTOS_inet_addr` generates
[image]

pablogs9, 2021-01-13:
Ok... sorry, check your configuration:

```
                "-DRMW_UXRCE_DEFAULT_UDP_IP=10.46.28.3",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=8888",
```

mrbhjv, 2021-01-13:
The colcon.meta is as I sent,

However, this is in the header generated

[image]

pablogs9, 2021-01-13:
Yes, your `colcon.meta` is wrong. Set it as I propose here:

> Ok... sorry, check your configuration:
> 
> ```
>                 "-DRMW_UXRCE_DEFAULT_UDP_IP=10.46.28.3",
>                 "-DRMW_UXRCE_DEFAULT_UDP_IP=8888",
> ```
> 
> The second should be `-DRMW_UXRCE_DEFAULT_UDP_PORT`
> 
> Rebuild your library and tell me what happens.

mrbhjv, 2021-01-13:
God damn it, small huge mistake there...I'll change it, completely missed that

mrbhjv, 2021-01-13:
[image]

Just to close it, thanks again for the quick and assertive response. On the bright side, it was aligned with what I thought, that was a mistake on my side on the configuration, unfortunately, I didn't notice it faster.

closed 2021-01-13 by mrbhjv

mrbhjv, 2021-01-13:
Replied here https://github.com/micro-ROS/micro_ros_setup/issues/246 👍