micro-ROS/micro_ros_setup #256: Retry connections

mrbhjv, 2021-01-14:
Hi,
First, I know this is not exactly the place for this, but I didn't know exactly where to place it.
If I use the examples for the apps, if the Agent is not present when initializing the app, the system will fail and it should try to continue, causing unexpected behaviors.

How/where, would be the solution thought by you to handle this problem? I want that, if the agent host disconnects by any circumstance, it also goes to this initialization again and retry (with low rate) until succeeded. I this that it's on `rclc_support_init` that it will fail and return something different to `RCL_RET_OK`, but maybe there's some other check I could do. I could also check that the host is available and the port is up, but that doesn't mean it's the agent who's up.

I'm running with FreeRTOS.

Thanks

pablogs9, 2021-01-14:
Hello @mrbhjv, the user app should be in charge of checking the result of the micro-ROS initialization and handle possible errors and agent disconnections.

Currently, we are working on a feature that allows checking if the agent is available to let the user check agent's status before trying to establish a session. You can find some related info here: https://github.com/micro-ROS/micro_ros_arduino/issues/40

When we finish this feature you will be able to call a fast function that determines if the agent is ready for connecting to it, do this solves your issue?

Maybe @jamoralp can tell us something more about this feature.

mrbhjv, 2021-01-14:
Hi @pablogs9, yes, that feature would be helpful. But I'm quite sure I can live with checking the return values for now. I didn't know about the spin_some return, which will do for the running portion.

Thanks :)

jamoralp, 2021-01-14:
Hi, as my colleague @pablogs9  was mentioning, yes, we are very close to finish a "ping" feature from the client application to the agent to easiliy query the latter status. Let's keep this issue closed for now, and I will ping you when it is ready to use :slightly_smiling_face:

mrbhjv, 2021-01-14:
One more question, sorry to do it on a closed issue, but, if I get disconnected, what should be my process. What entities should be destroyed, what entities won't need a change? I would prefer to work with static memory as much as possible, so, I would prefer not to have to re-instantiate anything. Should I need to run this again for example?
I'm wondering if what returns an error is while running and I have had already a successful connection. Do I need to reconnect by doing the init again?

```
rclc_support_init(&support, 0, NULL, &allocator);
rclc_node_init_default(&node, "freertos_int32_publisher", "", &support);
rclc_publisher_init_default(
      &publisher, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
      "freertos_int32_publisher")
...
```

pablogs9, 2021-01-14:
Here you have an example where a couple of FreeRTOS tasks are being created in order to publish and subscribe: https://github.com/micro-ROS/freertos_apps/blob/foxy/apps/crazyflie_demo/app.c

Be careful because multitask support for micro-ROS is nos fully tested, just use this to check how we are doing:

```c
        rc = rcl_subscription_fini(&sub_sensors, &node);
        rc = rcl_node_fini(&node);
        rc = rcl_shutdown(&context);
```

to close and destroy all the micro-ROS entities before trying to reconnect to the agent.

Regarding the memory, all the major memory consumption in micro-ROS is handled by static memory pools in RMW. So, you should only see some dynamic memory allocations during init and destroy because of some topic/node names strings.

mrbhjv, 2021-01-14:
What I don't have is the context, as it's initialized with the rclc_support_init. I'm wondering if I should do `rcl_shutdown(&support.context)` ? Or it won't be necessary?

None of the examples I looked, that initialize it with `rclc` have the shutdown of the context

pablogs9, 2021-01-14:
Sorry I shared an RCL example, check [here](https://github.com/ros2/rclc/blob/6e45370f4bb5434e5bd81974829795284604cf4b/rclc_examples/src/example_executor_convenience.c#L168) how to do this with the RCLC

mrbhjv, 2021-01-14:
Excellent, thanks. Now I'll close this.

closed 2021-01-14 by mrbhjv