micro-ROS/micro_ros_setup #260: Starting a project on MicroROS

Vishnu-Vijay95, 2021-01-19:
Hi All,
 I am going to start a project on micro Ros on freeRTOS on olimex STM32E407 board and i am using segger jlink for programing.
I want to know what are the steps i need to follow while starting a new project on microROS.

pablogs9, 2021-01-19:
Hello @Vishnu-Vijay95, have you tried the steps in our [Olimex STM32E407 + FreeRTOS tutorial](https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/freertos/)?

If so, could you detail which problems have you find following these steps?

pablogs9, 2021-01-19:
PD: maybe to get in touch with the micro-ROS build system you can test the [micro-ROS for Linux tutorial](https://micro-ros.github.io/docs/tutorials/core/first_application_linux/)

Vishnu-Vijay95, 2021-01-22:
Hai @pablogs9 , I tried to flashing the microROS firmware using the flashing script **ros2 run micro_ros_setup flash_firmware.sh** but this script is only for OpenOCD programmer. But i don't have this hardware and i am using segger jlink for programing. I have successfully flashed micro-ROS binary using an alternative method. Next have done the micro_ROS agent and stared the agent using command **ros2 run micro_ros_agent micro_ros_agent serial --dev [device]**. For the connection from USART to USB i have used a USB to TTL convertor, but i am getting an error **open device error  errno:2**.

Vishnu-Vijay95, 2021-01-22:
Hai @pablogs9 , I tried flashing the microROS firmware using the flashing script **ros2 run micro_ros_setup flash_firmware.sh** but this script is only for OpenOCD programmer, i don't have this hardware and i am using segger jlink for programing. I have successfully flashed micro-ROS binary using an alternative method. Next have done the micro_ROS agent and stared the agent using command **ros2 run micro_ros_agent micro_ros_agent serial --dev [device].** For the connection from USART to USB i have used a USB to TTL convertor, but i am getting an error **open device error errno:2**.

pablogs9, 2021-01-22:
Could you please share the whole command for `ros2 run micro_ros_agent micro_ros_agent serial --dev [device]` and the output of `ls /dev/serial/by-id/`?

Vishnu-Vijay95, 2021-01-22:
Thanks @pablogs9 for your support . i had rectified the problem. I was not giving the entire device name was just giving the device ID. Now tried to subscribe the ping topic and getting the warning : **topic [/microROS/ping] does not appear to be published yet**. whether it is due to incorrect flashing?

pablogs9, 2021-01-22:
Can you paste here the output of your agent? Remember to add the flag `-v6` for verbosity.

Vishnu-Vijay95, 2021-01-22:
This the output of the agent :
**command : ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0
[1611298569.543361] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3**

pablogs9, 2021-01-22:
please use the flag `-v6` in the agent command

Vishnu-Vijay95, 2021-01-22:
**command: ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 -v6
[1611302815.849160] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1611302815.849394] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6**
logger setup is active now.

pablogs9, 2021-01-22:
can you share the ouput of `ls /dev/serial/by-id/`?

Vishnu-Vijay95, 2021-01-22:
**ls /dev/serial/by-id/ : usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0**

pablogs9, 2021-01-22:
Can you try `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0 -v6`?

Vishnu-Vijay95, 2021-01-22:
command : `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0 -v6`
[1611304478.932185] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1611304478.932419] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6

command : `source /opt/ros/foxy/setup.bash`
command : `ros2 topic echo /microROS/ping`
WARNING: topic [/microROS/ping] does not appear to be published yet
Could not determine the type for the passed topic

pablogs9, 2021-01-22:
Can you show how have you connected the UART-USB cable to the board?

Vishnu-Vijay95, 2021-01-22:
yes. How can i share ?

pablogs9, 2021-01-22:
Documentation of your cable model and a photo of the conection to the board.

Vishnu-Vijay95, 2021-01-22:
For Connection i am using normal jumper cables.
Board connection, i am attaching here 
https://drive.google.com/file/d/1QapEIOaiNv0R2j23nJ7xyIljGz06Jq--/view?usp=sharing
https://drive.google.com/file/d/1JAs_UaKSWFtlOuhJmRkZ71rjoyPc4j-x/view?usp=sharing

pablogs9, 2021-01-22:
Have you checked the correctness of Tx and Rx?

Vishnu-Vijay95, 2021-01-22:
Yes i checked by looping the Tx and Rx i am getting the output. I don't know  whether its any flashing error. I had flashed micro-ROS.bin file to the MCU. I had used segger JLink for programming so the firmware flashing script didn't work for me as its written for OpenOCD programmer.

pablogs9, 2021-01-22:
Try to use the on board USB following the tutorial but using the following configuration command:

```bash
ros2 run micro_ros_setup configure_firmware.sh ping_pong --transport serial-usb
```

Vishnu-Vijay95, 2021-01-27:
Hi @pablogs9 i tried this configuration command and getting the following output
```
Configuring firmware for freertos platform olimex-stm32-e407
Using serial USB device. EXPERIMENTAL.
cp: cannot stat '/home/ignitarium/Desktop/mros_ws/firmware/freertos_apps/microros_olimex_e407_extensions/Src/olimex_e407_usb_transport.c': No such file or directory
cp: cannot stat '/home/ignitarium/Desktop/mros_ws/firmware/freertos_apps/microros_olimex_e407_extensions/Inc/olimex_e407_usb_transport.h': No such file or directory
Configured serial-usb mode with agent
```
Shows some header files are missing. Can you help me in flashing the board using  segger JLink

pablogs9, 2021-01-27:
Sorry I didn't remember that we had this USB transport as experimental. Try to use [this branch](https://github.com/micro-ROS/freertos_apps/tree/feature/usb_olimex) for the repo in firmware/freertos_apps.

pablogs9, 2021-01-27:
I just have replicated the tutorial using an Olimex STM32-E407 board and serial communication and everything works as expected. Do you have any way of checking if your board is generating the UART? Maybe using another USB-UART cable? Checking with an oscilloscope?

Vishnu-Vijay95, 2021-01-27:
Ok @pablogs9, How you flashed the the olimex Board? You used [ARM-USB-TINY-H ](https://www.olimex.com/Products/ARM/JTAG/ARM-USB-TINY-H/)for programing. I don't have this hardware.

pablogs9, 2021-01-27:
Sorry, I can't test with another programmer because I only have this one.

Vishnu-Vijay95, 2021-01-27:
Ok @pablogs9, How you flashed the the olimex Board? You used [ARM-USB-TINY-H ](https://www.olimex.com/Products/ARM/JTAG/ARM-USB-TINY-H/)for programing? I don't have this hardware.

Vishnu-Vijay95, 2021-01-27:
I am getting some packages stderr while building, do you encounter same issue.
```
Summary: 59 packages finished [3min 12s]
  37 packages had stderr output: action_msgs actionlib_msgs builtin_interfaces composition_interfaces diagnostic_msgs example_interfaces geometry_msgs libyaml_vendor lifecycle_msgs micro_ros_msgs microxrcedds_client nav_msgs rcl rcl_action rcl_interfaces rcl_lifecycle rcl_logging_noop rclc rclc_lifecycle rcutils rmw rmw_implementation rmw_microxrcedds rosgraph_msgs rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_microxrcedds_c sensor_msgs shape_msgs statistics_msgs std_msgs std_srvs stereo_msgs test_msgs trajectory_msgs unique_identifier_msgs visualization_msgs

```
but the build was complete with stderr.

pablogs9, 2021-01-27:
Yes, this is not a problem. All packages have been built but due to cross-compilation, there are some warnings.

Vishnu-Vijay95, 2021-01-27:
Ok Thanks @pablogs9,  i will try some alternative to flash.

Vishnu-Vijay95, 2021-01-27:
The freeRTOS + microROS ping_pong is available with the RTOS? or we have to port this package into freeRTOS? Why i asked because if i have to do a custom project on microROS + freeRTOS, whether i am able to use the same build system or i have to port microROS application to freeRTOS entirely?

pablogs9, 2021-01-27:
Not entirely but the parts that use the FreeRTOS API in your user code.

Vishnu-Vijay95, 2021-01-27:
In the ping_pong pgm i had listed all ros2 topics the output was
```
ros2 topic list
/parameter_events
/rosout
```
the topic micoROS/ping is not available. what will be the possible problem.

pablogs9, 2021-01-27:
Your micro-ROS client is not creating entities in the micro-ROS agent, please double-check the serial transport and if possible try with another serial-to-USB cable.

Vishnu-Vijay95, 2021-01-28:
I tested USB-serial communication on Zephyr its working fine for me. I tried USB-serial communication in freeRTOS with the repo provided by you i am getting an error while building I will state it below.
```

home/ignitarium/Desktop/microros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/serial/serial_transport_external.c:24: undefined reference to `CDC_DeInit_FS'
collect2: error: ld returned 1 exit status
make: *** [Makefile:386: /home/ignitarium/Desktop/microros_ws/firmware/freertos_apps/microros_olimex_e407_extensions/build/micro-ROS.elf] Error 1
```

the build is  failing with the ERROR code 1

pablogs9, 2021-01-28:
It is ok for you to use Zephyr?

Vishnu-Vijay95, 2021-01-28:
Actually looking forward to use freeRTOS. Can u help me?

Vishnu-Vijay95, 2021-01-28:
Thanks @pablogs9, i rectified it. It was my UART  problem. Thanks for your support.

pablogs9, 2021-01-28:
Nice, I'm closing this issue.

Please, don't hesitate to reopen if you have any question.

closed 2021-01-28 by pablogs9

Vishnu-Vijay95, 2021-02-01:
Hi @pablogs9 ,  I going to start a custom app in microROS +freeRTOS. where are the changes i have to make to register <my_app>

pablogs9, 2021-02-01:
just create your app in `firmware/freertos_apps/apps/` and execute the `ros2 run micro_ros_setup` and you should see your new folder

Vishnu-Vijay95, 2021-02-01:
Is this command complete? i just created `my_app` folder and saved `app.c `and `app-colon.meta` file in `firmware/freertos_apps/apps/`

pablogs9, 2021-02-01:
Paste here the output of `ros2 run micro_ros_setup`

Vishnu-Vijay95, 2021-02-01:
this it the output
```
usage: ros2 run [-h] [--prefix PREFIX] package_name executable_name ...
ros2 run: error: the following arguments are required: executable_name, argv
```

pablogs9, 2021-02-01:
oh sorry: `ros2 run micro_ros_setup configure_firmware.sh`

Vishnu-Vijay95, 2021-02-01:
Ya now its showing my_apps. Now it it does not require any additional configuration, right? But while building i am getting an error:
```
/home/microros_ws/firmware/toolchain/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /home/Desktop/microros_ws/firmware/freertos_apps/microros_olimex_e407_extensions/build/micro-ROS.elf section `.bss' will not fit in region `RAM'
/home/Desktop/microros_ws/firmware/toolchain/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: region `RAM' overflowed by 76472 bytes
collect2: error: ld returned 1 exit status
make: *** [Makefile:372: /home/Desktop/microros_ws/firmware/freertos_apps/microros_olimex_e407_extensions/build/micro-ROS.elf] Error 1
```
But i have written the same ping_pong code in my app.

pablogs9, 2021-02-01:
paste here your `app-colcon.meta`

Vishnu-Vijay95, 2021-02-01:
```
 "names": {
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=2",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=3",
                "-DRMW_UXRCE_MAX_SERVICES=0",
                "-DRMW_UXRCE_MAX_CLIENTS=0",
                "-DRMW_UXRCE_MAX_HISTORY=4",
            ]
        }
    }
}
```

pablogs9, 2021-02-01:
Try to build it in a clean environment. The ping_pong app should fit the Olimex STM32-E407 RAM, please try it in a clean environment, and if it still does not fit (super strange because the ping_pong app works), you can decrease the entities, for example, `DRMW_UXRCE_MAX_HISTORY`.

You can find more information about middleware optimization [here](https://micro-ros.github.io/docs/tutorials/core/microxrcedds_rmw_configuration/).

Vishnu-Vijay95, 2021-02-01:
I am able to build the `ping_pong `app successfully, i have copied and pasted the same code on `my_app` then its showing RAM overflow error. Are u sure that there is no additional settings needed to be done to register the my_apps folder means, any cmake list edit is required?

pablogs9, 2021-02-01:
As far as I know, you only need to create a new folder in the apps folder. You have copied literally the same code?

Vishnu-Vijay95, 2021-02-01:
Earlier i had made some changes as well, since it was not building i literally copied the same code and tried again. But the build was failing

pablogs9, 2021-02-01:
I guess that you are doing some changes. I have replicated here the procedure of copying the ping-pong app as we are talking and everything builds fine:

https://asciinema.org/a/Lh0hPfRIMOJhYRvEvA9DmGYSd

Vishnu-Vijay95, 2021-02-01:
Ok, let me check

Vishnu-Vijay95, 2021-02-01:
> I guess that you are doing some changes. I have replicated here the procedure of copying the ping-pong app as we are talking and everything builds fine:
> 
> https://asciinema.org/a/Lh0hPfRIMOJhYRvEvA9DmGYSd
Ok Thanks. Rectified

Vishnu-Vijay95, 2021-02-01:
I have an logical doubt. If i need to interface an sensor to mcu and thereby publish the sensor data, where i have to write my subscriber? whether its ros2 subscriber? if yes, what are the steps to interface microROS and ros2? i think my question is logically correct

pablogs9, 2021-02-01:
You will have a sensor connected to an MCU, this MCU will run micro-ROS and you need to create a publisher that send the sensor data.

This micro-ROS code needs to connect to the micro-ROS Agent, and once it is connected, you will be able to see the data being published in ROS 2 with the normal commands such as:

`ros2 topic echo ...`

Vishnu-Vijay95, 2021-03-02:
Hi @pablogs9, I am trying to write a microROS+ freertos program to control a kobuki turtle bot 2 robot. I had taken your [code ](https://github.com/pablogs9/kobuki_esp32_micro_ros/blob/96a65c9156edbc560c81849c45aecb680518b7e0/main/kobuki_microros.c#L102)  as reference. I am doing this on an olimex STM32E407 board. But while building i am getting an error `drive_base_msgs/msg/base_info.h: No such file or directory` . Thing this header file is missing.

pablogs9, 2021-03-02:
That's because [drive_base](https://github.com/micro-ROS/drive_base) is a custom message. If you have a common micro-ROS environment, just:

```bash
git clone https://github.com/micro-ROS/drive_base firmware/mcu_ws/uros/drive_base
ros2 run micro_ros_setup build_firmware.sh
```

And you should be able to use it.

Vishnu-Vijay95, 2021-03-02:
Thanks, now `kobuki.h` is also having same error. Header file missing.

[36 more replies]