micro-ROS/micro_ros_setup #262: About security with FreeRTOS and FreeRTOS+TCP

mrbhjv, 2021-01-21:
Hi,

I couldn't easily find anything that referred to encryption or security in general for the communication between the agent and a board (client?). I'd like to know if there's any plan for it, or maybe already some feature available? I know DDS offers the security component, is it already available on micro XRCE?

I'm currently working FreeRTOS and FreeRTOS+TCP

pablogs9, 2021-01-21:
Hello, there no active line of investigation or development related to micro-ROS and security.

As you say you can configure security in the DDS by using the [custom refs](https://micro-ros.github.io/docs/tutorials/core/create_dds_entities_by_ref/). This way the XRCE communication will be plain.

Also, there is the possibility of creating custom transports for XRCE. Currently, we have an undocumented approach for creating them but we are preparing a standard way of [implementing XRCE transports](https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/179). With these custom external transports, you should be able to use TCP+TLS or any secured approach for communicating between XRCE client and agent.

I hope that this answers, let us know and reopen if you have any other question.

closed 2021-01-21 by pablogs9

mrbhjv, 2021-01-21:
It does answer, thanks. Looking forward to trying out the custom transport