micro-ROS/micro_ros_setup #263: open device error when running microros agent ping_pong tutorial usb serial

luminize, 2021-01-21:
Hi,
Following the tutorial (verbatim) I've flashed my Olimex-STM32-407 board and built the micro-ros agent.
during building the agent for the first time I gave me "error" message output, and after some searching I guess this "error" can be ignored? Anyway, the second time the building did not give errors.

```
bdb@ThinkPad-P51:~/projects/microcos_ws$ ros2 run micro_ros_setup build_agent.sh
Building micro-ROS Agent
Starting >>> microcdr
[0.181s] WARNING:colcon.colcon_core.prefix_path.colcon:The path '/home/bdb/projects/microros_ws/install' in the environment variable COLCON_PREFIX_PATH doesn't exist
[0.182s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/bdb/projects/microros_ws/install/micro_ros_setup' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.182s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/bdb/projects/microros_ws/install/micro_ros_setup' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> microxrcedds_agent
Finished <<< microcdr [0.96s]                                     
Starting >>> rosidl_typesupport_microxrcedds_c
Finished <<< rosidl_typesupport_microxrcedds_c [0.67s]                     
Starting >>> micro_ros_msgs
Finished <<< micro_ros_msgs [4.75s]                                        
[Processing: microxrcedds_agent]                              
[Processing: microxrcedds_agent]                                     
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                        
--- stderr: microxrcedds_agent                                          
Cloning into 'spdlog'...
Cloning into 'sanitizers'...
Cloning into 'fastdds'...
Already on 'master'
CMake Warning (dev) at cmake/FindASan.cmake:25 (option):
  Policy CMP0077 is not set: option() honors normal variables.  Run "cmake
  --help-policy CMP0077" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  For compatibility with older versions of CMake, option is clearing the
  normal variable 'SANITIZE_ADDRESS'.
Call Stack (most recent call first):
  cmake/FindSanitizers.cmake:37 (find_package)
  tests/CMakeLists.txt:52 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

Switched to a new branch '2.0.x'
Submodule 'thirdparty/android-ifaddrs' (https://github.com/michalsrb/android-ifaddrs.git) registered for path 'thirdparty/android-ifaddrs'
Submodule 'thirdparty/asio' (https://github.com/chriskohlhoff/asio.git) registered for path 'thirdparty/asio'
Submodule 'thirdparty/fastcdr' (https://github.com/eProsima/Fast-CDR.git) registered for path 'thirdparty/fastcdr'
Submodule 'thirdparty/tinyxml2' (https://github.com/leethomason/tinyxml2.git) registered for path 'thirdparty/tinyxml2'
Cloning into '/home/bdb/projects/microcos_ws/build/microxrcedds_agent/fastdds/src/fastdds/thirdparty/android-ifaddrs'...
Cloning into '/home/bdb/projects/microcos_ws/build/microxrcedds_agent/fastdds/src/fastdds/thirdparty/asio'...
Note: switching to 'v1.4.2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 1549ff12 Replace STRING(PREPEND ..) in CMakeLists.txt to support older CMake versions
Cloning into '/home/bdb/projects/microcos_ws/build/microxrcedds_agent/fastdds/src/fastdds/thirdparty/fastcdr'...
Cloning into '/home/bdb/projects/microcos_ws/build/microxrcedds_agent/fastdds/src/fastdds/thirdparty/tinyxml2'...
---
Finished <<< microxrcedds_agent [2min 37s]
Starting >>> micro_ros_agent
--- stderr: micro_ros_agent                                    
/usr/bin/ld: warning: libfastrtps.so.2, needed by /opt/ros/foxy/lib/librmw_fastrtps_shared_cpp.so, may conflict with libfastrtps.so.2.0.2
---
Finished <<< micro_ros_agent [4.29s]

Summary: 5 packages finished [2min 41s]
  2 packages had stderr output: micro_ros_agent microxrcedds_agent

bdb@ThinkPad-P51:~/projects/microcos_ws$ ros2 run micro_ros_setup build_agent.sh
Building micro-ROS Agent
Starting >>> microcdr
[0.183s] WARNING:colcon.colcon_core.prefix_path.colcon:The path '/home/bdb/projects/microros_ws/install' in the environment variable COLCON_PREFIX_PATH doesn't exist
[0.183s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/bdb/projects/microros_ws/install/micro_ros_setup' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.183s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/bdb/projects/microros_ws/install/micro_ros_setup' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> microxrcedds_agent
Finished <<< microcdr [0.12s]                                                                     
Starting >>> rosidl_typesupport_microxrcedds_c
Finished <<< microxrcedds_agent [0.23s]                                                                                    
Finished <<< rosidl_typesupport_microxrcedds_c [0.18s]
Starting >>> micro_ros_msgs
Finished <<< micro_ros_msgs [0.36s]                     
Starting >>> micro_ros_agent
Finished <<< micro_ros_agent [0.14s]                  

Summary: 5 packages finished [0.91s]

```
When starting the agent I get a device opening error, suggesting I run as superuser.
```
bdb@ThinkPad-P51:~/projects/microcos_ws$ rource install/local_setup.bash
bdb@ThinkPad-P51:~/projects/microcos_ws$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-ZEPHYR_Zephyr_micro-ROS_20363942394E500A-if00
[1611223526.155757] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/serial/by-id/usb-ZEPHYR_Zephyr_micro-ROS_20363942394E500A-if00, errno: 13. Please re-run with superuser privileges.
Error while starting serial agent!
```
My setup:
- Ubuntu 20.04
- ROS2 Foxy, running from packages

Any help would be appreciated,
Best, Bas

pablogs9, 2021-01-21:
Hello @luminize, regarding the build error maybe @jamoralp can give us some support.

Regarding the runtime error, it seems that you don't have permission for read/write to the serial port. Check [this](https://unix.stackexchange.com/questions/14354/read-write-to-a-serial-port-without-root)

luminize, 2021-01-21:
Thanks @pablogs9 your link to adding myself to the dialout group solved this.
```
bdb@bdb-ThinkPad-P51:~$ groups bdb
bdb : bdb adm dialout cdrom sudo dip plugdev lpadmin lxd sambashare
bdb@bdb-ThinkPad-P51:~$ cd projects/microcos_ws/
bdb@bdb-ThinkPad-P51:~/projects/microcos_ws$ source /opt/ros/foxy/setup.bash 
bdb@bdb-ThinkPad-P51:~/projects/microcos_ws$ source install/local_setup.bash 
bdb@bdb-ThinkPad-P51:~/projects/microcos_ws$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-
/dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_0669FF323338424E43162338-if02
/dev/serial/by-id/usb-ZEPHYR_Zephyr_micro-ROS_20363942394E500A-if00
bdb@bdb-ThinkPad-P51:~/projects/microcos_ws$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-ZEPHYR_Zephyr_micro-ROS_20363942394E500A-if00 
[1611225798.867831] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1611225799.959007] info     | Root.cpp           | create_client            | create                 | client_key: 0x14790D24, session_id: 0x81
[1611225799.959045] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x343477540, address: 1
```
and now I can find the topics on the olimex board
```
bdb@bdb-ThinkPad-P51:~$ source /opt/ros/foxy/setup.bash 
bdb@bdb-ThinkPad-P51:~$ ros2 topic list
/microROS/ping
/microROS/pong
/parameter_events
/rosout
```
Would you like me to close this issue? or keep it open for the build warning?

luminize, 2021-01-21:
@pablogs9 Thanks for the quick help! Much appreciated.

jamoralp, 2021-01-21:
A little update regarding this: I found that the CMake CMP0077 warning you are getting is because of the sanitizer tool we are using, though it only appears when using CMake 3.16.3 or greater. I just updated my `microros/base:foxy` Docker image and I checked indeed that the CMake version has been bumped to that one, so it makes sense that this is being displayed now.
I already uploaded a fix in the `Agent` repo. Please check it out https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/207. This patch will also be incorporated into `foxy` branch, but I would really appreciate if you can test it on your local setup and confirm that the warning is not there anymore before merging the pull request. Thanks!

pablogs9, 2021-01-22:
Closing since https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/207 merged

closed 2021-01-22 by pablogs9