micro-ROS/micro_ros_setup #264: Q: best way to work with multiple types of microprocessor targets

luminize, 2021-01-21:
I have different boards for trying out various programs and tutorials.
When I tried to create the pin_pong app on the STM32 discovery board after I ran it on the Olimex-STM32-E407 I got the following message:
```
bdb@bdb-ThinkPad-P51:~/projects/microcos_ws$ ros2 run micro_ros_setup create_firmware_ws.sh zephyr discovery_l475_iot1
Firmware already created. Please delete /home/bdb/projects/microcos_ws/firmware folder if you want a fresh installation.
```
I removed the offending directory and reran the command.
now configuring the firmware without `[APP]` and `[OPTIONS]` shows I could choose between Zephyr and discovery_l475_iot?
```
bdb@bdb-ThinkPad-P51:~/projects/microcos_ws$ ros2 run micro_ros_setup configure_firmware.sh
micro-ROS application name must be provided: ros2 run micro_ros_setup configure_firmware.sh [app name] [options]
Available apps for Zephyr and discovery_l475_iot1:
+-- add_two_ints_service
+-- attitude_estimator
+-- int32_publisher
+-- int32_wifi_publisher
+-- openmanipulator_tof
+-- ping_pong
+-- sensors_publisher
+-- tof_ws2812
+-- vl53l1x_tof_sensor
```
My question boils down to: How do I add or change a platform for an RTOS, and flash to that specific platform (i.e. an ESP32 on my desktop)?. I could not find how to do that in the tutorials.

pablogs9, 2021-01-21:
Sorry but currently, the micro-ROS build system only supports one workspace per target because it needs to handle a lot of target-dependent files.
You need to create a new workspace for each of your boards.

Hope that this solves your question, I'm closing but feel free to reopen if you want to propose some change that allows having multitarget workspaces.

closed 2021-01-21 by pablogs9

luminize, 2021-01-21:
no, that's ok, I was wondering if I did something wrong. thx