micro-ROS/micro_ros_setup #265: how can I build a project by my own

Needrom, 2021-01-24:
I learn some thing from a close issue which is [https://github.com/micro-ROS/micro_ros_setup/issues/202](url)

but i didn't see any thing about how to add a base project with gpio lwip freertos by ourselves. I want to build it from a makefile project from stm32cubuMX, but don't know how. I try compare the project with a base project from  stm32CubeMX and a project in /firmware/......./microros_olimex_e407_extensions/   ,with no idea to transfer my base project to a project can built by a script which is "ros2 run micro_ros_setup build_firmware.sh"

- the board I use
the same chip with olimex_e407.   stm32f407zgt6

- if any could tell me how to add a project to firmware from a base Makefile project from stm32CubeMX?

Needrom, 2021-01-25:
Could not build it just use some thing like "ros2 run micro_ros_setup create_firmware_ws.sh freertos my-own-board-name" 
I think it just fetch the repo from github if the script can find that link

pablogs9, 2021-01-25:
You need to create micro-ROS support for your board as it is not supported. Some micro-ROS for freeRTOS board support are based in STMCubeMX, [check this](https://github.com/micro-ROS/freertos_apps/pull/46) for an example on how to create new board support. 

Although there is a lot of files, most of them are autogenerated, so at the end, you can analyze the `main.c` and the `Makefile` to see what is being done there.

Needrom, 2021-01-26:
> You need to create micro-ROS support for your board as it is not supported. Some micro-ROS for freeRTOS board support are based in STMCubeMX, [check [...]

Thanks for reply. I follow your step ,but I got this.
[image]

pablogs9, 2021-01-26:
I guess that these errors are not related to micro-ROS but related to your FreeRTOS port

Needrom, 2021-01-26:
> I guess that these errors are not related to micro-ROS but related to your FreeRTOS port

this may cause by 

```

# Removing heap4 manager while being polite with STM32CubeMX
#TMPVAR := $(C_SOURCES)
#C_SOURCES := $(filter-out Middlewares/Third_Party/FreeRTOS/Source/portable/MemMang/heap_4.c, $(TMPVAR))

```

I comment this two line so I can build it. But any thing I should Know if I put this two line disable

pablogs9, 2021-01-26:
Could you share some details about your project? Are you using a development hardware board that you can contribute to micro-ROS?

Can we close the issue?

Thanks!

Needrom, 2021-01-26:
sure. I'm going to have this to be detail in markdown in a new repo. I just use some base gpio function for now, and uart testing no yet. I'm going to add RS485 in it. I'm using a development hardware board, will share it all after I test a more bit later.

Needrom, 2021-01-28:
some thing is Not going so well. as like I found so much microros thing I haven't write it the freertos.c. After I add the code relate to rcl , wrong come out.

[image]

pablogs9, 2021-01-28:
Could you please provide steps for replicating this?

Needrom, 2021-01-29:
### Makefile of Mine

I Makefile little by little copy from the ```microros_nucleo_f446re_extensions``` 

And finaly , firmware can build without changing the freertos.c in $(my_project)/Core/Src/

With Makfile as follow.

```
##########################################################################################################################
# File automatically-generated by tool: [projectgenerator] version: [3.11.0-B13] date: [Mon Jan 25 13:25:54 CST 2021] 
##########################################################################################################################

# ------------------------------------------------
# Generic Makefile (based on gcc)
#
# ChangeLog :
#	2017-02-10 - Several enhancements + project update mode
#   2015-07-22 - first version
# ------------------------------------------------

######################################
# target
######################################
TARGET = micro-ROS

######################################
# building variables
######################################
# debug build?
DEBUG = 1
# optimization
ifeq ($(DEBUG), 1)
        OPT = -O0
        BUILD_TYPE = Debug
else
        OPT = -Og
        BUILD_TYPE = Release
endif

#######################################
# paths
#######################################
# Build path
EXTENSIONS_DIR = $(shell pwd)
TOPFOLDER = $(EXTENSIONS_DIR)/../..
UROS_DIR = $(TOPFOLDER)/mcu_ws
BUILD_DIR = $(EXTENSIONS_DIR)/build

######################################
# source
######################################
# C sources
C_SOURCES =  \
Core/Src/main.c \
Core/Src/gpio.c \
Core/Src/freertos.c \
Core/Src/usart.c \
Core/Src/allocators.c \
Core/Src/custom_memory_manager.c \
Core/Src/stm32f4xx_it.c \
Core/Src/stm32f4xx_hal_msp.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_eth.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc_ex.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_flash.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_flash_ex.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_flash_ramfunc.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_gpio.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_dma_ex.c \
[... 437 lines cut ...]
  osThreadAttr_t attributes;
  memset(&attributes, 0x0, sizeof(osThreadAttr_t));
  attributes.stack_size = 5 * 3000;
  attributes.priority = (osPriority_t)osPriorityNormal1;
  osThreadNew(appMain, NULL, &attributes);
  osDelay(500);
  char ptrTaskList[500];
  vTaskList(ptrTaskList);
  printf("**********************************\n");
  printf("Task  State   Prio    Stack    Num\n");
  printf("**********************************\n");
  printf(ptrTaskList);
  printf("**********************************\n");

  TaskHandle_t xHandle;
  xHandle = xTaskGetHandle("microROS_app");
```

and this One

```
 if (eTaskGetState(xHandle) != eSuspended && availableNetwork)
```

the Building error comes out.

pablogs9, 2021-01-29:
Could you please provide steps for replicating this?

This is:
- a code that I can download
- a set of steps for having the exact same situation as you
- a build command that ends in your very same error

Needrom, 2021-02-03:
sorry for taking so long, here is my repo that you can check the project I use and the config file I use
[https://github.com/Needrom/Microros-on-DIscovery-board](url)

pablogs9, 2021-02-03:
With these changes, I have been able to build it: https://github.com/Needrom/Microros-on-DIscovery-board/pull/1

Needrom, 2021-02-03:
> With these changes, I have been able to build it: [Needrom/Microros-on-DIscovery-board#1](https://github.com/Needrom/Microros-on-DIscovery-board/pull/1)

Yes!It can build now ! appreciate for that

closed 2021-02-03 by Needrom

pablogs9, 2021-02-03:
Cool, let us know about your progress and projects!

Needrom, 2021-02-04:
Sure. any way I could contribute my project?