micro-ROS/micro_ros_setup #268: Olimex with NuttX - no connection from client to agent

JanStaschulat, 2021-01-26:
## Issue template

- Hardware description: Olimex board with STM32
- RTOS: NuttX
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue

I followed all installation steps on page
https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/nuttx/

I am using a Linux laptop with Ubuntu 18.04 and build ROS 2 foxy from the sources. 

compiling the micro-ros agent finished but with this warnnig that libfastrtps.so.2 may conflict with libfastrtps.so.2.0.2:
```C
Starting >>> micro_ros_agent
--- stderr: micro_ros_agent                                       
/usr/bin/ld: warning: libfastrtps.so.2, needed by /home/jst3si/ros2_foxy/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so, may conflict with libfastrtps.so.2.0.2
---
Finished <<< micro_ros_agent [10.5s]
```
Then I connected all cables to the Olimex board. 
- ARM-USB-OCD-H cable
- serial cable
- mini-USB cable

```C
jst3si@RNGX5596:~/olimex_ws/log$ ls /dev/serial/by-id/
usb-15ba_Olimex_OpenOCD_JTAG_ARM-USB-OCD-H_OLC96F84-if01-port0
usb-FTDI_TTL232R-3V3_FTAPR6AT-if00-port0
usb-NuttX_CDC_ACM_Serial_0-if00
```
I am in dialout group

```C
jst3si@RNGX5596:~/olimex_ws/log$ groups
jst3si dialout sudo tracing docker
```

1. I started the agent in a terminal
2. I connected with screen to Olimex board and typed uros_pingpong
3. I checked incoming ROS 2 topic in third window.

Error: there is not connection established between client and agent.  Please support. 

#### Expected behavior
working demo with uros_pingpong

#### Actual behavior

##### Agent terminal:
```C
^Cjst3si@RNGX5596:~/olimex_ws$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-FTDITL232R-3V3_FTAPR6AT-if00-port0
[1611692529.175394] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1611692875.210808] info     | Root.cpp           | create_client            | create                 | client_key: 0x57D17F07, session_id: 0x81
[1611692875.210939] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1473347335, address: 1
[1611692875.242586] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1473347335, address: 1
[1611692875.290598] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1473347335, address: 1
[1611692875.386420] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1473347335, address: 1
[1611692875.594591] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1473347335, address: 1
[1611692875.993963] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1473347335, address: 1
[1611692876.793547] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1473347335, address: 1
[1611692878.392016] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1473347335, address: 1
[1611692881.589075] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1473347335, address: 1
[1611692887.999066] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1473347335, address: 1
^Cjst3si@RNGX5596:~/olimex_ws$ 
```

##### nsh shell output (on Olimex board):

```C
NuttShell (NSH)
nsh>
nsh>uros_pingpong
Serial mode => dev: /dev/ttyS0
[ERROR] [0946684931.554000000] [rclc]: [rclc_init] Error in rcl_init: failed to create node session on Micro ROS Agent., at /home/jst3si/olimex_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c:317, at /home/jst3si/olimex_ws/firmware/mcu_ws/uros/rcl/rcl/src/rcl/init.c:233

Failed status on line 86: 1. Aborting.
nsh>
```
The error message came after some timeout.

##### Linux shell with ROS 2:

```C
jst3si@RNGX5596:~$ ros2 topic list
/parameter_events
/rosout
jst3si@RNGX5596:~$ ros2 topic echo /microROS/ping
WARNING: topic [/microROS/ping] does not appear to be published yet
Could not determine the type for the passed topic
```

#### Additional information
1) I assume that the serial cables are correctly conncted, because:
    a) the agent responses to the request from the Olimex client
    b) I exchanged Tx and Rx cable on the board, then there was no output of the Agent when starting uros_pingpong on the NSH shell
2) The client "failed to create node session on Micro ROS Agent" is the error message on the nsh shell. 
 Is there any log file of the Agent?

3) I googled for: micro-ros agent "failed to create node session on Micro ROS Agent"
    but did not find any solution there.

pablogs9, 2021-01-27:
Hello @JanStaschulat I have just replicated the tutorial with your very same hardware setup and I have it working:

[image]

Could you try to replicate it and follow the steps carefully?

pablogs9, 2021-01-27:
@jamoralp could you tell us something about this fast library?

JanStaschulat, 2021-01-27:
```C
jst3si@RNGX5596:~/olimex2_ws$ source ~/ros2_foxy/install/setup.bash 
jst3si@RNGX5596:~/olimex2_ws$ git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
Klone nach 'src/micro_ros_setup' ...
remote: Enumerating objects: 113, done.
remote: Counting objects: 100% (113/113), done.
remote: Compressing objects: 100% (93/93), done.
remote: Total 2286 (delta 46), reused 70 (delta 15), pack-reused 2173
Empfange Objekte: 100% (2286/2286), 490.26 KiB | 1.59 MiB/s, Fertig.
Löse Unterschiede auf: 100% (1530/1530), Fertig.
jst3si@RNGX5596:~/olimex2_ws$ 
jst3si@RNGX5596:~/olimex2_ws$ sudo apt update && rosdep update
OK:1 http://mirror-osd.de.bosch.com/ubuntu bionic InRelease
OK:2 http://mirror-osd.de.bosch.com/ubuntu bionic-security InRelease                          
OK:3 http://mirror-osd.de.bosch.com/ubuntu bionic-updates InRelease                           
Ign:4 http://mirror-osd.de.bosch.com/osd bionic InRelease                                     
OK:5 http://mirror-osd.de.bosch.com/partner bionic InRelease                                  
OK:6 http://mirror-osd.de.bosch.com/ansible bionic InRelease                                  
OK:7 http://mirror-osd.de.bosch.com/cepces bionic InRelease                                   
OK:8 http://mirror-osd.de.bosch.com/openconnect bionic InRelease                              
OK:9 http://mirror-osd.de.bosch.com/ms-teams stable InRelease                                 
OK:10 http://mirror-osd.de.bosch.com/osd bionic Release                                       
OK:11 http://ppa.launchpad.net/lttng/ppa/ubuntu bionic InRelease                              
OK:12 http://packages.ros.org/ros2/ubuntu bionic InRelease                                    
Holen:14 https://repo.iovisor.org/apt/bionic bionic InRelease [2.419 B]                       
Fehl:14 https://repo.iovisor.org/apt/bionic bionic InRelease
  Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY 91B31D79C30E5EE6
Paketlisten werden gelesen... Fertig
W: GPG-Fehler: https://repo.iovisor.org/apt/bionic bionic InRelease: Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY 91B31D79C30E5EE6
E: Das Depot »https://repo.iovisor.org/apt/bionic bionic InRelease« ist nicht signiert.
N: Eine Aktualisierung von solch einem Depot kann nicht auf eine sichere Art durchgeführt werden, daher ist es standardmäßig deaktiviert.
N: Weitere Details zur Erzeugung von Paketdepots sowie zu deren Benutzerkonfiguration finden Sie in der Handbuchseite apt-secure(8).
jst3si@RNGX5596:~/olimex2_ws$ rosdep install --from-path src --ignore-src -y
#All required rosdeps installed successfully
jst3si@RNGX5596:~/olimex2_ws$ sudo apt-get install python3-pip
[sudo] Passwort für jst3si: 
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
python3-pip ist schon die neueste Version (9.0.1-2.3~ubuntu1.18.04.4).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 39 nicht aktualisiert.
jst3si@RNGX5596:~/olimex2_ws$ colcon build
Starting >>> micro_ros_setup
Finished <<< micro_ros_setup [4.72s]                  

Summary: 1 package finished [4.90s]
jst3si@RNGX5596:~/olimex2_ws$ source install/local_setup.bash 
jst3si@RNGX5596:~/olimex2_ws$ ros2 run micro_ros_setup create_firmware_ws.sh nuttx olimex-stm32-e407
Creating firmware for nuttx platform olimex-stm32-e407
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Add distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"
Skip end-of-life distro "groovy"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Add distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
[... 372 lines cut ...]
Cloning into '/root/microros_ws/build/microxrcedds_agent/fastdds/src/fastdds/thirdparty/android-ifaddrs'...
Cloning into '/root/microros_ws/build/microxrcedds_agent/fastdds/src/fastdds/thirdparty/asio'...
Cloning into '/root/microros_ws/build/microxrcedds_agent/fastdds/src/fastdds/thirdparty/fastcdr'...
Cloning into '/root/microros_ws/build/microxrcedds_agent/fastdds/src/fastdds/thirdparty/tinyxml2'...
---
Finished <<< microxrcedds_agent [6min 0s]
Starting >>> micro_ros_agent
--- stderr: micro_ros_agent                                   
/usr/bin/ld: warning: libfastrtps.so.2, needed by /opt/ros/foxy/lib/librmw_fastrtps_shared_cpp.so, may conflict with libfastrtps.so.2.0.2
---
Finished <<< micro_ros_agent [8.08s]

Summary: 5 packages finished [6min 8s]
  2 packages had stderr output: micro_ros_agent microxrcedds_agent
```
I had no access to connect to serial port. So I added root to dialout group. 
However, this needed a logout. With exit, the docker container was gone: 

```C
jst3si@RNGX5596:~$ docker ps
CONTAINER ID        IMAGE               COMMAND             CREATED             STATUS              PORTS               NAMES
jst3si@RNGX5596:~$ 
```

So I have to do everything again. How can I logout/login, so that the group information is updated in a docker container?

pablogs9, 2021-01-27:
Are you using `--privileged` and `-v /dev:/dev` in the `docker run` command?

JanStaschulat, 2021-01-27:
yes:
docker run -it --net=host -v /dev:/dev -v /home/jst3si/ws_micro_ros:/home/root/microros_host_ws --privileged ros:foxy

pablogs9, 2021-01-27:
And the agent output when you try to use it?

JanStaschulat, 2021-01-27:
Compiling everything again in new docker container:
Error message for buillihg the agent:
```C
Starting >>> micro_ros_agent
--- stderr: micro_ros_agent                                    
/usr/bin/ld: warning: libfastrtps.so.2, needed by /opt/ros/foxy/lib/librmw_fastrtps_shared_cpp.so, may conflict with libfastrtps.so.2.0.2
---
Finished <<< micro_ros_agent [7.16s]

Summary: 5 packages finished [5min 22s]
  2 packages had stderr output: micro_ros_agent microxrcedds_agent
root@RNGX5596:/home/root/microros_ws# 
```

output of the agent:
```
root@RNGX5596:/home/root/microros_ws# ros2 run micro_ros_agent micro_ros_agent serial --dev usb-FTDI_TTL232R-3V3_FTAPR6AT-if00-port0
[1611745064.530750] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: usb-FTDI_TTL232R-3V3_FTAPR6AT-if00-port0, errno: 2
Error while starting serial agent!
root@RNGX5596:/home/root/microros_ws# 
```

pablogs9, 2021-01-27:
try `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-FTDI_TTL232R-3V3_FTAPR6AT-if00-port0`

JanStaschulat, 2021-01-27:
```
root@RNGX5596:/home/root/microros_ws# groups root
root : root dialout
root@RNGX5596:/home/root/microros_ws# 
```
however, i only added root to the dialout group, but did not a reboot or logout/login. Because I don't know how to do that within docker. Any idea?

JanStaschulat, 2021-01-27:
that worked:
```
root@RNGX5596:/home/root/microros_ws# ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-FTDI_TTL232R-3V3_FTAPR6AT-if00-port0
[1611745263.760440] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3

```

pablogs9, 2021-01-27:
and use `-v6` if you want to have a verbose output to check if everything is working fine

JanStaschulat, 2021-01-27:
but no connection from client to host:
nsh window:
```C
NuttShell (NSH)
nsh>
nsh>uros_pingpong
Serial mode => dev: /dev/ttyS0
[ERROR] [0946690326.899000000] [rclc]: [rclc_init] Error in rcl_init: failed to create node session on Micro ROS Agent., at /home/root/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c:317, at /home/root/microros_ws/firmware/mcu_ws/uros/rcl/rcl/src/rcl/init.c:233

Failed status on line 86: 1. Aborting.
nsh>
```

Agent window:
```C
root@RNGX5596:/home/root/microros_ws# ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-FTDI_TTL232R-3V3_FTAPR6AT-if00-port0
[1611745263.760440] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1611745379.431320] info     | Root.cpp           | create_client            | create                 | client_key: 0x169225AE, session_id: 0x81
[1611745379.431435] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x378676654, address: 1
[1611745379.447106] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x378676654, address: 1
[1611745379.494958] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x378676654, address: 1
[1611745379.607292] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x378676654, address: 1
[1611745379.798865] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x378676654, address: 1
[1611745380.198369] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x378676654, address: 1
[1611745380.997535] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x378676654, address: 1
[1611745382.612169] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x378676654, address: 1
[1611745385.808730] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x378676654, address: 1
[1611745392.202438] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x378676654, address: 1

```

JanStaschulat, 2021-01-27:
This is the same error as with my native Linux (Ubuntu 18:04) installation. Maybe with some configuration of the board? I tested it with two olimex boards. Both failed. So maybe some missing NuttX configuration about the serial connection?

pablogs9, 2021-01-27:
copy here the output of the agent with the `-v6` flag and upload a photo of your serial connection with the board.

JanStaschulat, 2021-01-27:
With -v I get this output:
```
root@RNGX5596:/home/root/microros_ws# ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-FTDI_TTL232R-3V3_FTAPR6AT-if00-port0 -v6
[1611745937.574139] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1611745937.574319] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1611745950.512150] info     | Root.cpp           | create_client            | create                 | client_key: 0x1EACD79D, session_id: 0x81
[1611745950.512268] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x514643869, address: 1
[1611745950.512489] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x1EACD79D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611745950.543967] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x1EACD79D, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 1E AC D7 9D 81 00 FC 01
[1611745950.544166] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x514643869, address: 1
[1611745950.544509] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x1EACD79D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611745950.591966] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x1EACD79D, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 1E AC D7 9D 81 00 FC 01
[1611745950.592107] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x514643869, address: 1
[1611745950.592240] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x1EACD79D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611745950.704037] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x1EACD79D, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 1E AC D7 9D 81 00 FC 01
[1611745950.704190] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x514643869, address: 1
[1611745950.704373] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x1EACD79D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611745950.895683] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x1EACD79D, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 1E AC D7 9D 81 00 FC 01
[1611745950.895811] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x514643869, address: 1
[1611745950.895930] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x1EACD79D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611745951.295455] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x1EACD79D, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 1E AC D7 9D 81 00 FC 01
[1611745951.295591] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x514643869, address: 1
[1611745951.295726] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x1EACD79D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611745952.094484] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x1EACD79D, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 1E AC D7 9D 81 00 FC 01
[1611745952.094664] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x514643869, address: 1
[1611745952.094883] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x1EACD79D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611745953.811231] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x1EACD79D, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 1E AC D7 9D 81 00 FC 01
[1611745953.811277] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x514643869, address: 1
[1611745953.811312] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x1EACD79D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611745956.905508] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x1EACD79D, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 1E AC D7 9D 81 00 FC 01
[1611745956.905723] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x514643869, address: 1
[1611745956.905923] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x1EACD79D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[... cut ...] ]        | client_key: 0x1EACD79D, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 1E AC D7 9D 81 00 FC 01
[1611745963.298770] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x514643869, address: 1
[1611745963.298891] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x1EACD79D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
```

picture is coming

JanStaschulat, 2021-01-27:
[image]
[image]

pablogs9, 2021-01-27:
As far as I see in the agent log, the agent is receiving the client messages but the client is not receiving the agent messages:

```
client ---_-> agent
client <--/-- agent
```

so the client keeps trying to establish a session since it does not receive the session created confirmation. Could you check your serial cable? Please try to bypass Tx and Rx and check if you have echo using `miniterm` or `minicom` .

JanStaschulat, 2021-01-27:
USB           serial           Port on Olimex
cable          cable
-------------------------------------------------------
black    ----   black     -----  Gnd
orange ----  grey        ----- RX3 
yellow  ----   white     -----  TX3

pablogs9, 2021-01-27:
Here you have compressed my .hex and .bin. With you very same configuration: Nuttx + Olimex E407, for me it works. Maybe if you flash this binary we can check if it is a configuration issue, communication issue...
[uros_nuttx_olimex.zip](https://github.com/micro-ROS/micro_ros_setup/files/5879810/uros_nuttx_olimex.zip)

JanStaschulat, 2021-01-27:
so the client keeps trying to establish a session since it does not receive the session created confirmation. Could you check your serial cable? Please try to bypass Tx and Rx  .

What do you mean "try to bypass Tx and Rx and check if you have echo using miniterm or minicom"?

pablogs9, 2021-01-27:
Just connect your Tx and Rx in the serial cable (make electrical contact between the white and the gray) and check if when you send a character using `minicom -D [device] -b 115200` the character appears in the terminal.

If `local echo` is disabled programs such as `minicom` just print what enters by the Rx and sends what you type. This way you can check your serial cable is working correctly.

JanStaschulat, 2021-01-27:
with your nuttx image:
```C
root@RNGX5596:/home/root/microros_ws# ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-FTDI_TTL232R-3V3_FTAPR6AT-if00-port0 -v6
[1611747568.031487] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1611747568.031675] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1611747575.521648] info     | Root.cpp           | create_client            | create                 | client_key: 0x5BA481F1, session_id: 0x81
[1611747575.521783] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1537507825, address: 1
[1611747575.521981] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5BA481F1, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611747575.553397] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5BA481F1, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 5B A4 81 F1 81 00 FC 01
[1611747575.553568] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1537507825, address: 1
[1611747575.553970] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5BA481F1, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611747575.601334] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5BA481F1, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 5B A4 81 F1 81 00 FC 01
[1611747575.601478] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1537507825, address: 1
[1611747575.601636] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5BA481F1, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611747575.697445] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5BA481F1, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 5B A4 81 F1 81 00 FC 01
[1611747575.697592] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1537507825, address: 1
[1611747575.697748] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5BA481F1, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611747575.905023] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5BA481F1, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 5B A4 81 F1 81 00 FC 01
[1611747575.905177] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1537507825, address: 1
[1611747575.905354] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5BA481F1, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611747576.304711] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5BA481F1, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 5B A4 81 F1 81 00 FC 01
[1611747576.304861] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1537507825, address: 1
[1611747576.305010] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5BA481F1, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611747577.104025] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5BA481F1, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 5B A4 81 F1 81 00 FC 01
[1611747577.104190] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1537507825, address: 1
[1611747577.104397] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5BA481F1, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611747578.702215] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5BA481F1, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 5B A4 81 F1 81 00 FC 01
[1611747578.702378] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1537507825, address: 1
[1611747578.702653] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5BA481F1, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611747582.015723] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5BA481F1, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 5B A4 81 F1 81 00 FC 01
[1611747582.015861] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1537507825, address: 1
[1611747582.016241] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5BA481F1, len: 19, data: 
[... cut ...]
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
```

nuttx shell:
```C
NuttShell (NSH)
nsh>uros_pingpong
Serial mode => dev: /dev/ttyS0
[ERROR] [0946692527.301000000] [rclc]: [rclc_init] Error in rcl_init: failed to create node session on Micro ROS Agent., at /workspaces/foxy_ws/uros_nuttx_olimex/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c:317, at /workspaces/foxy_ws/uros_nuttx_olimex/firmware/mcu_ws/uros/rcl/rcl/src/rcl/init.c:233

Failed status on line 86: 1. Aborting.
nsh>
```

JanStaschulat, 2021-01-27:
[image]
I connected Tx and Rx cable, ground was not connected (see picture). Is that correct?

Send a character from the shell with:
`root@RNGX5596:/home/root/microros_ws# echo 'A' >/dev/serial/by-id/usb-FTDI_TTL232R-3V3_FTAPR6AT-if00-port0`

opended minicom with:
`root@RNGX5596:/# minicom -D /dev/serial/by-id/usb-FTDI_TTL232R-3V3_FTAPR6AT-if00-port0 -b 115200`

but did not see anything.

JanStaschulat, 2021-01-27:
I am in home-office, here I don't have any other serial cable with connectors. I'll go to office now and search for some other cables.

pablogs9, 2021-01-27:
Don't do the echo, just ` minicom -D /dev/serial/by-id/usb-FTDI_TTL232R-3V3_FTAPR6AT-if00-port0 -b 115200` and write something in the shell inside the minicom. Does the character appears in the screen?

The GND unconnected is ok.

JanStaschulat, 2021-01-27:
@pablogs9 , I replaced both cables for RX and TX. 
Now the output is:
Agent
```C
root@RNGX5596:/home/root/microros_ws# ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-FTDI_TTL232R-3V3_FTAMCSN5-if00-port0  -v6
[1611760658.569773] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1611760658.570061] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1611760689.891115] info     | Root.cpp           | create_client            | create                 | client_key: 0x5218AEFC, session_id: 0x81
[1611760689.891228] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1377349372, address: 1
[1611760689.891484] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5218AEFC, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1611760690.022808] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1611760690.023242] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1611760690.057633] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1611760690.057830] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1611760690.106799] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1611760690.107035] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1611760690.237929] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1611760690.238248] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1611760690.272480] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1611760690.272676] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1611760690.320458] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1611760690.320716] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1611760690.570724] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1611760690.570930] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1611760690.717610] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1611760690.717887] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1611760690.800549] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1611760690.800845] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1611760691.050737] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1611760691.051012] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1611760691.182164] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1611760691.182442] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1611760691.264628] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5218AEFC, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[... cut ...]
0000: 81 00 00 00 0A 01 05 00 01 00 00 01 80
```
Client:
```C
NuttShell (NSH)
nsh>
nsh>uros_pingpong
Serial mode => dev: /dev/ttyS0
[ERROR] [0946684860.790000000] [rclc]: [rclc_publisher_init_default] Error in rcl_publisher_init: Issues running micro XRCE-DDS session, at /workspaces/foxy_ws/uros_nuttx_olimex/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/utils.c:38, at /workspaces/foxy_ws/uros_nuttx_olimex/firmware/mcu_ws/uros/rcl/rcl/src/rcl/publisher.c:187

Failed status on line 93: 1. Aborting.
nsh>
```

JanStaschulat, 2021-01-27:
I also noticed the following:
after flashing: 
- then the red and green light are on. 
- The micro-usb is listed in /dev after I unplug/plug-in the USB cable to my Laptop

when I power down and power up again:
- only the red light goes on, green light is out. 
- after unplug/plug-in the USB cable of the micro-usb cable,the micro-usb cable is not lised in /dev and 
  an not able to connect to Olimex board with screen/minicom.

I can only connect with the board immediately after flashing. Is this the same for you?

pablogs9, 2021-01-27:
Yes, it seems that you have to plug the USB after flashing if you want it to be recognized by the PC. It should be a Nuttx issue

Regarding the LEDs, I'm not sure, tomorrow I'll test it.

Regarding the connection, just changing the cables you achieve this XRCE communication? In that traces still there are a lot of ACK/NACK that can be seen as a nonstable communication. Do you have another USB-to-Serial cable to test?

JanStaschulat, 2021-01-27:
Yes, I have another USB-serial cable and three other serial cable. 
Now it works :)
[image]
[image]

pablogs9, 2021-01-28:
I'm closing Jan, reopen if you have any other problem.

closed 2021-01-28 by pablogs9