micro-ROS/micro_ros_setup #270: Generating static library for STM32f407 discovery board

Altaibaatar, 2021-02-03:
Hello ROS team,

Recently, I am trying to generate a micro-ROS static library(.a, .h) for my custom board. I am new to ROS & ROS2. So before that, I would like to generate it for the "olimex STM32-E407 discovery board" which already you guys provided some running examples. 
I am sorry for the amateur questions. (It seems like a duplicating with issue #137. I did try to mimic it. But couldn't find any success)

## Issue
Can not generate a static library.

- Hardware description: [STM32-E407](https://www.olimex.com/Products/ARM/ST/STM32-E407/open-source-hardware)
- RTOS: FreeRTOS
- Installation type: Generating a static library -> Import the generated library to the STM32cubeIDE
- Version or commit hash: Foxy
- Compiler: arm-none-eabi-

#### Steps

```
ros2 run micro_ros_setup create_firmware_ws.sh generate_lib

cp ./serial_transport_custom.c firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/serial/serial_transport_external.c

cp ./serial_transport_custom.h firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/profile/transport/serial/serial_transport_external.h 

ros2 run micro_ros_setup build_firmware.sh /home/ksj/uros_projects/install/micro_ros_setup/config/generate_lib/generic/toolchain-stm.cmake /home/ksj/uros_projects/install/micro_ros_setup/config/generate_lib/generic/client-colcon.meta
```

#### Expected behavior
.a and .h files to be generated in the build directory

#### Actual behavior

```
<-- Errors (I will post it separately) -->
Summary: 59 packages finished [42.7s]
  37 packages had stderr output: action_msgs actionlib_msgs builtin_interfaces composition_interfaces diagnostic_msgs example_interfaces geometry_msgs libyaml_vendor lifecycle_msgs micro_ros_msgs microxrcedds_client nav_msgs rcl rcl_action rcl_interfaces rcl_lifecycle rcl_logging_noop rclc rclc_lifecycle rcutils rmw rmw_implementation rmw_microxrcedds rosgraph_msgs rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_microxrcedds_c sensor_msgs shape_msgs statistics_msgs std_msgs std_srvs stereo_msgs test_msgs trajectory_msgs unique_identifier_msgs visualization_msgs
ls: cannot access '*.o': No such file or directory
``` 
#### Additional information
About the toolchain.cmake, I copied and edited it from [here (posted by pablogs9)](https://github.com/micro-ROS/micro_ros_setup/issues/137#issuecomment-648723389)

Altaibaatar, 2021-02-03:
My **toolchain.cmake**
```include(CMakeForceCompiler)

set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_SYSTEM_PROCESSOR arm)
set(CMAKE_CROSSCOMPILING 1)
set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)
set(PLATFORM_NAME "LwIP")

set(CMAKE_SYSROOT /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions)

set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)

# External transports
# set(EXTERNAL_TRANSPORT_HEADER "@EXTERNAL_TRANSPORT_HEADER@" CACHE PATH "" FORCE)
# set(EXTERNAL_TRANSPORT_SRC "@EXTERNAL_TRANSPORT@" CACHE PATH "" FORCE) 

# Makefile flags
set(CROSSDEV /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/../../toolchain/bin/arm-none-eabi-)
set(ARCH_CPU_FLAGS "-mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_HAL_DRIVER -DSTM32F407xx -D_TIMEVAL_DEFINED -O0 -Wall -fdata-sections -ffunction-sections")
set(ARCH_OPT_FLAGS "")

# Compiler tools
foreach(tool gcc ld ar)
	string(TOUPPER ${tool} TOOL)
    find_program(${TOOL} ${CROSSDEV}${tool})
	if(NOT ${TOOL})
		message(FATAL_ERROR "could not find ${CROSSDEV}${tool}")
	endif()
endforeach()

set(CMAKE_C_COMPILER ${CROSSDEV}gcc)
set(CMAKE_CXX_COMPILER ${CROSSDEV}g++)

set(CMAKE_C_FLAGS_INIT "-std=c11 ${ARCH_CPU_FLAGS} ${ARCH_OPT_FLAGS}" CACHE STRING "" FORCE)
set(CMAKE_CXX_FLAGS_INIT "-std=c++14 ${ARCH_CPU_FLAGS} ${ARCH_OPT_FLAGS} " CACHE STRING "" FORCE)

include_directories(SYSTEM 
	/home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/FreeRTOS-Plus-POSIX/include
 /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/include
 /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/include/private
 /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/include/FreeRTOS_POSIX
 /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/include/FreeRTOS_POSIX/sys
 /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/src/hal/interface
 /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/src/modules/interface
 /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/src/utils/interface
 /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/src/config
 /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/src/drivers/interface
 /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/Middlewares/Third_Party/LwIP/src/include/posix
 /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/Middlewares/Third_Party/LwIP/src/include
 /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/Inc
 /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/Drivers/STM32F4xx_HAL_Driver/Inc
 /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/Drivers/CMSIS/Device/ST/STM32F4xx/Include
 /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/Drivers/CMSIS/Include
 /home/user11/uros_examples/stm32discovery/firmware/freertos_apps/microros_olimex_e407_extensions/Middlewares/Third_Party/LwIP/system

    )
    
set(__BIG_ENDIAN__ 0)
```

Altaibaatar, 2021-02-03:
**Errors:**

```
Finished <<< rosidl_generator_dds_idl [0.46s]                             
Finished <<< tinydir_vendor [2.59s]
Starting >>> rcutils
--- stderr: rcutils                                                      
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_get_cwd':
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c:60:24: warning: unused parameter 'buffer' [-Wunused-parameter]
 rcutils_get_cwd(char * buffer, size_t max_length)
                 ~~~~~~~^~~~~~
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c:60:39: warning: unused parameter 'max_length' [-Wunused-parameter]
 rcutils_get_cwd(char * buffer, size_t max_length)
                                ~~~~~~~^~~~~~~~~~
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_is_directory':
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c:83:35: warning: unused parameter 'abs_path' [-Wunused-parameter]
 rcutils_is_directory(const char * abs_path)
                      ~~~~~~~~~~~~~^~~~~~~~
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_is_file':
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c:102:30: warning: unused parameter 'abs_path' [-Wunused-parameter]
 rcutils_is_file(const char * abs_path)
                 ~~~~~~~~~~~~~^~~~~~~~
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_exists':
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c:121:29: warning: unused parameter 'abs_path' [-Wunused-parameter]
 rcutils_exists(const char * abs_path)
                ~~~~~~~~~~~~~^~~~~~~~
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_is_readable':
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c:136:34: warning: unused parameter 'abs_path' [-Wunused-parameter]
 rcutils_is_readable(const char * abs_path)
                     ~~~~~~~~~~~~~^~~~~~~~
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_is_writable':
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c:158:34: warning: unused parameter 'abs_path' [-Wunused-parameter]
 rcutils_is_writable(const char * abs_path)
                     ~~~~~~~~~~~~~^~~~~~~~
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_is_readable_and_writable':
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c:180:47: warning: unused parameter 'abs_path' [-Wunused-parameter]
 rcutils_is_readable_and_writable(const char * abs_path)
                                  ~~~~~~~~~~~~~^~~~~~~~
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_mkdir':
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c:257:13: warning: implicit declaration of function 'mkdir' [-Wimplicit-function-declaration]
   int ret = mkdir(abs_path, 0775);
             ^~~~~
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_calculate_directory_size':
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c:267:47: warning: unused parameter 'directory_path' [-Wunused-parameter]
 rcutils_calculate_directory_size(const char * directory_path, rcutils_allocator_t allocator)
                                  ~~~~~~~~~~~~~^~~~~~~~~~~~~~
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c:267:83: warning: unused parameter 'allocator' [-Wunused-parameter]
 rcutils_calculate_directory_size(const char * directory_path, rcutils_allocator_t allocator)
                                                               ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_get_file_size':
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/filesystem.c:325:36: warning: unused parameter 'file_path' [-Wunused-parameter]
 rcutils_get_file_size(const char * file_path)
                       ~~~~~~~~~~~~~^~~~~~~~~
In function 'rcutils_char_array_strncat',
    inlined from 'rcutils_char_array_strcat' at /home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/char_array.c:239:10:
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/char_array.c:224:3: warning: 'strncat' output truncated before terminating nul copying as many bytes from a string as its length [-Wstringop-truncation]
   strncat(char_array->buffer, src, n);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/char_array.c: In function 'rcutils_char_array_strcat':
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/char_array.c:239:10: note: length computed here
   return rcutils_char_array_strncat(char_array, src, strlen(src));
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/process.c: In function 'rcutils_get_pid':
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/process.c:51:15: warning: implicit declaration of function 'getpid'; did you mean 'getw'? [-Wimplicit-function-declaration]
   return (int)getpid();
               ^~~~~~
               getw
/home/ksj/uros_projects/firmware/mcu_ws/uros/rcutils/src/logging.c: In function 'rcutils_get_env_var_zero_or_one':
[... cut ...]
Finished <<< rcl_logging_noop [0.85s]
--- stderr: rosidl_runtime_c                                                                           
CMake Warning at /home/ksj/uros_projects/firmware/mcu_ws/install/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rcutils' exports library 'dl' which couldn't be found
Call Stack (most recent call first):
  /home/ksj/uros_projects/firmware/mcu_ws/install/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  CMakeLists.txt:15 (find_package)

```

pablogs9, 2021-02-03:
Hello, 

```bash
<-- Errors (I will post it separately) -->
Summary: 59 packages finished [42.7s]
  37 packages had stderr output: action_msgs actionlib_msgs builtin_interfaces composition_interfaces diagnostic_msgs example_interfaces geometry_msgs libyaml_vendor lifecycle_msgs micro_ros_msgs microxrcedds_client nav_msgs rcl rcl_action rcl_interfaces rcl_lifecycle rcl_logging_noop rclc rclc_lifecycle rcutils rmw rmw_implementation rmw_microxrcedds rosgraph_msgs rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_microxrcedds_c sensor_msgs shape_msgs statistics_msgs std_msgs std_srvs stereo_msgs test_msgs trajectory_msgs unique_identifier_msgs visualization_msgs
ls: cannot access '*.o': No such file or directory
```

This is not telling you that the build fails, just that some packages (a lot of them in fact) have warnings (stderr), but usually, this is kind of normal in cross-compiling.

Have you look in `firmware/build/libmicroros.a`?

Altaibaatar, 2021-02-03:
I have it. I imported it to my STM32CubeIDE and compiled the project successfully with .a and .h files. 
So I tried to run ping-pong app. But it seems **rclc_support_init()** function is not returning. So I thought the problem maybe related to generating the library.
```
	rcl_allocator_t allocator = rcl_get_default_allocator();
	rclc_support_t support;

	// create init_options
	RCCHECK(rclc_support_init(&support, 0, NULL, &allocator)); <-- Stopped HERE

	// create node
	rcl_node_t node;
	RCCHECK(rclc_node_init_default(&node, "pingpong_node", "", &support));
```

pablogs9, 2021-02-03:
Two things:
- Are you building the project inside `firmware/freertos_apps/microros_olimex_e407_extensions` or just using these headers in your `toolchain.cmake`?
- Can I see your transports?

Altaibaatar, 2021-02-03:
- Just using the headers. After generating a static library, I am importing .a, .h files to stm32cubeide on my WIndows computer.
- Do you mean [serial_t_ransport_custom.*] as transports?

pablogs9, 2021-02-03:
Yes, I mean the serial transports.

Be careful because any version mismatch of the header that we have in `freertos_apps/microros_olimex_e407_extensions` and your own headers can result in an ABI incompatibility that will crash the app. You should be building your library with your own headers.

Altaibaatar, 2021-02-03:
**serial_transport.h**
```
#ifndef _UXR_CLIENT_SERIAL_TRANSPORT_OLIMEXSTM32E407_H_
#define _UXR_CLIENT_SERIAL_TRANSPORT_OLIMEXSTM32E407_H_

#ifdef __cplusplus
extern "C"
{
#endif

#include "stm32f4xx_hal.h"

extern UART_HandleTypeDef huart1;
extern UART_HandleTypeDef huart3;
extern UART_HandleTypeDef huart6;
extern DMA_HandleTypeDef hdma_usart3_rx;

typedef struct uxrSerialPlatform
{
    UART_HandleTypeDef * uart;
} uxrSerialPlatform;

#ifdef __cplusplus
}
#endif

#endif //_UXR_CLIENT_SERIAL_TRANSPORT_OLIMEXSTM32E407_H_
```

Altaibaatar, 2021-02-03:
**Serial_transport.c**

```
#include <uxr/client/profile/transport/serial/serial_transport_external.h>
#include "stm32f4xx_hal_dma.h"

#include <unistd.h>
#include <stdio.h>
#include <string.h>
#include <stdbool.h>

#define UART_DMA_BUFFER_SIZE 2048

static uint8_t dma_buffer[UART_DMA_BUFFER_SIZE];
static size_t dma_head = 0, dma_tail = 0;

bool uxr_init_serial_platform(struct uxrSerialPlatform* platform, int fd, uint8_t remote_addr, uint8_t local_addr)
{
  switch ( fd ){
      case 1:
        platform->uart = &huart1;
        break;
      case 3:
        platform->uart = &huart3;
        break;
      case 6:
        platform->uart = &huart6;
        break;
      default:
        return false;
  }
  HAL_UART_Receive_DMA(platform->uart, dma_buffer, UART_DMA_BUFFER_SIZE);
  return true;
}

bool uxr_close_serial_platform(struct uxrSerialPlatform* platform)
{   
  HAL_UART_DMAStop(platform->uart);
  return true;
}

size_t uxr_write_serial_data_platform(uxrSerialPlatform* platform, uint8_t* buf, size_t len, uint8_t* errcode)
{
  
  HAL_StatusTypeDef ret;

  if (platform->uart->gState == HAL_UART_STATE_READY){
    ret = HAL_UART_Transmit_DMA(platform->uart, buf, len);
    while (ret == HAL_OK && platform->uart->gState != HAL_UART_STATE_READY){
      osDelay(1);
    }

    return (ret == HAL_OK) ? len : 0;
  }else{
    return 0;
  }
}

size_t uxr_read_serial_data_platform(uxrSerialPlatform* platform, uint8_t* buf, size_t len, int timeout, uint8_t* errcode)
{ 
  __disable_irq();
  dma_tail = UART_DMA_BUFFER_SIZE - __HAL_DMA_GET_COUNTER(platform->uart->hdmarx);
  __enable_irq();

  size_t wrote = 0;
  while ((dma_head != dma_tail) && (wrote < len)){
    buf[wrote] = dma_buffer[dma_head];
    dma_head = (dma_head + 1) % UART_DMA_BUFFER_SIZE;
    wrote++;
  }
 
  return wrote;
}
```

Altaibaatar, 2021-02-19:
Hello @pablogs9 , 

Thank you for your support. 

I've imported the libmicroros.a and include files to my IDE. The project was compiled and flashed the stm32 controller. But when I tried to publish ros message, the app crashes. I did step by step debugging. But since I am using the static library, I am not able to access (.c) files. 
If it is possible, could you please provide or publish (.c) and (.h) files that used for STM32 discovery board?

pablogs9, 2021-02-19:
in which function is it crashing?

Altaibaatar, 2021-02-19:
It's like

```
appMain() at app.c
↓
rclc_support_init()
↓
rcl_get_zero_initialized_init_options()
↓
rcl_support_init()
↓
rcl_init_options_init()
↓
rcutils_allocator_is_valid()
↓
rcl_init_options_init()
↓
__freertos_allocate(size_t size, void * state)
↓
pvPortMalloc( size_t xWantedSize )
↓
vTaskSuspendAll();

```

pablogs9, 2021-02-19:
Which allocation scheme are you using in your FreeRTOS system?

Check [this](https://github.com/micro-ROS/freertos_apps/blob/foxy/microros_olimex_e407_extensions/Src/custom_memory_manager.c), [this](https://github.com/micro-ROS/freertos_apps/blob/foxy/microros_olimex_e407_extensions/Src/allocators.c), [this](https://github.com/micro-ROS/freertos_apps/blob/74cd38c0a283e66741881d62b5f0967640927179/microros_olimex_e407_extensions/Src/main.c#L457), [this](https://github.com/micro-ROS/freertos_apps/blob/74cd38c0a283e66741881d62b5f0967640927179/microros_olimex_e407_extensions/Makefile#L161) and [this](https://github.com/micro-ROS/freertos_apps/blob/74cd38c0a283e66741881d62b5f0967640927179/microros_olimex_e407_extensions/Makefile#L166).

You need to update your memory management to something similar to our approach.

Altaibaatar, 2021-02-25:
Hello @pablogs9 

The allocation problem was fixed. Thank you for your direction and help. 

After that I tried to establish ROS connection and the session established successfully with udp4. 

But when I tried to initialize a publisher with the command below, 
```
RCCHECK(rclc_publisher_init_default(&ping_publisher, &node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Header), "/microROS/ping"));
```

it gives **the error:**
```
app.c:95: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Header'
```

-----
Inside the "std_msgs", there is header.h file. 
```
#ifndef STD_MSGS__MSG__HEADER_H_
#define STD_MSGS__MSG__HEADER_H_

#include "std_msgs/msg/detail/header__struct.h"
#include "std_msgs/msg/detail/header__functions.h"
#include "std_msgs/msg/detail/header__type_support.h"

#endif  // STD_MSGS__MSG__HEADER_H_
```

-----
In the "detail" folder, there are:
```
header__functions.c
header__functions.h
header__rosidl_typesupport_microxrcedds_c.h
header_struct.h
header__type_support.h
```

Could you please help me to figure out this problem?

Thank you for your support

pablogs9, 2021-02-25:
Have you included `#include <std_msgs/Header.h>`?

You can check if that symbol is included in your library using:

```bash
nm libmicroros.a | grep rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Header
```

Also, you can check if this symbol is even build if you do something like:

```bash
find firmware -name "*.obj" -exec nm -A {} \; | grep rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Header
```

Let me know what do you find.

pablogs9, 2021-03-12:
Hello! 

Check this: https://github.com/micro-ROS/micro_ros_stm32cubemx_utils

Altaibaatar, 2021-03-13:
Hello,

Thank you for your help and effort @pablogs9. Recently I am a bit separated from the project for awhile. I will try it check that soon. 

Thank you!

closed 2021-03-13 by Altaibaatar