micro-ROS/micro_ros_setup #271: Error building firmware

h-valdes, 2021-02-09:
Hi, 

I am following the [introductory tutorial](https://micro-ros.github.io//docs/tutorials/core/first_application_linux/) from the website. I tried it in a Docker container and in a fresh install of Ubuntu 20.04. I am failing
to execute:

- ros2 run micro_ros_setup build_firmware.sh

The error:

`In file included from /home/humboldt/micro2/src/uros/micro-ROS-demos/rclc/autodiscover_agent/main.c:11:
/home/humboldt/micro2/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/include/rmw_uros/options.h:22:10: fatal error: ucdr/microcdr.h: No such file or directory
   22 | #include <ucdr/microcdr.h>
`
And the final output of colcon:
`
Summary: 29 packages finished [6min 23s]
  1 package failed: micro_ros_demos_rclc
  4 packages aborted: micro_ros_demos_rclcpp nav_msgs sensor_msgs visualization_msgs
  2 packages had stderr output: micro_ros_demos_rclc micro_ros_demos_rclcpp
  2 packages not processed
`

### Additional information
Yesterday the repository was also failing, but the error was in rmw-microxrcedds. 
The error was the number of parameters in the function uxr_init_serial_transport. I removed &context_impl->udp_platform from rmw_microxrcedds_c/src/rmw_init.c and rmw_microxrcedds_c/src/rmw_uros_options.c and it worked (a similar fix was introduced in [commit](https://github.com/micro-ROS/rmw-microxrcedds/commit/a02183b1609f9ad34d33e48a3e3a30950ad2e8c3)).

h-valdes, 2021-02-09:
Following the error, the problem that I have is related with [this include in rmw-microxrcedss](https://github.com/micro-ROS/rmw-microxrcedds/blob/foxy/rmw_microxrcedds_c/include/rmw_uros/options.h#L22):

pablogs9, 2021-02-09:
Hello @h-valdes we are aware of this. Please wait a couple of hours because we are updating some repos and we will fix this. Sorry for the inconvenience.

h-valdes, 2021-02-09:
No Problem! Thanks for the quick answer!

pablogs9, 2021-02-09:
I will notify you here when this is solved.

pablogs9, 2021-02-09:
Hello, you should be able to try it now.

I'm closing, do not hesitate to reopen if you have any problem.

closed 2021-02-09 by pablogs9

h-valdes, 2021-02-09:
@pablogs9 Thanks a lot, it works! I did a test of the basic example and also with the esp32 firmware.
Both are working. 

Thank you!