micro-ROS/micro_ros_setup #276: Generate Static Library for STM32G474 series

masoudir, 2021-02-11:
Hello,

I want to generate static library for STM32G474 series. I know that this microcontroller is in CortexM4F family and this is not listed in [this links](https://github.com/micro-ROS/micro_ros_arduino/tree/foxy/extras/library_generation).

Maybe this is a new topic for this kind of ARM M4F family and there is no cmake or colcon file for that. Can any one help me?

By the way, I have tried to build for cortex-m3 using this [file](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/extras/library_generation/cortex_m3_toolchain.cmake) as cmake and this [file](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/extras/library_generation/colcon.meta) as colcon. But got this error:

##########

ros2 run micro_ros_setup build_firmware.sh '/home/a/uros_ws/tt/cortex-m3-toolchain.cmake' '/home/a/uros_ws/tt/colcon.meta' 
Crosscompiled environment: cleaning path
Building firmware for generate_lib platform generic
Using provided meta: /home/a/uros_ws/tt/colcon.meta
Starting >>> tinydir_vendor
Starting >>> rosidl_adapter
Starting >>> rosidl_typesupport_interface
Starting >>> microcdr
Finished <<< rosidl_typesupport_interface [1.11s]                   
Starting >>> rmw_implementation_cmake
Finished <<< rosidl_adapter [1.32s]                                 
Starting >>> rosidl_parser
--- stderr: microcdr
gcc: warning: ‘-mcpu=’ is deprecated; use ‘-mtune=’ or ‘-march=’ instead
gcc: error: unrecognized command line option ‘-mthumb’
make[5]: *** [CMakeFiles/microcdr.dir/build.make:66: CMakeFiles/microcdr.dir/src/c/common.c.obj] Error 1
make[5]: *** Waiting for unfinished jobs....
gcc: warning: ‘-mcpu=’ is deprecated; use ‘-mtune=’ or ‘-march=’ instead
gcc: error: unrecognized command line option ‘-mthumb’
make[5]: *** [CMakeFiles/microcdr.dir/build.make:79: CMakeFiles/microcdr.dir/src/c/types/basic.c.obj] Error 1
gcc: warning: ‘-mcpu=’ is deprecated; use ‘-mtune=’ or ‘-march=’ instead
gcc: error: unrecognized command line option ‘-mthumb’
make[5]: *** [CMakeFiles/microcdr.dir/build.make:92: CMakeFiles/microcdr.dir/src/c/types/string.c.obj] Error 1
make[4]: *** [CMakeFiles/Makefile2:79: CMakeFiles/microcdr.dir/all] Error 2
make[3]: *** [Makefile:133: all] Error 2
make[2]: *** [CMakeFiles/ucdr.dir/build.make:115: ucdr-prefix/src/ucdr-stamp/ucdr-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:79: CMakeFiles/ucdr.dir/all] Error 2
make: *** [Makefile:87: all] Error 2
---
Failed   <<< microcdr [1.31s, exited with code 2]
Aborted  <<< rosidl_parser [0.02s]
Aborted  <<< rmw_implementation_cmake [0.61s]                                  
Aborted  <<< tinydir_vendor [5.53s]                                
                                 
Summary: 2 packages finished [6.28s]
  1 package failed: microcdr
  3 packages aborted: rmw_implementation_cmake rosidl_parser tinydir_vendor
  1 package had stderr output: microcdr
  53 packages not processed

pablogs9, 2021-02-12:
Which compiler and version are you using?

Could you please provide a sample CMake/Makefile/GCC command of any other code that you are building for your STM32G474 platform?

masoudir, 2021-02-12:
I am using STM32CubeIDE as IDE and also arm-none-eabi-gcc as compiler.
But in case of that micro-ros uses cmake, may I have to use Makefile created by STM32CubeIDE? How can I use that makefile instead of that Cortex-M3 Cmake I have mentioned?

My sample Makefile is [this](https://drive.google.com/file/d/1N5n7XCz0KwrkYlzpEhd19KTVQGfzsTC0/view?usp=sharing)
This file is automatically generated by STM32CubeIDE as Makefile project.

How can I create cmake suitable for building micro-ros static lib for this kind of microcontroller?

Thanks very much

pablogs9, 2021-02-15:
Some points:
- As you can see the error that you are having is about the compiler flags, probably you have not set the correct compiler with lines:
```cmake
set(CMAKE_C_COMPILER [YOUR PATH])
set(CMAKE_CXX_COMPILER [YOUR PATH]) 
```
Use here the same path to the compiler that STM32Cube is using
- Using the `generate_lib` option of the micro-ROS setup you don't need to mix the micro-ROS build with your app build, just link a static library within your Makefile.
- Sorry but this Makefile that you share does not build for Cortex-M3 architecture, it is for `cortex-m4`.

<details>
  <summary>Click to expand</summary>
  
```makefile
##########################################################################################################################
# File automatically-generated by tool: [projectgenerator] version: [3.11.0-B13] date: [Fri Feb 12 10:37:00 PST 2021]
##########################################################################################################################

# ------------------------------------------------
# Generic Makefile (based on gcc)
#
# ChangeLog :
#	2017-02-10 - Several enhancements + project update mode
#   2015-07-22 - first version
# ------------------------------------------------

######################################
# target
######################################
TARGET = test

######################################
# building variables
######################################
# debug build?
DEBUG = 1
# optimization
OPT = -Og

#######################################
# paths
#######################################
# Build path
BUILD_DIR = build

######################################
# source
######################################
# C sources
C_SOURCES =  \
Core/Src/main.c \
Core/Src/stm32g4xx_it.c \
Core/Src/stm32g4xx_hal_msp.c \
Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_tim.c \
Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_tim_ex.c \
Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_pwr_ex.c \
Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal.c \
Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_rcc.c \
Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_rcc_ex.c \
Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_flash.c \
Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_flash_ex.c \
Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_flash_ramfunc.c \
Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_gpio.c \
Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_exti.c \
Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_dma.c \
Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_dma_ex.c \
Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_pwr.c \
Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_cortex.c \
Core/Src/system_stm32g4xx.c  

[... 109 lines cut ...]
$(BUILD_DIR)/%.bin: $(BUILD_DIR)/%.elf | $(BUILD_DIR)
	$(BIN) $< $@	
	
$(BUILD_DIR):
	mkdir $@		

#######################################
# clean up
#######################################
clean:
	-rm -fR $(BUILD_DIR)
  
#######################################
# dependencies
#######################################
-include $(wildcard $(BUILD_DIR)/*.d)

# *** EOF ***
```
</details>

- When you have the correct Makefile, just fill the the flag line in the toolchain with the very same compiler flag of your Makefile:
```cmake
set(FLAGS [YOUR FLAGS HERE] CACHE STRING "" FORCE)
```

masoudir, 2021-02-15:
I have edited cmake file as this:

####################################          
SET(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_CROSSCOMPILING 1)
set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)

set(CMAKE_C_COMPILER arm-none-eabi-gcc)
set(CMAKE_CXX_COMPILER arm-none-eabi-g++)

SET(CMAKE_C_COMPILER_WORKS 1 CACHE INTERNAL "")
SET(CMAKE_CXX_COMPILER_WORKS 1 CACHE INTERNAL "")

set(FLAGS "-mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DF_CPU=84000000L -D'RCUTILS_LOG_MIN_SEVERITY=RCUTILS_LOG_MIN_SEVERITY_NONE'" CACHE STRING "" FORCE)

set(CMAKE_C_FLAGS_INIT "-std=c11 ${FLAGS} -DCLOCK_MONOTONIC=0 -D'__attribute__(x)='" CACHE STRING "" FORCE)
set(CMAKE_CXX_FLAGS_INIT "-std=c++11 ${FLAGS} -fno-rtti -DCLOCK_MONOTONIC=0 -D'__attribute__(x)='" CACHE STRING "" FORCE)

set(__BIG_ENDIAN__ 0)
#######################################

but got multiple errors. I have copied all flags from my makefile as you have mentioned. Also I have tried with different flags. But again I have got errors.

Some of these errors were:

################
---
Finished <<< visualization_msgs [14.1s]
--- stderr: sensor_msgs                                       
CMake Warning at /home/a/uros_ws/firmware/mcu_ws/install/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rcutils' exports library 'dl' which couldn't be found
Call Stack (most recent call first):
  /home/a/uros_ws/firmware/mcu_ws/install/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  /home/a/uros_ws/firmware/mcu_ws/install/share/rosidl_runtime_c/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/a/uros_ws/firmware/mcu_ws/install/share/rosidl_runtime_c/cmake/rosidl_runtime_cConfig.cmake:41 (include)
  /home/a/uros_ws/firmware/mcu_ws/install/share/builtin_interfaces/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/a/uros_ws/firmware/mcu_ws/install/share/builtin_interfaces/cmake/builtin_interfacesConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

---
Finished <<< sensor_msgs [28.3s]
Starting >>> stereo_msgs
--- stderr: stereo_msgs                                      
CMake Warning at /home/a/uros_ws/firmware/mcu_ws/install/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rcutils' exports library 'dl' which couldn't be found
Call Stack (most recent call first):
  /home/a/uros_ws/firmware/mcu_ws/install/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  /home/a/uros_ws/firmware/mcu_ws/install/share/rmw/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/a/uros_ws/firmware/mcu_ws/install/share/rmw/cmake/rmwConfig.cmake:41 (include)
  /home/a/uros_ws/firmware/mcu_ws/install/share/rosidl_typesupport_microxrcedds_c/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/a/uros_ws/firmware/mcu_ws/install/share/rosidl_typesupport_microxrcedds_c/cmake/rosidl_typesupport_microxrcedds_cConfig.cmake:41 (include)
  /home/a/uros_ws/firmware/mcu_ws/install/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:13 (find_package)
  /home/a/uros_ws/firmware/mcu_ws/install/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /home/a/uros_ws/firmware/mcu_ws/install/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /home/a/uros_ws/firmware/mcu_ws/install/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

---
Finished <<< stereo_msgs [3.59s]
Starting >>> common_interfaces
Finished <<< common_interfaces [0.83s]                          

Summary: 59 packages finished [1min 35s]
  37 packages had stderr output: action_msgs actionlib_msgs builtin_interfaces composition_interfaces diagnostic_msgs example_interfaces geometry_msgs libyaml_vendor lifecycle_msgs micro_ros_msgs microxrcedds_client nav_msgs rcl rcl_action rcl_interfaces rcl_lifecycle rcl_logging_noop rclc rclc_lifecycle rcutils rmw rmw_implementation rmw_microxrcedds rosgraph_msgs rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_microxrcedds_c sensor_msgs shape_msgs statistics_msgs std_msgs std_srvs stereo_msgs test_msgs trajectory_msgs unique_identifier_msgs visualization_msgs
ls: cannot access '*.o': No such file or directory

################

Is there any valid cmake file for even a STM32 cortex-m3 microcontroller to test it? 

I do not know how to solve this issue? I think micro-ros does not provide any good tutorials on how to use generate_lib for custom microcontrollers!!!!

pablogs9, 2021-02-16:
Some points:
1. Please use [Mardown style](https://docs.github.com/es/github/writing-on-github/basic-writing-and-formatting-syntax) for formatting your issues.
2. You are building for Cortex-M3 with the flag `-mcpu=cortex-m4`
3. Your building **DOES NOT** have errors, just warnings. Read carefully. You should have `firmware/build/libmicroros.a` and  `firmware/include`, please check it.

> I do not know how to solve this issue? I think micro-ros does not provide any good tutorials on how to use generate_lib for custom microcontrollers!!!!

Regarding this, yes we have [a good tutorial](https://micro.ros.org/docs/tutorials/advanced/create_custom_static_library/) on how to use the custom library generation. But you need to know how to use CMake toolchains for cross-compilation, be familiar with ROS 2 build systems and read carefully.

masoudir, 2021-02-25:
Hello @pablogs9 ,

I have switched to platformio in order to build the code.

Also I have used platformio build flags for CMAKE in micro-ros.

# The cmake is this:
  `
    SET(CMAKE_SYSTEM_NAME Generic)
  set(CMAKE_CROSSCOMPILING 1)
  set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)
  set(CMAKE_C_COMPILER arm-none-eabi-gcc)
  set(CMAKE_CXX_COMPILER arm-none-eabi-g++)
  
  SET(CMAKE_C_COMPILER_WORKS 1 CACHE INTERNAL "")
  SET(CMAKE_CXX_COMPILER_WORKS 1 CACHE INTERNAL "")
  
  set(FLAGS "-Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m4 --specs=nano.specs --specs=nosys.specs -Wl,--start-group -lc -lgcc -lm -lstdc++ -lnosys" CACHE STRING "" FORCE)
  
  set(CMAKE_C_FLAGS_INIT "-std=c11 ${FLAGS} -DCLOCK_MONOTONIC=0 -D'__attribute__(x)='" CACHE STRING "" FORCE)
  set(CMAKE_CXX_FLAGS_INIT "-std=c++11 ${FLAGS} -fno-rtti -DCLOCK_MONOTONIC=0 -D'__attribute__(x)='" CACHE STRING "" FORCE)
  
  set(__BIG_ENDIAN__ 0)
  `

# The colcon file:

`
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        },
        "rosidl_typesupport": {
            "cmake-args": [
                "-DROSIDL_TYPESUPPORT_SINGLE_TYPESUPPORT=ON"
            ]
        },
        "rcl": {
            "cmake-args": [
                "-DBUILD_TESTING=OFF",
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
                "-DRCL_LOGGING_ENABLED=OFF"
            ]
        }, 
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_NO_THREAD_SUPPORT=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON"
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-UCLIENT_PROFILE_STREAM_FRAMING=ON",
                "-DUCLIENT_PROFILE_CUSTOM_TRANSPORT=ON"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=10",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=4",
                "-DRMW_UXRCE_TRANSPORT=custom"
            ]
        }
    }
}
`

I have created platformio project for stm32g474 with framework of STM32Cube.
Also I included this flag for introducing static lib to platformio `build_flags = -I include -L lib -l microros`

But got these errors when linking to micro-ros static library file:

`
lib/libmicroros.a(librclc-executor.c.obj): In function `rclc_executor_spin_one_period':
executor.c:(.text+0x152a): undefined reference to `usleep'
lib/libmicroros.a(librclc-executor.c.obj): In function `rclc_executor_spin_period':
executor.c:(.text+0x15b8): undefined reference to `usleep'
lib/libmicroros.a(librcutils-filesystem.c.obj): In function `rcutils_mkdir':
filesystem.c:(.text+0x9e): undefined reference to `mkdir'
lib/libmicroros.a(librcutils-time_unix.c.obj): In function `rcutils_system_time_now':
time_unix.c:(.text+0xc): undefined reference to `clock_gettime'
lib/libmicroros.a(librcutils-time_unix.c.obj): In function `rcutils_steady_time_now':
time_unix.c:(.text+0x5c): undefined reference to `clock_gettime'
lib/libmicroros.a(libmicroxrcedds_client-time.c.obj): In function `uxr_millis':
time.c:(.text.uxr_millis+0x8): undefined reference to `clock_gettime'
lib/libmicroros.a(libmicroxrcedds_client-time.c.obj): In function `uxr_nanos':
time.c:(.text.uxr_nanos+0x8): undefined reference to `clock_gettime'
`

The error is related to some functions of timers inside libmrcroros.a file.

How can I resolve it?

Thanks in advance

pablogs9, 2021-02-26:
The library link is complaining about some missing functions:
- `clock_gettime`: you have to implement it somewhere using functions of your hardware/platform, something like [this](https://github.com/micro-ROS/micro_ros_arduino/blob/ec8fc92f5e9d2cc8192f4ac632d451e14f46c33c/src/opencr_transports.c.in#L6)
- `usleep`: you have to implement it somewhere, same as above.
- `mkdir`: this function is not important, you can define it as an empty function, like [this](https://github.com/micro-ROS/micro_ros_arduino/blob/ec8fc92f5e9d2cc8192f4ac632d451e14f46c33c/src/arduino_transports.c#L13)

pablogs9, 2021-03-12:
Hello!

Please check this: https://github.com/micro-ROS/micro_ros_stm32cubemx_utils

closed 2021-03-12 by pablogs9