micro-ROS/micro_ros_setup #277: some thing wrong with rclc_support_init in appMain

Needrom, 2021-02-20:
Error output with appMain thread open.

with no idea of gdb in linux with openocd so I add some printf in code to detect where the code had ran.

printf output.
```
printf is ready 
endif----- 
freeRTOS allocator success 
Into appMain 
test printf 0 
test printf 1 
rclc_support_init counter : 0 
rclc_support_init counter : 1 
rclc_support_init counter : 2 
rclc_support_init counter : 3 
rclc_support_init counter : 4 
get to end rclc_support_init counter : 5 
rclc_support_init_with_options counter : 0 
rclc_support_init_with_options counter : 1 
rclc_support_init_with_options counter : 2 
rclc_support_init_with_options counter : 3 
rclc_support_init_with_options counter : 4 
rclc_support_init_with_options counter : 5 
rclc_support_init_with_options counter : 6 

```

It mean the code run into appMain and stop at this line 
```
RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));
```

and finally it stop it this line 
./mcu_ws/uros/rclc/rclc/src/rclc/init.c

```
rcl_ret_t
rclc_support_init_with_options(
  rclc_support_t * support,
  int argc,
  char const * const * argv,
  rcl_init_options_t * init_options,
  rcl_allocator_t * allocator)
{
 int counter = 0;
 printf("%s counter : %d \r\n", __func__, counter++);
 RCL_CHECK_FOR_NULL_WITH_MSG(
    support, "support is a null pointer", return RCL_RET_INVALID_ARGUMENT);
   printf("%s counter : %d \r\n", __func__, counter++);
 RCL_CHECK_FOR_NULL_WITH_MSG(
    init_options, "init_options is a null pointer", return RCL_RET_INVALID_ARGUMENT);
   printf("%s counter : %d \r\n", __func__, counter++);
 RCL_CHECK_FOR_NULL_WITH_MSG(
    allocator, "allocator is a null pointer", return RCL_RET_INVALID_ARGUMENT);
  rcl_ret_t rc = RCL_RET_OK;
  printf("%s counter : %d \r\n", __func__, counter++);

  memcpy(&support->init_options, init_options, sizeof(rcl_init_options_t));
  printf("%s counter : %d \r\n", __func__, counter++);

   printf("%s counter : %d \r\n", __func__, counter++);
 support->context = rcl_get_zero_initialized_context();
   printf("%s counter : %d \r\n", __func__, counter++);
 rc = rcl_init(argc, argv, &support->init_options, &support->context);              //stop here
   printf("%s counter : %d \r\n", __func__, counter++);
 if (rc != RCL_RET_OK) {
    PRINT_RCLC_ERROR(rclc_init, rcl_init);
    return rc;
  }
  support->allocator = allocator;
  printf("%s counter : %d \r\n", __func__, counter++);

  rc = rcl_clock_init(RCL_STEADY_TIME, &support->clock, support->allocator);
  if (rc != RCL_RET_OK) {
    PRINT_RCLC_ERROR(rclc_init, rcl_clock_init);
  }
   printf("%s counter : %d \r\n", __func__, counter++);
 return rc;
}

```

## the change I had make.
- I just add some printf.
1.  in Ping-Pong app.c

```
void appMain(void *argument)
{
	int test_printf_count = 0;
	printf("Into %s \r\n", __func__);
	rcl_allocator_t allocator = rcl_get_default_allocator();
	printf("test printf %d \r\n", test_printf_count++);
	rclc_support_t support;

	printf("test printf %d \r\n", test_printf_count++);
// create init_options
	RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

	printf("test printf %d \r\n", test_printf_count++);
// create node
	rcl_node_t node;
	RCCHECK(rclc_node_init_default(&node, "pingpong_node", "", &support));

	printf("test printf %d \r\n", test_printf_count++);
// Create a reliable ping publisher
	RCCHECK(rclc_publisher_init_default(&ping_publisher, &node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Header), "/microROS/ping"));

	printf("test printf %d \r\n", test_printf_count++);
// Create a best effort pong publisher
	RCCHECK(rclc_publisher_init_best_effort(&pong_publisher, &node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Header), "/microROS/pong"));

	printf("test printf %d \r\n", test_printf_count++);
// Create a best effort ping subscriber
	RCCHECK(rclc_subscription_init_best_effort(&ping_subscriber, &node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Header), "/microROS/ping"));

	printf("test printf %d \r\n", test_printf_count++);
// Create a best effort  pong subscriber
	RCCHECK(rclc_subscription_init_best_effort(&pong_subscriber, &node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Header), "/microROS/pong"));

	printf("test printf %d \r\n", test_printf_count++);
// Create a 3 seconds ping timer timer,
	rcl_timer_t timer;
	RCCHECK(rclc_timer_init_default(&timer, &support, RCL_MS_TO_NS(2000), ping_timer_callback));

	printf("test printf %d \r\n", test_printf_count++);
// Create executor
	rclc_executor_t executor;
	RCCHECK(rclc_executor_init(&executor, &support.context, 3, &allocator));
	RCCHECK(rclc_executor_add_timer(&executor, &timer));
	RCCHECK(rclc_executor_add_subscription(&executor, &ping_subscriber, &incoming_ping,
		&ping_subscription_callback, ON_NEW_DATA));
	RCCHECK(rclc_executor_add_subscription(&executor, &pong_subscriber, &incoming_pong,
		&pong_subscription_callback, ON_NEW_DATA));

	printf("test printf %d \r\n", test_printf_count++);
// Create and allocate the pingpong messages
	char outcoming_ping_buffer[STRING_BUFFER_LEN];
	outcoming_ping.frame_id.data = outcoming_ping_buffer;
	outcoming_ping.frame_id.capacity = STRING_BUFFER_LEN;

	char incoming_ping_buffer[STRING_BUFFER_LEN];
	incoming_ping.frame_id.data = incoming_ping_buffer;
	incoming_ping.frame_id.capacity = STRING_BUFFER_LEN;

	char incoming_pong_buffer[STRING_BUFFER_LEN];
	incoming_pong.frame_id.data = incoming_pong_buffer;
	incoming_pong.frame_id.capacity = STRING_BUFFER_LEN;

	device_id = rand();
	
	printf("test printf %d \r\n", test_printf_count++);

	while(1){
		rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10));
		usleep(10000);
	}

	// Free resources
	RCCHECK(rcl_publisher_fini(&ping_publisher, &node));
	RCCHECK(rcl_publisher_fini(&pong_publisher, &node));
	RCCHECK(rcl_subscription_fini(&ping_subscriber, &node));
	RCCHECK(rcl_subscription_fini(&pong_subscriber, &node));
	RCCHECK(rcl_node_fini(&node));
}
```

2. firmware/mcu_ws/uros/rclc/rclc/src/rclc/init.c
```
rcl_ret_t
rclc_support_init(
  rclc_support_t * support,
  int argc,
  char const * const * argv,
  rcl_allocator_t * allocator)
{
  int counter = 0;
  printf("%s counter : %d \r\n", __func__, counter++);
  RCL_CHECK_FOR_NULL_WITH_MSG(
    support, "support is a null pointer", return RCL_RET_INVALID_ARGUMENT);
  printf("%s counter : %d \r\n", __func__, counter++);
RCL_CHECK_FOR_NULL_WITH_MSG(
    allocator, "allocator is a null pointer", return RCL_RET_INVALID_ARGUMENT);
  rcl_ret_t rc = RCL_RET_OK;
printf("%s counter : %d \r\n", __func__, counter++);

 printf("%s counter : %d \r\n", __func__, counter++);
 rcl_init_options_t init_options = rcl_get_zero_initialized_init_options();
  printf("%s counter : %d \r\n", __func__, counter++);
rc = rcl_init_options_init(&init_options, (*allocator) );
  if (rc != RCL_RET_OK) {
    PRINT_RCLC_ERROR(rclc_support_init, rcl_init_options_init);
    return rc;
  }
printf("get to end %s counter : %d \r\n", __func__, counter++);

  rc = rclc_support_init_with_options(support, argc, argv, &init_options, allocator);

  printf("get to end %s counter : %d \r\n", __func__, counter++);

  return rc;
}
```

3. firmware/mcu_ws/uros/rclc/rclc/src/rclc/init.c
```
rcl_ret_t
rclc_support_init_with_options(
  rclc_support_t * support,
  int argc,
  char const * const * argv,
  rcl_init_options_t * init_options,
  rcl_allocator_t * allocator)
{
 int counter = 0;	
 printf("%s counter : %d \r\n", __func__, counter++);
 RCL_CHECK_FOR_NULL_WITH_MSG(
    support, "support is a null pointer", return RCL_RET_INVALID_ARGUMENT);
   printf("%s counter : %d \r\n", __func__, counter++);
 RCL_CHECK_FOR_NULL_WITH_MSG(
    init_options, "init_options is a null pointer", return RCL_RET_INVALID_ARGUMENT);
   printf("%s counter : %d \r\n", __func__, counter++);
 RCL_CHECK_FOR_NULL_WITH_MSG(
    allocator, "allocator is a null pointer", return RCL_RET_INVALID_ARGUMENT);
  rcl_ret_t rc = RCL_RET_OK;
  printf("%s counter : %d \r\n", __func__, counter++);

  memcpy(&support->init_options, init_options, sizeof(rcl_init_options_t));
  printf("%s counter : %d \r\n", __func__, counter++);

   printf("%s counter : %d \r\n", __func__, counter++);
 support->context = rcl_get_zero_initialized_context();
   printf("%s counter : %d \r\n", __func__, counter++);
 rc = rcl_init(argc, argv, &support->init_options, &support->context);
   printf("%s counter : %d \r\n", __func__, counter++);
 if (rc != RCL_RET_OK) {
    PRINT_RCLC_ERROR(rclc_init, rcl_init);
    return rc;
  }
  support->allocator = allocator;
  printf("%s counter : %d \r\n", __func__, counter++);

  rc = rcl_clock_init(RCL_STEADY_TIME, &support->clock, support->allocator);
  if (rc != RCL_RET_OK) {
    PRINT_RCLC_ERROR(rclc_init, rcl_clock_init);
  }
   printf("%s counter : %d \r\n", __func__, counter++);
 return rc;
}

```

any thing help?

pablogs9, 2021-02-22:
Which RTOS are you using?
Which board are you using exactly?
Which transport are you using?

Could you share the output of the micro-ROS Agent with a `-v6` flag?

Thanks!

Needrom, 2021-02-22:
sorry about the the link below .It can't navigate to the project before.Now it could.

It can't run up yet.some initialization can't be finish. Don't know the mean about ouput with ```-v6``` flag.

some bad new is the code is running to this. 

/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c

```
  if (!uxr_init_udp_transport(
      &context_impl->transport, &context_impl->udp_platform, ip_protocol,
      context_impl->connection_params.agent_address, context_impl->connection_params.agent_port))
  {
    RMW_SET_ERROR_MSG("Can not create an udp connection");
    return RMW_RET_ERROR;
  }

```
but I'm in the serial mode not udp.

pablogs9, 2021-02-22:
Recently we have changed the approach of the transports. Please check this: https://github.com/Needrom/Microros-on-DIscovery-board/pull/2

I have not built it and maybe you need to change something, but this is the new way of using serial transports in micro-ROS.

Don't hesitate to ask anything if you have any more doubts.

Needrom, 2021-02-22:
OK

with some error output it seem it didn't choose a rigth Transport mode 
```
--- stderr: rcutils
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_get_cwd':
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c:60:24: warning: unused parameter 'buffer' [-Wunused-parameter]
 rcutils_get_cwd(char * buffer, size_t max_length)
                 ~~~~~~~^~~~~~
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c:60:39: warning: unused parameter 'max_length' [-Wunused-parameter]
 rcutils_get_cwd(char * buffer, size_t max_length)
                                ~~~~~~~^~~~~~~~~~
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_is_directory':
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c:83:35: warning: unused parameter 'abs_path' [-Wunused-parameter]
 rcutils_is_directory(const char * abs_path)
                      ~~~~~~~~~~~~~^~~~~~~~
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_is_file':
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c:102:30: warning: unused parameter 'abs_path' [-Wunused-parameter]
 rcutils_is_file(const char * abs_path)
                 ~~~~~~~~~~~~~^~~~~~~~
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_exists':
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c:121:29: warning: unused parameter 'abs_path' [-Wunused-parameter]
 rcutils_exists(const char * abs_path)
                ~~~~~~~~~~~~~^~~~~~~~
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_is_readable':
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c:136:34: warning: unused parameter 'abs_path' [-Wunused-parameter]
 rcutils_is_readable(const char * abs_path)
                     ~~~~~~~~~~~~~^~~~~~~~
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_is_writable':
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c:158:34: warning: unused parameter 'abs_path' [-Wunused-parameter]
 rcutils_is_writable(const char * abs_path)
                     ~~~~~~~~~~~~~^~~~~~~~
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_is_readable_and_writable':
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c:180:47: warning: unused parameter 'abs_path' [-Wunused-parameter]
 rcutils_is_readable_and_writable(const char * abs_path)
                                  ~~~~~~~~~~~~~^~~~~~~~
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_mkdir':
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c:257:13: warning: implicit declaration of function 'mkdir' [-Wimplicit-function-declaration]
   int ret = mkdir(abs_path, 0775);
             ^~~~~
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_calculate_directory_size':
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c:267:47: warning: unused parameter 'directory_path' [-Wunused-parameter]
 rcutils_calculate_directory_size(const char * directory_path, rcutils_allocator_t allocator)
                                  ~~~~~~~~~~~~~^~~~~~~~~~~~~~
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c:267:83: warning: unused parameter 'allocator' [-Wunused-parameter]
 rcutils_calculate_directory_size(const char * directory_path, rcutils_allocator_t allocator)
                                                               ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c: In function 'rcutils_get_file_size':
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/filesystem.c:325:36: warning: unused parameter 'file_path' [-Wunused-parameter]
 rcutils_get_file_size(const char * file_path)
                       ~~~~~~~~~~~~~^~~~~~~~~
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/logging.c: In function 'rcutils_get_env_var_zero_or_one':
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/logging.c:127:35: warning: unused parameter 'zero_semantic' [-Wunused-parameter]
   const char * name, const char * zero_semantic,
                      ~~~~~~~~~~~~~^~~~~~~~~~~~~
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/logging.c:128:16: warning: unused parameter 'one_semantic' [-Wunused-parameter]
   const char * one_semantic)
   ~~~~~~~~~~~~~^~~~~~~~~~~~
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/logging.c: In function 'rcutils_logging_console_output_handler':
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/logging.c:797:35: warning: implicit declaration of function 'isatty' [-Wimplicit-function-declaration]
 # define IS_STREAM_A_TTY(stream) (isatty(fileno(stream)) != 0)
                                   ^~~~~~
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/logging.c:807:22: note: in expansion of macro 'IS_STREAM_A_TTY'
       is_colorized = IS_STREAM_A_TTY(g_output_stream); \
                      ^~~~~~~~~~~~~~~
/home/mego-ros2/microros_ws/firmware/mcu_ws/uros/rcutils/src/logging.c:928:3: note: in expansion of macro 'IS_OUTPUT_COLORIZED'
   IS_OUTPUT_COLORIZED(is_colorized)
   ^~~~~~~~~~~~~~~~~~~
[... cut ...]
  Transport not supported.  Use "serial", "udp", "custom"

---
Failed   <<< rmw_microxrcedds [0.64s, exited with code 1]
make: *** [Makefile:334: colcon_compile] Error 1

```

and the configuration step configure with UART not UART1
```
ros2 run micro_ros_setup configure_firmware.sh ping_pong -t serial -d 1
echo test
/home/mego-ros2/microros_ws/install/micro_ros_setup
Configuring firmware for freertos platform stm32MakefileTest
Using serial device USART.
Configured serial mode with agent at USART

```

pablogs9, 2021-02-22:
Rebuild all from scratch, create a new micro_ros_setup workspace.

Needrom, 2021-02-23:
tough way to rebuild when I try to create the firmware again. The last time a create the firmware I try for multiple time to get it done. If anything a I should know before I create the firmware so I can make it easy.
```
Creating firmware for freertos platform olimex-stm32-e407
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml]:
	Failed to download target platform data for gbpdistro:
	<urlopen error timed out>
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Add distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"
ERROR: error loading sources list:
	<urlopen error <urlopen error [Errno 0] Error> (https://raw.githubusercontent.com/ros/rosdistro/master/foxy/distribution.yaml)>

```

-  another question
I have retarget my printf with to file which is /Core/Src/retarget.c and /Core/Inc/retarget.h. If this two file will affect project working not properly?

Needrom, 2021-02-23:
is there any wrong in here ```firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/CMakeLists.txt```

```
 77 if(${RMW_UXRCE_TRANSPORT} STREQUAL "serial")
 78   set(RMW_UXRCE_TRANSPORT_SERIAL ON)
 79   set(RMW_UXRCE_DEFAULT_SERIAL_DEVICE "/dev/ttyAMA0" CACHE STRING "Sets the agent se    rial port.")
 80 elseif(${RMW_UXRCE_TRANSPORT} STREQUAL "udp")
 81   set(RMW_UXRCE_TRANSPORT_UDP ON)
 82   if(${RMW_UXRCE_IPV} STREQUAL "ipv4")
 83     set(RMW_UXRCE_TRANSPORT_IPV4 ON)
 84     set(RMW_UXRCE_DEFAULT_UDP_IP "127.0.0.1" CACHE STRING "Sets the agent IP address    .")
 85   elseif(${RMW_UXRCE_IPV} STREQUAL "ipv6")
 86     set(RMW_UXRCE_TRANSPORT_IPV6 ON)
 87     set(RMW_UXRCE_DEFAULT_UDP_IP "::1" CACHE STRING "Sets the agent IP address.")
 88   endif()
 89   set(RMW_UXRCE_DEFAULT_UDP_PORT "8888" CACHE STRING "Sets the agent IP port.")
 90 elseif(${RMW_UXRCE_TRANSPORT} STREQUAL "custom_serial")
 91   set(RMW_UXRCE_TRANSPORT_CUSTOM_SERIAL ON)
 92 else()
 93   message(FATAL_ERROR "Transport not supported. Use \"serial\", \"udp\", \"custom\""    )
 94 endif()

```

for No "custom" I can find,because the error said ```CMake Error at CMakeLists.txt:93 (message):
  Transport not supported.  Use "serial", "udp", "custom"
```

pablogs9, 2021-02-23:
Please check:
- https://github.com/Needrom/Microros-on-DIscovery-board/pull/3
- https://asciinema.org/a/lO4kRkiqpGM1VIa97GozbmmLF

Needrom, 2021-02-23:
> Please check:
> 
> * [Needrom/Microros-on-DIscovery-board#3](https://github.com/Needrom/Microros-on-DIscovery-board/pull/3)
> * https://asciinema.org/a/lO4kRkiqpGM1VIa97GozbmmLF

OK  thanks

checking the other question for now. can't even rosdep update.Didn't know why.

Needrom, 2021-02-24:
Can you add an option which the micro_ros_setup create_firmware.sh can keep download git repo if it failed ?

pablogs9, 2021-02-24:
Sorry, we have not planned to implement this feature, if you improve the micro-ROS build system, feel free to contribute with a PR.

Needrom, 2021-02-24:
OK, may be later. have to take some time to test it. 
But I just download it one by one to the firmware.and the environment not is working.

But the problem is still here.
The code can get here ,at the line 208 in file freertos.c 
```
for(;;)
209   {
210     printf("into the printf \r\n");
211     if (eTaskGetState(xHandle) != eSuspended && availableNetwork)
212     //if(1)
213     {
214         HAL_GPIO_WritePin(GPIOF, GPIO_PIN_10, GPIO_PIN_SET);
215         osDelay(150);
216         HAL_GPIO_WritePin(GPIOF, GPIO_PIN_10, GPIO_PIN_RESET);
217         osDelay(150);
218         HAL_GPIO_WritePin(GPIOF, GPIO_PIN_9, GPIO_PIN_SET);
219         osDelay(150);
220         HAL_GPIO_WritePin(GPIOF, GPIO_PIN_9, GPIO_PIN_RESET);
221         osDelay(150);
222     }
223     else {
224 //        HAL_GPIO_WritePin(GPIOF, GPIO_PIN_10, GPIO_PIN_SET);    
225 //      osDelay(1000);
226 //      HAL_GPIO_WritePin(GPIOF, GPIO_PIN_10, GPIO_PIN_RESET);
227 //      osDelay(1000);  
228         HAL_GPIO_WritePin(GPIOF, GPIO_PIN_9, GPIO_PIN_SET);
229         osDelay(1000);
230         HAL_GPIO_WritePin(GPIOF, GPIO_PIN_9, GPIO_PIN_RESET);
231         osDelay(1000);
232     }
233   }
```
and the led is right to blink.

but the appMain may not working. and the printf after the 
```
osThreadNew(appMain, NULL, &attributes);

```
is disappear.

pablogs9, 2021-02-24:
Check your stack, how many bytes have you configured for your main stack and for appMain stack?

Needrom, 2021-02-24:
still not work with 3 * 10000;
```
osThreadAttr_t attributes;
  memset(&attributes, 0x0, sizeof(osThreadAttr_t));
  attributes.name = "microROS_app";
  attributes.stack_size = 3 * 10000;
  attributes.priority = (osPriority_t)osPriorityNormal1;
  printf("before appMain \r\n");
  osThreadNew(appMain, NULL, &attributes);
  Retarget_Init(&huart1);
  osDelay(500);
  char ptrTaskList[500];
  vTaskList(ptrTaskList);
  printf("**********************************\n");
  printf("Task  State   Prio    Stack    Num\n");
  printf("**********************************\n");
  printf(ptrTaskList);
  printf("**********************************\n");

```

Needrom, 2021-02-24:
I may think 3 * 10000 is big enough. but still not working.
```
osThreadAttr_t attributes;
  memset(&attributes, 0x0, sizeof(osThreadAttr_t));
  attributes.name = "microROS_app";
  attributes.stack_size = 3 * 10000;
  attributes.priority = (osPriority_t)osPriorityNormal1;
  printf("before appMain \r\n");
  osThreadNew(appMain, NULL, &attributes);
  Retarget_Init(&huart1);
  osDelay(500);
  char ptrTaskList[500];
  vTaskList(ptrTaskList);
  printf("**********************************\n");
  printf("Task  State   Prio    Stack    Num\n");
  printf("**********************************\n");
  printf(ptrTaskList);
  printf("**********************************\n");

```

pablogs9, 2021-02-24:
And can you debug where is your micro-ROS app failing?

Needrom, 2021-02-27:
### debug step 
- appMain -> rclc_support_init -> rclc_support_init_with_options -> rmw_init -> uxr_create_session -> uxr_create_session_retries -> wait_session_status -> send_message -> send_custom_msg -> uxr_write_framed_msg -> uxr_framing_write_transport -> write_cb

step in ``` bool sent = session->comm->send_msg(session->comm->instance, buffer, length); ``` come to send_custom_msg  

step in ```write_cb``` turn to here ```freertos_serial_write```

```HAL_UART_Transmit_DMA``` is success

stuck in here ```microros_transports.c``` into the ```freertos_serial_write``` function

```
while (ret == HAL_OK && uart->gState != HAL_UART_STATE_READY){                                             │
   │36                  osDelay(1);                                                                                                │
   │37                  }     
```
is seem the  ```uart->gState``` here is allway ```HAL_UART_STATE_BUSY_TX```

Needrom, 2021-02-28:
OK I didn't noticed that you have pull a request to fix the DMA below.
but after merge that too. The printf still stuck in the rclc_support_init in appMain
but when I add the DMA option from CubeMX some new happen,but still can't connect to the ros2
here is the new

[image]

Needrom, 2021-03-01:
Sorry for learning openocd recently.No use it before

the code stuck in here now
```
3334        static portTASK_FUNCTION( prvIdleTask, pvParameters )                     │
   │3335        {                                                                         │
   │3336                /* Stop warnings. */                                              │
   │3337                ( void ) pvParameters;                                            │
   │3338                                                                                  │
   │3339                /** THIS IS THE RTOS IDLE TASK - WHICH IS CREATED AUTOMATICALLY WH│
   │3340                SCHEDULER IS STARTED. **/                                         │
   │3341                                                                                  │
   │3342                /* In case a task that has a secure context deletes itself, in whi│
   │3343                the idle task is responsible for deleting the task's secure contex│
   │3344                any. */                                                           │
   │3345                portALLOCATE_SECURE_CONTEXT( configMINIMAL_SECURE_STACK_SIZE );   │
   │3346                                                                                  │
   │3347                for( ;; )                                                         │
   │3348                {                                                                 │
   │3349                        /* See if any tasks have deleted themselves - if so then t│
   │3350                        is responsible for freeing the deleted task's TCB and stac│
  >│3351                        prvCheckTasksWaitingTermination();        
```

pablogs9, 2021-03-01:
It seems that your serial transport is not sending data. Have you tried to send data over DMA? Have you changed any configuration related?

Needrom, 2021-03-01:
> It seems that your serial transport is not sending data. Have you tried to send data over DMA? Have you changed any configuration related?

Actually ```B+>│34                  ret = HAL_UART_Transmit_DMA(uart, buf, len);   ``` this line will send a little bit.
in HEX
```
7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 06 C3 81 83 81 00 FC 01 6A B5 
```

in Str
```
~€XRCE脕儊?j?
```

pablogs9, 2021-03-01:
That's correct, this is the XRCE init frame. So if this is being sent over the serial the agent should answer with a "session ok" message.

The point is why is it hanging in the `while (ret == HAL_OK && uart->gState != HAL_UART_STATE_READY)`?

Needrom, 2021-03-01:
> That's correct, this is the XRCE init frame. So if this is being sent over the serial the agent should answer with a "session ok" message.
> 
> The point is why is it hanging in the `while (ret == HAL_OK && uart->gState != HAL_UART_STATE_READY)`?

Yes the state seem to stay in the HAL_UART_STATE_TX

pablogs9, 2021-03-01:
Maybe you should review your serial device configuration because it should exit `HAL_UART_STATE_TX` as soon as all bytes are sent.

Have you checked if the `len` in this function is correct? The function should send `len` bytes from `buf`.

Needrom, 2021-03-02:
All serial device configuration is auto generate, some relate function which may change the serial option had comment.And the ```uart->gState``` as usual.

Needrom, 2021-03-03:
yeah fix that with add global interrupt for usart, and Then it won't stuck in the state of TX.

[image]

and it can connect now

closed 2021-03-03 by Needrom