micro-ROS/micro_ros_setup #278: Trying to run the ping_pong example with Zephyr on a ST Nucleo F446RE board

triveria, 2021-02-21:
# Trying to run the ping_pong example with Zephyr on a ST Nucleo F446RE board
I'm trying to get started with micro-ROS/Zephyr on a ST Nucleo F446RE board by using the docker image ros:foxy and trying to follow the ping_pong example from the tutorial: https://micro.ros.org/docs/tutorials/core/first_application_rtos/zephyr/
I can compile and flash it, but when I try to subscribe to the micro-ROS ping topic, I always get the message

```
root@WML:/microros_ws# ros2 topic echo /microROS/ping
WARNING: topic [/microROS/ping] does not appear to be published yet
Could not determine the type for the passed topic
```

# Here is what I've tried so far
Following this reply from pablogs9 here https://github.com/micro-ROS/micro_ros_setup/issues/161#issuecomment-665126290, I've:
* added nucleo_f446re to /microros_ws/src/micro_ros_setup/config/zephyr/generic/supported_platforms
* added to /microros_ws/src/micro_ros_setup/config/zephyr/generic/build.sh the lines responsible for platform renaming
* `colcon build`
* `ros2 run micro_ros_setup create_firmware_ws.sh zephyr nucleo_f446re`
* added my board name to the list of compatible boards in /microros_ws/firmware/zephyr_apps/apps/ping_pong/CMakeLists.txt

## Here it starts to get fuzzy
```
root@WML:/microros_ws# ros2 run micro_ros_setup configure_firmware.sh ping_pong --transport serial
Please check firmware/zephyr_apps/microros_extensions/microros_transports.h
for configuring serial device before build.
```

I've looked at /microros_ws/firmware/zephyr_apps/microros_extensions/microros_transports.h and microros_transports.c, but don't know what I should do here, since it looks very platform independent.

Moving on, I build + flash the firmware and create + build + run the agent:

```
root@WML:/microros_ws# ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_066FFF495057717867191616-if02 -v6
[1613898205.382793] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 4
[1613898205.382972] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
```
`root@WML:/microros_ws# ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_066FFF495057717867191616-if02 -d1 -v6`

and
`root@WML:/microros_ws# ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_066FFF495057717867191616-if02 -d2 -v6`
yield the same output.

When I attach another console to the docker instance and try to subscribe to the ping topic, ros2 exits with the following message:
```
root@WML:/microros_ws# ros2 topic echo /microROS/ping
WARNING: topic [/microROS/ping] does not appear to be published yet
Could not determine the type for the passed topic
```

## Testing around
For testing purposes I've done the same steps again and modified the firmware/zephyr_apps/apps/ping_pong/serial.conf to include
```
CONFIG_PRINTK=y
CONFIG_STDOUT_CONSOLE=y
```

and looked at the UART output via minicom. There I continuously receive the message
`Serial device not found`
which originates from firmware/zephyr_apps/microros_extensions/microros_transports.c

# Question
So my question is: What is missing so that the ping_pong example will run on the nucleo_f446re?

- Hardware description: ST Nucleo F446RE, https://docs.zephyrproject.org/latest/boards/arm/nucleo_f446re/doc/index.html
- RTOS: Zephyr
- Version or commit hash: ros:foxy docker image (downloaded 19.02.2020)

triveria, 2021-02-21:
I've solved the issue now and the ping_pong example works flawlessly on the nucleo_f446re.

The issues consisted of 2 subissues:
* micro-ROS picks the wrong UART
* nucleo needs to boot **after** the agent was started. This needs to be done every time we stop and restart the agent

I write my steps here in detail, in hope that someone else will benefit from it and more boards get added to the list of supported micro-ROS+Zephyr boards.

# micro-ROS picked the wrong UART

To detect that issue, I modified firmware/zephyr_apps/apps/ping_pong/serial.conf
Here added / modified the following lines:
`CONFIG_PRINTK=y`
otherwise, printk-debugging will not work.

After that, I modified /microros_ws/firmware/zephyr_apps/microros_extensions/microros_transports.c:

After the line
`sprintf(uart_descriptor,"UART_%d", params->fd);`
I added
`printk("Used uart_descriptor is %s\n", uart_descriptor);`
via minicom I then saw
`Used uart_descriptor is UART_0`

From previous experience with Zephyr on my nucleo I know, that the usable UARTS are UART_1 and UART_2.
I've not seen UART_0 before.
UART_1 has to be accessed with a separate UART-USB-device.
UART_2 is the one connected to the USB port of the nucleo, so we go for that one.
(Here I recommend that you play around with Zephyr separately and simply try to get the hello_world sample there to print "Hello World" to various UARTS. More infos here:
https://docs.zephyrproject.org/2.4.0/samples/hello_world/README.html
https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_UART_CONSOLE_ON_DEV_NAME.html
board reference. In my case: https://docs.zephyrproject.org/latest/boards/arm/nucleo_f446re/doc/index.html
)

To get it to use another UART, I added in /microros_ws/firmware/zephyr_apps/microros_extensions/microros_transports.c
`params->fd = 2;`
before the line
`sprintf(uart_descriptor,"UART_%d", params->fd);`

Now I compiled and flashed it again.

# nucleo needs to boot _after_ the agent is started

This I rather stumbled upon by accident.
I first needed to start the agent, and then boot the nucleo (either via the RESET button or by pluging it in only afterwards)
Now I see ping messages from the nucleo and it responds to fake_pings.

# Open questions

This leads to 2 questions:

1. Is there a .cfg, where I can set which UART the nucleo uses for its ROS messages?
2. Why does the agent have to be started first, before I can boot the nucleo? Can this be circumvented somehow or can I reset the communication via software? In case the agent crashes, I do not want to unplug and replug the nucleo.

triveria, 2021-02-21:
I'm closing this issue now, since I've found the answer to question 2.

For someone to stumble upon this issue in the future:
The reason why the agent needs to be started before the board is not a config in ROS or something, but a [line](https://github.com/micro-ROS/zephyr_apps/blob/foxy/apps/ping_pong/src/main.c#L98) in the ping_pong example:
`RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));`
When there is no agent, rclc_support_init(..) will fail and thus RCCHECK(..) will prematurely exit the main function.

For question about the default UART I will open a separate issue.

closed 2021-02-21 by triveria

jamoralp, 2021-02-22:
Good morning  @triveria ,
Indeed, you found the answer to your question. Anyway, if you want to leverage your application's logic, you can always use the recently implemented `ping` feature to know if the micro-ROS Agent is already up and running, or not. Take a look at [this example.](https://github.com/micro-ROS/micro-ROS-demos/blob/foxy/rclc/ping_uros_agent/main.c)

triveria, 2021-02-22:
Hi @jamoralp,
I can definitely use that code example and ping feature. Thank you for showing it to me.