micro-ROS/micro_ros_setup #279: Micro-ROS on Zephyr picks the wrong UART as default UART on nucleo_f446re

triveria, 2021-02-21:
## Micro-ROS on Zephyr picks the wrong UART as default UART on nucleo_f446re

- Hardware description: [ST Nucleo F446RE](https://docs.zephyrproject.org/latest/boards/arm/nucleo_f446re/doc/index.html)
- RTOS: Zephyr
- Installation type: ros2 run micro_ros_setup create_firmware_ws.sh zephyr nucleo_f446re
- Version or commit hash: foxy

#### Steps to reproduce the issue

Basically trying to get the [ping_pong](https://micro.ros.org/docs/tutorials/core/first_application_rtos/zephyr/) example to work, with board = nucleo_f446re and transport = serial

```
# Create step
ros2 run micro_ros_setup create_firmware_ws.sh zephyr nucleo_f446re

# Configure step with ping_pong app and serial transport
ros2 run micro_ros_setup configure_firmware.sh ping_pong --transport serial
```

in [serial.conf](https://github.com/micro-ROS/zephyr_apps/blob/foxy/apps/ping_pong/serial.conf#L15) set
`CONFIG_PRINTK=y`

```
# Build step
ros2 run micro_ros_setup build_firmware.sh

# Flash step
ros2 run micro_ros_setup flash_firmware.sh
```

Check serial communication with nucleo:
```
$ minicom -b 115200 -D /dev/ttyACM0
Serial device not found
Serial device not found
...
```

#### Expected behavior

[zephyr_transport_open](https://github.com/micro-ROS/zephyr_apps/blob/foxy/microros_extensions/microros_transports.c#L44) should be called such that it picks UART_1 or UART_2 or at least the default UART specified in the device tree by Zephyr.

Maybe through an additional command line option? A suggestion:
`ros2 run micro_ros_setup configure_firmware.sh ping_pong --transport serial --uart 2`
Or is something like that already available?

#### Actual behavior

[zephyr_transport_open](https://github.com/micro-ROS/zephyr_apps/blob/foxy/microros_extensions/microros_transports.c#L44) gets called such that it picks UART_0, which does not exist on the nucleo_f446re.

#### Additional information

nucleo_f446re has UART_1(accessible via GPIO pins) and UART_2 (accessible via USB).
Zephyr also only references these 2 and chooses UART_2 for both console and shell as can be seen [here](https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/nucleo_f446re/nucleo_f446re.dts#L17).

pablogs9, 2021-02-22:
As you can see [here](https://github.com/micro-ROS/zephyr_apps/blob/2557814c132e9865429471a04024ad682afb1daf/apps/ping_pong/src/main.c#L86) default parameters for the Zephyr transports are being set to [UART_0](https://github.com/micro-ROS/zephyr_apps/blob/2557814c132e9865429471a04024ad682afb1daf/microros_extensions/microros_transports.h#L46)

Is it just a matter of providing a different `zephyr_transport_params_t` to the `rmw_uros_set_custom_transport`, you can take this param from the Zephyr config file.

Is this a good solution for you?

Thanks!

triveria, 2021-02-22:
Yes, that solution fits. Thank you very much for showing that to me.

However, without your comment I would not have known about this.
I added a little comment in this pull request [here](https://github.com/triveria/zephyr_apps/pull/1) so that others know about this option right away.

closed 2021-02-22 by triveria