micro-ROS/micro_ros_setup #280: Issue for reconnect through

ncnynl, 2021-02-22:
## Issue template

- Hardware description: 
Olimex STM32-E407 

- RTOS: 
freertos

- Installation type: 
micro_ros_agent 

- Version or commit hash: 
foxy 

#### Steps to reproduce the issue

#### Expected behavior
Can be executed multiple times using “ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 -v6 “ when  stop using ctrc + c

#### Actual behavior

At present, using micros can connect to the board through “ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 -v6“, but if I stop using ctrc + c, I can't connect successfully if I want to execute it again.

#### Additional information

I can only connect normally after running the command “ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 -v6“ after executing “ros2 run micro_ros_setup flash_firmware.sh“  every time

pablogs9, 2021-02-22:
You should be running `flash_firmware.sh` once. 

Then, you need to have an open agent before the micro-ROS code starts to run. So when using the `int32_example`, you have to reset your board if you close the agent.

Have I understood well your question?

ncnynl, 2021-02-22:
Examples are as follows:

1. Power on the board
2. Run the agent on the computer through the command "ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 -v6"
3. It can be connected to the board ports normally and work normally
4. Turn off the agent on the computer, and press ctrl + c
5. Run the agent again on the computer, "ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 -v6"
6. Stay in
[1605573834.109076] info | TermiosAgentLinux.cpp | init | running... | fd: 3
[1605573834.109228] info | Root.cpp | set_verbose_level | logger setup | verbose_level: 6
7. I tried many times and it was the same.
8. Re-power on my board or flash it again to work normally again.

pablogs9, 2021-02-22:
Obviously, none of the examples takes care of handling the reconnection between client and agent. So if you destroy the agent, the client will not create the entities again since the examples are not like this.

You can find a reconnection example [here](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino).

ncnynl, 2021-02-22:
@pablogs9  thank you ， i have a look !

pablogs9, 2021-02-22:
Closing since not an issue, we can continue talking here if you have any other doubts.

closed 2021-02-22 by pablogs9