micro-ROS/micro_ros_setup #281: Error in rcl_init

davidm312, 2021-02-24:
## Issue template
Hey Guys, i was trying to publish data from an ultrasonic sensor to my ROS client with the esp32 (wifi). I modified the int32_publisher program to get it work but even the unmodified program doesn't work properly. It works sometimes but most of the time i am getting and rclc ERROR. I am just a student and new to micro-ros so sorry if i am missing something but i can't get what's wrong.

- ESP32: 
- FREERTOS: 
- Foxy: 

#### Steps to reproduce the issue

**These are the steps i typed in the shell->

ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32 

ros2 run micro_ros_setup configure_firmware.sh int32_publisher -t udp -i 127.178.***.** -p 8888

docker run -it --rm --net=host microros/micro-ros-agent:foxy udp4 --port 8888 -v6

//declaring the Wifi SSID and password
ros2 run micro_ros_setup build_firmware.sh menuconfig

ros2 run micro_ros_setup build_firmware.sh

ros2 run micro_ros_setup flash_firmware.sh

ros2 run micro_ros_setup build_firmware.sh monitor

and then when i monitor the process i am getting this error:

ros2 run micro_ros_setup build_firmware.sh monitor**

**I (2154) wifi station: connected to ap SSID:FRITZ!Box 7560
micro-ROS transport: connected using UDP mode, ip: '192.168.***.**', port: '8888'
[ERROR] [0000000011.575562000] [rclc]: [rclc_init] Error in rcl_init: failed to create node session on Micro ROS Agent., at /home/david/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c:242, at /home/david/uros_ws/firmware/mcu_ws/uros/rcl/rcl/src/rcl/init.c:233**

Failed status on line 37: 1. Aborting.

pablogs9, 2021-02-24:
You have literally written `127.178.***.**` or just hide your actual IP in the copy-paste?

pablogs9, 2021-02-24:
In any case, it seems weird that you have configured `127.178.***.**` and it is trying to connect to `'192.168.***.**'`, let me check

pablogs9, 2021-02-24:
I just have tested your procedure and it works well, let us know if you are setting the IP correctly.

davidm312, 2021-02-24:
Oh sorry ye i forgot to mention. The ip address is fine i guess, i just changed them manually in the copy paste and hided the last 5 numbers.I am using the ip address from cmd: **hostname -I**.

pablogs9, 2021-02-24:
But, is it a local IP (LAN) or Internet IP (WAN)?

pablogs9, 2021-02-24:
You are not setting well this IP, after running the configuration example you should be able to find the IP set in `firmware/mcu_ws/colcon.meta`.

davidm312, 2021-02-24:
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        },
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_NO_THREAD_SUPPORT=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON"
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-DUCLIENT_PROFILE_UDP=ON",
                "-DUCLIENT_PROFILE_TCP=OFF"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_XML_BUFFER_LENGTH=400",
                "-DRMW_UXRCE_TRANSPORT=udp",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=192.168.178.59",
                "-DRMW_UXRCE_DEFAULT_UDP_PORT=8888"
            ]
        }
    }
}

pablogs9, 2021-02-24:
I have seen that you have edited the IP in the first comment...

The point is that if you are using a Local Area Network IP it should work using the steps in the first comment.

In fact, the configure command just copy your IP to this meta file and this is used as default IP.

If you are having troubles you also can [set the agent IP at runtime](https://github.com/micro-ROS/micro-ROS-demos/blob/985e6eca545beb897fcbb4a3782516cc914488d0/rclc/configuration_example/configured_publisher/main.c#L43).

closed 2021-02-24 by davidm312