micro-ROS/micro_ros_setup #284: Issue setting up agent on jetson nano

mrbhjv, 2021-02-25:
## Issue template

- Hardware description: Jetson nano dev kit
- Installation type: Following guide to build agent
- Version or commit hash: Foxy

#### Steps to reproduce the issue
On a dockerfile on the Jetson

```
FROM ros:foxy-ros-base-l4t-r32.4.4

RUN git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup

RUN rosdep update && rosdep install --from-path src --ignore-src -y

RUN . /opt/ros/$ROS_DISTRO/setup.sh && colcon build
RUN . install/setup.sh \
    && ros2 run micro_ros_setup create_agent_ws.sh \
    && ros2 run micro_ros_setup build_agent.sh
```

#### Expected behavior
No errors and agent builds correctly. I have an identical container targetting just foxy that builds with no problem on x86_64 (my pc).

#### Actual behavior
running `rosdep` gives me 
```
Warning: running 'rosdep update' as root is not recommended.
  You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Add distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"
Skip end-of-life distro "groovy"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Add distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in /root/.ros/rosdep/sources.cache
Error(s) in package '/opt/ros/foxy/build/pluginlib/prefix/share/test_pluginlib/package.xml':
Package 'test_pluginlib' must declare at least one maintainer
The package node must contain at least one "license" tag
```
I've tried running it with `non-root` user as well, with the same result.

I get the same error with
```
ros2 run micro_ros_setup create_agent_ws.sh
```
```
ros2 run micro_ros_setup create_agent_ws.sh
.......
=== ./eProsima/Micro-CDR (git) ===
Cloning into '.'...
=== ./eProsima/Micro-XRCE-DDS-Agent (git) ===
Cloning into '.'...
=== ./uros/drive_base (git) ===
Cloning into '.'...
=== ./uros/micro-ROS-Agent (git) ===
Cloning into '.'...
=== ./uros/micro_ros_msgs (git) ===
Cloning into '.'...
=== ./uros/rmw_microxrcedds (git) ===
Cloning into '.'...
=== ./uros/rosidl_typesupport_microxrcedds (git) ===
Cloning into '.'...
Error(s) in package '/opt/ros/foxy/build/pluginlib/prefix/share/test_pluginlib/package.xml':
Package 'test_pluginlib' must declare at least one maintainer
The package node must contain at least one "license" tag
```

#### Additional information

It'd be nice to have a docker image for the agent that can run on a jetson, that way I wouldn't even need to go through this steps.

Thanks in advance.

pablogs9, 2021-02-25:
I guess that this issue is not related to micro-ROS but with this `test_pluginlib` package.

Regarding the micro-ROS Agent, maybe it is easier for you to try to build a basic Micro XRCE-DDS Agent (the core of micro-ROS agent). You can do it with:

```bash
git clone -b foxy https://github.com/eProsima/Micro-XRCE-DDS-Agent
cd Micro-XRCE-DDS-Agent
mkdir build && cd build
cmake ..
make 
```

You should be able to build an Agent that is compatible with the basic features of micro-ROS.

Let me know if you find any problem doing this.

mrbhjv, 2021-02-25:
Dumb question, but could you point me to `how to run the agent`?
The reason why I added it here, is because I don't get the same issue when running it on a different system so that you're aware of it.

When building the agent manually, I also get
```
Switched to a new branch '2.0.x'
Submodule 'thirdparty/android-ifaddrs' (https://github.com/michalsrb/android-ifaddrs.git) registered for path 'thirdparty/android-ifaddrs'
Submodule 'thirdparty/asio' (https://github.com/chriskohlhoff/asio.git) registered for path 'thirdparty/asio'
Submodule 'thirdparty/fastcdr' (https://github.com/eProsima/Fast-CDR.git) registered for path 'thirdparty/fastcdr'
Submodule 'thirdparty/tinyxml2' (https://github.com/leethomason/tinyxml2.git) registered for path 'thirdparty/tinyxml2'
Cloning into '/uros_ws/build/microxrcedds_agent/fastdds/src/fastdds/thirdparty/android-ifaddrs'...
Cloning into '/uros_ws/build/microxrcedds_agent/fastdds/src/fastdds/thirdparty/asio'...
Cloning into '/uros_ws/build/microxrcedds_agent/fastdds/src/fastdds/thirdparty/fastcdr'...
Cloning into '/uros_ws/build/microxrcedds_agent/fastdds/src/fastdds/thirdparty/tinyxml2'...
---
Finished <<< microxrcedds_agent [13min 58s]
Starting >>> micro_ros_agent
[Processing: micro_ros_agent]                                   
--- stderr: micro_ros_agent                                        
/usr/bin/ld: warning: libfastrtps.so.2, needed by /opt/ros/foxy/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so, may conflict with libfastrtps.so.2.0.2
---
```

It could be because the jetson image is outdated I guess?

pablogs9, 2021-02-25:
I think so... Maybe the docker is not up to date.
Is possible for you to try the instructions for building the agent I provided in the last comment?

mrbhjv, 2021-02-25:
Building with
```
git clone -b foxy https://github.com/eProsima/Micro-XRCE-DDS-Agent
cd Micro-XRCE-DDS-Agent
mkdir build && cd build
cmake ..
make 
```
works with no errors. I see the client connect to the agent, but I can't see anything from the ROS (topics, nodes, etc) tree. I guess this is an old thing that was added to micro-ros.

I manually built the agent following the same steps I showed on the docker container, simply ignoring the errors and the agent does build and seems to work fine. It's just a strange error with that plugin.

Anyways, would be nice if there's an image that I could just run on a jetson.

Thanks

closed 2021-02-25 by mrbhjv