micro-ROS/micro_ros_setup #287: ping_pong app not so right on my board

Needrom, 2021-03-04:
## Issue template

- Hardware description: https://github.com/Needrom/Microros-on-DIscovery-board
- RTOS: freertos
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
run ``` ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 ```
and echo the topic ```/microROS/ping```

#### Expected behavior
it should any way print the message until I kill the process or trun off my board.

#### Actual behavior
the topic just print a little.The message down below is all it printf.
```
stamp:
  sec: 4
  nanosec: 640000000
frame_id: '467594128_37129206'
---
stamp:
  sec: 6
  nanosec: 633000000
frame_id: '1691377550_37129206'
---
stamp:
  sec: 8
  nanosec: 636000000
frame_id: '1471595416_37129206'
---
stamp:
  sec: 10
  nanosec: 632000000
frame_id: '262628173_37129206'
---
stamp:
  sec: 12
  nanosec: 635000000
frame_id: '1198695175_37129206'
---
stamp:
  sec: 14
  nanosec: 632000000
frame_id: '636211359_37129206'
---
stamp:
  sec: 16
  nanosec: 634000000
frame_id: '1129188923_37129206'
---
stamp:
  sec: 18
  nanosec: 632000000
frame_id: '60369457_37129206'
---

```

#### Additional information

pablogs9, 2021-03-04:
I don't understand `the topic just print a little.The message down below is all it printf.`

In any case, you should provide more information:
- Is your board freezing at some point?
- How is the communication with the agent when the topic stops being published?
- Have you traced or debugged where and when the board stops publishing?

Needrom, 2021-03-04:
it just like the the timer here won't call the ```ping_timer_callback``` any more after 8 times or  11 times the topic message prints out.

timer in appMain
```
121         rcl_timer_t timer;
122         RCCHECK(rclc_timer_init_default(&timer, &support, RCL_MS_TO_NS(2000), ping_timer_callback));
```

I set a breakpoint at ```ping_timer_callback```
and just can get in the ```ping_timer_callback``` a few times like 8 times or 11 times, and then I can't not reach this breakpoint any more, the code will keep running and won't reach this callback again.

pablogs9, 2021-03-04:
Use the agent with the flag `-v6` and paste here the output.

Have you checked how much stack is being configured for your FreeRTOS application?

Needrom, 2021-03-08:
here is the output.
```
[1615190679.854852] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1615190679.856261] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1615190680.443659] info     | Root.cpp           | create_client            | create                 | client_key: 0x59B93166, session_id: 0x81
[1615190680.444038] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1505309030, address: 0
[1615190680.444523] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1615190680.462362] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x59B93166, len: 100, data: 
0000: 81 80 00 00 01 05 5C 00 00 0A 00 01 01 02 00 00 4E 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 70 69 6E 67 70 6F 6E 67 5F 6E 6F 64 65 3C
0040: 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70 61 6E 74 3E 3C 2F 64 64 73
0060: 3E 00 00 00
[1615190680.466207] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1615190680.472849] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1615190680.475250] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1615190680.478674] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x59B93166, participant_id: 0x000(1)
[1615190680.479438] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1615190680.479513] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1615190680.479569] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1615190680.479622] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1615190680.479676] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1615190680.483598] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1615190680.483861] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1615190680.486878] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1615190680.499678] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x59B93166, len: 128, data: 
0000: 81 80 01 00 01 05 77 00 00 0B 00 02 02 02 00 00 69 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 72 74 2F 6D 69 63 72 6F 52 4F 53 2F 70 69 6E 67 3C 2F 6E 61 6D 65 3E 3C 64 61
0040: 74 61 54 79 70 65 3E 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 48 65 61 64
0060: 65 72 5F 3C 2F 64 61 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73 3E 00 00 01 00
[1615190680.500215] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x59B93166, topic_id: 0x000(2), participant_id: 0x000(1)
[1615190680.500367] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1615190680.500439] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1615190680.504137] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 01 00 80
[1615190680.505372] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1615190680.506894] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
[1615190680.507027] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[... cut ...]
[1615190698.536227] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 0D 00 00 00 80
[1615190698.538308] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 0A 00 15 00 80
[1615190698.538421] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 16 00 00 00 80

```

Needrom, 2021-03-08:
[image]

Needrom, 2021-03-08:
some time the connect session is failed.
```
[1615191164.965756] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1615191164.966682] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1615191166.263610] info     | Root.cpp           | create_client            | create                 | client_key: 0x59B93166, session_id: 0x81
[1615191166.264038] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1505309030, address: 0
[1615191166.264305] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1615191167.573630] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x59B93166, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 59 B9 31 66 81 00 FC 01
[1615191167.574096] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1505309030, address: 0
[1615191167.574536] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1615191168.893141] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x59B93166, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 59 B9 31 66 81 00 FC 01
[1615191168.893497] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1505309030, address: 0
[1615191168.893634] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1615191170.261246] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x59B93166, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 59 B9 31 66 81 00 FC 01
[1615191170.261293] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1505309030, address: 0
[1615191170.261314] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1615191171.630837] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x59B93166, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 59 B9 31 66 81 00 FC 01
[1615191171.631177] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1505309030, address: 0
[1615191171.631310] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x59B93166, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1615191172.998491] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x59B93166, len: 12, data: 
0000: 81 00 00 00 03 01 04 00 00 02 FF FE
[1615191172.998833] info     | Root.cpp           | delete_client            | delete                 | client_key: 0x59B93166
[1615191172.998942] info     | SessionManager.hpp | destroy_session          | session closed         | client_key: 0x59B93166, address: 0

```

pablogs9, 2021-03-08:
Are you using `heap_4` of [this](https://github.com/Needrom/Microros-on-DIscovery-board/blob/main/freertos_apps/stm32MakefileTest/Core/Src/custom_memory_manager.c)?

pablogs9, 2021-03-08:
Regarding the image above, you are using 50k of heap, but how much stack have you configured?

Needrom, 2021-03-09:
actually I already add this line in Makefile
```
# Removing heap4 manager while being polite with STM32CubeMX
TMPVAR := $(C_SOURCES)
C_SOURCES := $(filter-out Middlewares/Third_Party/FreeRTOS/Source/portable/MemMang/heap_4.c, $(TMPVAR))

```

and defaultTask here
```
osThreadId_t defaultTaskHandle;
const osThreadAttr_t defaultTask_attributes = {
  .name = "defaultTask",
  .priority = (osPriority_t) osPriorityNormal,
  .stack_size = 1500 * 4
};

```

and appMain here, for the issue  previous.
It't not necessary for such a size.
```
  attributes.name = "microROS_app";
  attributes.stack_size = 3 * 10000;

```

pablogs9, 2021-03-09:
The point is that your client is not receiving data from the agent, that's why it tries to reconnect and then ends closing the session. Have you checked the serial connections? Everything works well with the serial transport in the read operation?

Needrom, 2021-03-26:
Actually, My usart seem workable if I Don't start the appMain.

Needrom, 2021-03-26:
when I want to build it with TCP option no with the serial. 
Got this error
```
Crosscompiled environment: cleaning path
Building firmware for freertos platform stm32MakefileTest
Selected app: ping_pong
rm -fR /home/mego/microros_ws/firmware/freertos_apps/stm32MakefileTest/build
rm -f /home/mego/microros_ws/firmware/freertos_apps/stm32MakefileTest/arm_toolchain.cmake; \
cat /home/mego/microros_ws/firmware/freertos_apps/stm32MakefileTest/arm_toolchain.cmake.in | \
	sed "s/@CROSS_COMPILE@/\/home\/mego\/microros_ws\/firmware\/freertos_apps\/stm32MakefileTest\/..\/..\/toolchain\/bin\/arm-none-eabi-/g" | \
	sed "s/@FREERTOS_TOPDIR@/\/home\/mego\/microros_ws\/firmware\/freertos_apps\/stm32MakefileTest\/..\/../g" | \
	sed "s/@ARCH_CPU_FLAGS@/\"-mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_HAL_DRIVER -DSTM32F407xx -D_TIMEVAL_DEFINE -O0 -Wall -fdata-sections -ffunction-sections\"/g" | \
	sed "s/@ARCH_OPT_FLAGS@/\"\"/g" | \
	sed "s/@INCLUDES@/\/home\/mego\/microros_ws\/firmware\/freertos_apps\/stm32MakefileTest\/lib\/FreeRTOS-Plus-POSIX\/include\n \/home\/mego\/microros_ws\/firmware\/freertos_apps\/stm32MakefileTest\/lib\/include\n \/home\/mego\/microros_ws\/firmware\/freertos_apps\/stm32MakefileTest\/lib\/include\/private\n \/home\/mego\/microros_ws\/firmware\/freertos_apps\/stm32MakefileTest\/lib\/include\/FreeRTOS_POSIX\n \/home\/mego\/microros_ws\/firmware\/freertos_apps\/stm32MakefileTest\/lib\/include\/FreeRTOS_POSIX\/sys\n \/home\/mego\/microros_ws\/firmware\/freertos_apps\/stm32MakefileTest\/Core\/Inc\n \/home\/mego\/microros_ws\/firmware\/freertos_apps\/stm32MakefileTest\/Drivers\/STM32F4xx_HAL_Driver\/Inc\n \/home\/mego\/microros_ws\/firmware\/freertos_apps\/stm32MakefileTest\/Drivers\/CMSIS\/Device\/ST\/STM32F4xx\/Include\n \/home\/mego\/microros_ws\/firmware\/freertos_apps\/stm32MakefileTest\/Drivers\/CMSIS\/Include\n/g" \
	> /home/mego/microros_ws/firmware/freertos_apps/stm32MakefileTest/arm_toolchain.cmake
cd /home/mego/microros_ws/firmware/freertos_apps/stm32MakefileTest/../../mcu_ws; \
colcon build \
	--packages-ignore-regex=.*_cpp \
	--metas /home/mego/microros_ws/firmware/freertos_apps/stm32MakefileTest/../../mcu_ws/colcon.meta /home/mego/microros_ws/firmware/freertos_apps/apps/ping_pong/app-colcon.meta \
	--cmake-args \
	"--no-warn-unused-cli" \
	-DCMAKE_POSITION_INDEPENDENT_CODE=OFF \
	-DTHIRDPARTY=ON \
	-DBUILD_SHARED_LIBS=OFF \
	-DBUILD_TESTING=OFF \
	-DCMAKE_BUILD_TYPE=Debug \
	-DCMAKE_TOOLCHAIN_FILE=/home/mego/microros_ws/firmware/freertos_apps/stm32MakefileTest/arm_toolchain.cmake \
	-DCMAKE_VERBOSE_MAKEFILE=ON; \

Starting >>> tinydir_vendor
Starting >>> rosidl_adapter
Finished <<< rosidl_adapter [0.60s]                                 
Starting >>> rosidl_typesupport_interface
Finished <<< rosidl_typesupport_interface [0.48s]                       
Starting >>> rosidl_parser
Finished <<< rosidl_parser [0.52s]                                      
Starting >>> rosidl_cmake
Finished <<< rosidl_cmake [0.52s]                                       
Starting >>> microcdr
Finished <<< microcdr [1.29s]                                           
Starting >>> microxrcedds_client
--- stderr: microxrcedds_client                                         
In file included from /home/mego/microros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/transport.h:32,
                 from /home/mego/microros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/util/ping.h:24,
                 from /home/mego/microros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/util/ping.c:1:
/home/mego/microros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/profile/transport/ip/udp/udp_transport.h:33:27: error: field 'platform' has incomplete type
     struct uxrUDPPlatform platform;
                           ^~~~~~~~
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:261: CMakeFiles/microxrcedds_client.dir/src/c/util/ping.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Makefile2:79: CMakeFiles/microxrcedds_client.dir/all] Error 2
make[1]: *** [Makefile:133: all] Error 2
---
Failed   <<< microxrcedds_client [1.45s, exited with code 2]
Aborted  <<< tinydir_vendor [1min 9s]                                          

Summary: 5 packages finished [1min 9s]
  1 package failed: microxrcedds_client
  1 package aborted: tinydir_vendor
  2 packages had stderr output: microxrcedds_client tinydir_vendor
  52 packages not processed
make: *** [Makefile:343: colcon_compile] Interrupt

```

pablogs9, 2021-03-26:
TCP is not supported by micro-ROS just UDP and Serial by default and custom transports.

Needrom, 2021-03-26:
> The point is that your client is not receiving data from the agent, that's why it tries to reconnect and then ends closing the session. Have you checked the serial connections? Everything works well with the serial transport in the read operation?

Can't find any wrong from the project,May be the way I downloads the eProsima/Micro-XRCE-DDS-Client is wrong, I just use ```git clone ``` to download not with the ```vcs``` in the create.sh

Needrom, 2021-03-26:
> TCP is not supported by micro-ROS just UDP and Serial by default and custom transports.

my bad.   just udp be it still go wrong in the building option.

Needrom, 2021-03-26:
> > TCP is not supported by micro-ROS just UDP and Serial by default and custom transports.
> 
> my bad. just udp be it still go wrong in the building

Needrom, 2021-03-28:
the serial is still read while the connection down,and I send it to USART3 and print out to this 
in normally print.
```

	
	
~	

	
~	
	
亐
	
~	
~	
	
	
	
	
~	
~	

	
	
	
	
	
~	
	
~	
	
	
	
	
	
	
	
M~M~	
	
	
	
昆?'	
31596089	
)	
	
	
	
115691_1O~	
231596089	

)	
	
	
@?'	
325632_12弤	
31596089	
)	
	
	
	
	
	
231596089	

)	
	
	
	
739871_12O~	
31596089	
)	
	
	
	
97361_12弤	
31596089	
)	
	
	
	
73427_12O~	
31596089	

)	
	
	
@?'	
491776_12O~	
31596089	
)	
	
	
@?'	
46314_12弤	
31596089	
)	
	
	
	
406624_1巭	
231596089	
	
靱	
	
~	
	
	
	
~	
~	

	
	
~	
	
~	
	
	
	
~	
	
~	
	

	
	
~	
	
~	
	
	
	
~	
	
~	
	
	
	
~	
	
~	
	
	
~	
~	
M~(~	
	
	
~	
~	
	
	
	
~	

	
~	
	
亐
	
~	
~	
	
	
	
	
~	
~	

	
	
	
	
	
~	
	
~	
	
	
	
	
	
	
	
M~M~	
	
	
	
昆?'	
31596089	
)	
	
	
	
115691_1O~	
231596089	

)	
	
	
@?'	
325632_12弤	
31596089	
)	
	
	
	
	
	
231596089	

)	
	
	
	
739871_12O~	
31596089	
)	
	
	
	
97361_12弤	
31596089	
)	
	
	
	
73427_12O~	
31596089	

)	
	
	
@?'	
491776_12O~	
31596089	
)	
	
	
@?'	
46314_12弤	
31596089	
)	
	
	
	
406624_1巭	
231596089	
	
靱	
	
~	
	
	
	
~	

```

In Hex
```
36 31 5F 31 32 8F 7E 09 0A 33 31 35 39 36 30 38 39 09 0A 01 29 0B 09 0A 05 09 0A 09 0A 09 0A 37 33 34 32 37 5F 31 32 4F 7E 09 0A 33 31 35 39 36 30 38 39 09 0A 0A 29 0C 09 0A 01 05 09 0A 09 0A 40 18 3F 27 15 09 0A 34 39 31 37 37 36 5F 31 32 4F 7E 09 0A 33 31 35 39 36 30 38 39 09 0A 01 29 0B 09 0A 05 09 0A 09 0A 40 18 3F 27 15 09 0A 34 36 33 31 34 5F 31 32 8F 7E 09 0A 33 31 35 39 36 30 38 39 09 0A 01 29 0B 09 0A 05 09 0A 09 0A 09 0A 34 30 36 36 32 34 5F 31 8E 7E 09 0A 32 33 31 35 39 36 30 38 39 09 0A 09 0A EC 7E 09 0A 01 06 09 0A 7E 09 0A 09 0A 09 0A 09 0A 7E 09 0A 7E 09 0A 0D 09 0A 09 0A 7E 09 0A 09 0A 7E 09 0A 09 0A 09 0A 09 0A 7E 09 0A 09 0A 7E 09 0A 09 0A 0A 01 05 09 0A 09 0A 7E 09 0A 01 05 09 0A 7E 09 0A 09 0A 09 0A 09 0A 7E 09 0A 06 09 0A 7E 09 0A 09 0A 09 0A 09 0A 7E 09 0A 09 0A 7E 09 0A 09 0A 09 0A 7E 09 0A 7E 09 0A 4D 7E 28 7E 09 0A 09 0A 09 0A 7E 09 0A 7E 09 0A 09 0A 09 0A 09 0A 7E 09 0A 0D 09 0A 7E 09 0A 09 0A 81 80 0A 09 0A 7E 09 0A 7E 09 0A 09 0A 09 0A 09 0A 09 0A 7E 09 0A 7E 09 0A 0A 01 05 09 0A 09 0A 09 0A 05 09 0A 09 0A 7E 09 0A 09 0A 7E 09 0A 09 0A 09 0A 09 0A 09 0A 09 0A 09 0A 09 0A 7F 4D 7E 4D 7E 09 0A 09 0A 09 0A 09 0A C0 A5 3F 27 15 09 0A 33 31 35 39 36 30 38 39 09 0A 01 29 0B 09 0A 05 09 0A 09 0A 09 0A 31 31 35 36 39 31 5F 31 4F 7E 09 0A 32 33 31 35 39 36 30 38 39 09 0A 0A 29 0C 09 0A 01 05 09 0A 09 0A 40 18 3F 27 15 09 0A 33 32 35 36 33 32 5F 31 32 8F 7E 09 0A 33 31 35 39 36 30 38 39 09 0A 01 29 0B 09 0A 05 09 0A 09 0A 09 0A 09 0A 09 0A 32 33 31 35 39 36 30 38 39 09 0A 0A 29 0C 09 0A 01 05 09 0A 09 0A 09 0A 37 33 39 38 37 31 5F 31 32 4F 7E 09 0A 33 31 35 39 36 30 38 39 09 0A 01 29 0B 09 0A 05 09 0A 09 0A 09 0A 39 37 33 36 31 5F 31 32 8F 7E 09 0A 33 31 35 39 36 30 38 39 09 0A 01 29 0B 09 0A 05 09 0A 09 0A 09 0A 37 33 34 32 37 5F 31 32 4F 7E 09 0A 33 31 35 39 36 30 38 39 09 0A 0A 29 0C 09 0A 01 05 09 0A 09 0A 40 18 3F 27 15 09 0A 34 39 31 37 37 36 5F 31 32 4F 7E 09 0A 33 31 35 39 36 30 38 39 09 0A 01 29 0B 09 0A 05 09 0A 09 0A 40 18 3F 27 15 09 0A 34 36 33 31 34 5F 31 32 8F 7E 09 0A 33 31 35 39 36 30 38 39 09 0A 01 29 0B 09 0A 05 09 0A 09 0A 09 0A 34 30 36 36 32 34 5F 31 8E 7E 09 0A 32 33 31 35 39 36 30 38 39 09 0A 09 0A EC 7E 09 0A 01 06 09 0A 7E 09 0A 09 0A 09 0A 09 0A 7E 09 0A 
```

Needrom, 2021-03-29:
it seem like the appMain stop running any more so that the ```rclc_executor_spin_some``` won't be call any more
I place a printf at the while loop of the appMain and it don't print any more.

Needrom, 2021-04-01:
Code has stop here. ```rcl_wait``` 

```
  // wait up to 'timeout_ns' to receive notification about which handles reveived
  // new data from DDS queue.
  rc = rcl_wait(&executor->wait_set, timeout_ns);
  RCLC_UNUSED(rc);

  printf("%s : 2 \r\n", __func__);

```

Needrom, 2021-04-05:
Yeah you right, It still the usart DMA thing that fail the connection.

closed 2021-04-06 by Needrom

Needrom, 2021-04-07:
[image]

is here the setting of usart RX not with ```circular``` mode that make the topic publish wrong

pablogs9, 2021-04-07:
Is this setting also misconfigured in the micro-ROS port for Olimex LTD STM32-E407?