micro-ROS/micro_ros_setup #288: During firmware create firmware zephyr host - fails on not resolving github for some libraries

wegunterjr, 2021-03-04:
## Issue template

- Hardware description: Linux/Ubuntu/Host
- RTOS: zephyr
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
I am running `ros2 run micro_ros_setup create_firmware_ws.sh zephyr host`  and towards the end of the step, i get errors:
`=== ./ament/ament_cmake (git) ===
Could not determine ref type of version: fatal: unable to access 'https://github.com/ament/ament_cmake.git/': Could not resolve host: github.com
=== ./ament/ament_index (git) ===
Could not determine ref type of version: fatal: unable to access 'https://github.com/ament/ament_index.git/': Could not resolve host: github.com
=== ./ament/ament_lint (git) ===
Could not determine ref type of version: fatal: unable to access 'https://github.com/ament/ament_lint.git/': Could not resolve host: github.com
=== ./ament/ament_package (git) ===
Could not determine ref type of version: fatal: unable to access 'https://github.com/ament/ament_package.git/': Could not resolve host: github.com
=== ./ament/googletest (git) ===
Could not determine ref type of version: fatal: unable to access 'https://github.com/ament/googletest.git/': Could not resolve host: github.com
=== ./ament/uncrustify_vendor (git) ===
Could not determine ref type of version: fatal: unable to access 'https://github.com/ament/uncrustify_vendor.git/': Could not resolve host: github.com
=== ./ros2/ament_cmake_ros (git) ===
Could not determine ref type of version: fatal: unable to access 'https://github.com/ros2/ament_cmake_ros.git/': Could not resolve host: github.com
`

#### Expected behavior
I expected the create firmware to finish the install, download the libraries

#### Actual behavior
Step finishes, but appears to be missing important libraries.  It is able to clone some libraries, but not all.
`=== ./ros2/common_interfaces (git) ===
Cloning into '.'...
=== ./ros2/example_interfaces (git) ===
Cloning into '.'...
=== ./ros2/libyaml_vendor (git) ===
Cloning into '.'...
=== ./ros2/rcl (git) ===
`

#### Additional information

wegunterjr, 2021-03-04:
```Could not determine ref type of version: fatal:```

pablogs9, 2021-03-05:
This seems to be a network connection. Have you tried on another computer or with another internet connection?

wegunterjr, 2021-03-08:
yeah, that was it. I changed networks, and no issue.  Weird.   Thanks!!!

closed 2021-03-08 by wegunterjr