micro-ROS/micro_ros_setup #289: Examples for MultiArrays?

ottensmeyer, 2021-03-08:
## Issue template

- Hardware description: OpenCR using Arduino IDE
- RTOS: n/a
- Installation type: 
- Version or commit hash: foxy?

#### Steps to reproduce the issue

The question: 
Where can I find examples (or actual code) for publishing MultiArrays (of any data type) in similar format to the publisher, subscriber and other standard examples?  If there aren't any, could one be prepared for inclusion with the other examples?

The background:
I'm trying to generate an Int32MultiArray message to send a long series of integers from an OpenCR to ros2-web-bridge for display on an html-based UI.  (Later will do the same with a float array, but the problem's the same).

I've gone through the publisher example, can get different single scalar integers to show up properly (e.g. 32 or 64-bit), but I'm a bit baffled in trying to set up a MultiArray.  .create() functions appear to provide for MultiArray myArray, myArray.layout, and myArray.layout.dim, looks like malloc is required for the myArray.layout.dim.data.label.data.   myArray.data has an .init(), as does the lower level myArray.data.data.  I'm apparently getting something wrong, as my timer_callback() hangs while publishing.

I know the general advice is to create a custom msg by assembling the available ROS message types (rather than generic arrays), however I'd still want to have an array(s) within the custom message rather than labeling each integer individually (e.g. .x, .y, .z, .a, .b, ...) for convenience of indexing through for-loops at both the publisher and subscriber ends.

I've looked at the tf_publisher example, found https://github.com/micro-ROS/micro-ROS-demos/blob/foxy/rclc/complex_msg_publisher/main.c, which got me part way.

I've found limited MultiArray examples/code implemented for rosserial that have slightly different formats from the micro-ros versions, so they're of limited help.

I could use, for example, use an AccelWithCovariance messageType, but then I'd have to use ".accel" and ".covariance" in my variable names, which would violate the paradigm of meaningfully naming message elements.

thanks much!

#### Expected behavior

#### Actual behavior

#### Additional information

pablogs9, 2021-03-08:
You can check the structure of the datatype [in the headers of micro-ROS](https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/src/example_interfaces/msg/detail/int32_multi_array__struct.h). So the only thing that you have to care about is that, in micro-ROS, **sequences should always be initialized** by the user.

So for example:

```c
#include <example_interfaces/msg/int32_multiarray.h>

...

rclc_publisher_init_default(
&publisher,
&node,
ROSIDL_GET_MSG_TYPE_SUPPORT(example_interfaces, msg, Int32MultiArray),
"my_pub"));

...

example_interfaces__msg__Int32MultiArray msg;

#define DATA_SIZE 100

// Alloc the data
msg.data.data = (uint32_t *) malloc(DATA_SIZE * sizeof(uint32_t) );
msg.data.capacity = DATA_SIZE;
msg.data.size = 0;

// Fill the data
for(size_t i = 0; i < DATA_SIZE; i++){
    msg.data.data[i] = i;
    msg.data.size++;
}

// Alloc the layout with 3 dimensions
msg.layout.data_offset = ????;
msg.layout.dim.data = (example_interfaces__msg__MultiArrayDimension *) malloc(3 * sizeof(example_interfaces__msg__MultiArrayDimension));
msg.layout.dim.size = 3;
msg.layout.dim.capacity = 3;

// Fill each dimension

msg.layout.dim.data[0].size = ????;
msg.layout.dim.data[0].stride = ????;
msg.layout.dim.data[0].label.data = "my_label";
msg.layout.dim.data[0].label.size = strlen(msg.layout.dim.data[0].label.data);
msg.layout.dim.data[0].label.capacity = msg.layout.dim.data[0].label.size;

...
```

Let me know if this works for you.

pablogs9, 2021-03-10:
> One note - for this example, the malloc of the data for the Int32MultiArray didn't work until I changed it to:
> msg.data.data = (int32_t *) malloc(DATA_SIZE * sizeof(int32_t) );
> I assume that the malloc needs to use the same data type as the underlying elements (Int32 for this, Float32 for a Float32MultiArray, etc.)?

You are right, just a mistake on my side.

> Follow up questions:
> What is the purpose of the .init, .fini, .create and .destroy functions in the XXX_function.h files if one needs to set up the structures by hand? I've found the compiled libmicroros.a files, but I haven't figured out where to dig deeper to see what's under the hood.

These functions are supposed to do the same as we are doing here manually but I have never tested them... The point is that these functions will use malloc/free by default and using the method I exposed you are also able to set the sequences to static memory.

> Are the values here arbitrary (as long as they're >0)?
> msg.layout.dim.data[0].size = ????;
> msg.layout.dim.data[0].stride = ????;
> Given that the layout.dim.data structure only holds a string describing the given dimension of the array, when would these size and stride values need to be >1?

I just put `????` because I don't know the data that you want to set in these variables.

ottensmeyer, 2021-03-11:
Much appreciated - no longer stuck on this - thanks for the help!

closed 2021-03-11 by ottensmeyer