micro-ROS/micro_ros_setup #290: about error of std::bad_alloc in Raspberry Pi 4

ncnynl, 2021-03-10:
I have some strange errors. I run the agent on the Raspberry Pi 4B/2G memory. It is normal to run the cartographer on the PC, but when running navigation2, the agent on the Raspberry Pi will report an error and appear: what(): std:: bad_alloc, this seems to be insufficient memory. I run the agent in NUC/8G memory, whether it is cartographer or navigation2, the agent is ok.

What I am more curious about is why navigation2 affects the operation of the agent.

pablogs9, 2021-03-10:
This is a known issue: https://github.com/micro-ROS/micro_ros_arduino/issues/40#issuecomment-779860217

It is related to the undelaying Fast DDS middleware in the Agent. Could you please detail which version (foxy, rolling...) of the agent are you using and your installation (docker, from sources, as ROS 2 package...).

ncnynl, 2021-03-10:
The current problem is that the agent is running on the Raspberry Pi, but it does not appear on the nuc.

Development board:
Based on the self-developed stm32f407 development board, the micro-ros firmware tested in nuc is normal. It is used to control the car and can create maps and navigate.

nuc:

The ros2 version is foxy, installed through apt.
The version of the system is ubunut 20.04 mate
The code of the micro-ros workspace, compiled from source code, foxy version, not the latest version
Micro-XRCE-DDS-Client. git log shows as: commit 3286eb version
cartographer and navigation2, installed via apt
middleware is the default, it should be fastDDS
nuc connects to the development board, and can complete mapping and navigation.

raspberry pie: 

The ros2 version is foxy, installed through apt.
The version of the system is ubunut 20.04 mate
The code of the micro-ros workspace is copied through the code on the nuc, after deleting the build and install directories, recompile
cartographer and navigation2, installed via apt
middleware is the default, it should be fastDDS
Raspberry Pi connects to the development board, and realizes mapping and navigation through remote pc (nuc or mini pc for development), and it will appear if you run the navigation. You can't make mistakes when you build a map.

The current guess is that the source code of the Raspberry Pi may be copied from nuc, which is not suitable for the Raspberry Pi. Compilation will cause a conflict between need by libfastrtps.so.2 and libfastrtps.so.2.0.2
I will try to install the latest version of micro-ros on the Raspberry Pi. Specify the corresponding version respectively to see if there is a problem with the version of libfastrtps.so.

ncnynl, 2021-03-11:
I check difference nuc and rpi ：

nuc：
git clone https://github.com/eProsima/Micro-XRCE-DDS.git
commit 272c5ad91ea931b007fa31472a90421db4e346aa (HEAD, tag: v1.3.0)  

/opt/ros/foxy/lib/:
libfastrtps.so.2.0.2 ，but i  cp  lib/libfastrtps.so.2.0.0 to here ，replace libfastrtps.so.2.0.2 
Because compile miscros_ws will prompt you need libfastrtps.so.2 and the current libfastrtps.so.2.0.2 may conflict

/usr/local/lib:
lib/libfastrtps.so.2.0.0

I compile firmware in nuc 

raspberry pi:  

git clone https://github.com/eProsima/Micro-XRCE-DDS.git
master version

/opt/ros/foxy/lib/:
libfastrtps.so.2.0.2  

/usr/local/lib:
lib/libfastrtps.so.2.0.2 

I have a try two ways:

One:  libfastrtps.so.2.0.2  change into lib/libfastrtps.so.2.0.0 in rpi 

Two: nuc update  to newest version . libfastrtps.so.2.0.2 and  flash new firmware

ncnynl, 2021-03-15:
@pablogs9  I re-tested the Raspberry Pi 4, I increased the swap size to 4G, and everything is normal in the test.

ncnynl, 2021-03-16:
I found that after running the agent, if I run a few more launch files, errors will be reported. About 6, 7 launches

pablogs9, 2021-03-16:
Could you try to comment all calls to `add_middleware_callback` in [this file](https://github.com/micro-ROS/micro-ROS-Agent/blob/0220dd6febd616f14234f05522d1b76d938595ec/micro_ros_agent/src/agent/Agent.cpp#L157) and rebuild it?

Let me know if the error is solved by doing this.

ncnynl, 2021-03-19:
Hello @pablogs9  

this is a good news.

I did not comment this code, but I reinstalled all the latest code yesterday, and it seems to be stable without errors.

step:

1. Remove ros2 installed by apt and re-use ros2 compiled from source code
2. Follow the process of first_application_rtos/freertos/ to recompile all micro-ROS packages
3. Compile the package developed by yourself
4. RPI4 test starts the agent and starts the radar
5. NUC launches odom release and navigation package

This error does not reappear at this time.
I re-run more than 20 nodes without any errors.

I also tested it on another x86 industrial computer. There is no error at the same time.

ncnynl, 2021-03-19:
I upgraded the firmware code at the same time, using the rmw_uros_ping_agent function and microros_transports.c, and it has been running for 5 hours without any problems.

pablogs9, 2021-03-22:
Cool, so can we close the issue by now?

closed 2021-03-22 by ncnynl