micro-ROS/micro_ros_setup #291: Build Failure

eden-desta, 2021-03-16:
Tried to peer at the other issues it appears you might have a release build going on. But i noticed this on my end trying to rebuild my env:

```
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup

rosdep update
rosdep install --from-path src --ignore-src -y
colcon build
```

```
Starting >>> micro_ros_setup
Starting >>> microrosapp
--- stderr: microrosapp                                             
CMake Error at CMakeLists.txt:11 (include):
  include could not find load file:

    /cmake/app/boilerplate.cmake

CMake Error at CMakeLists.txt:20 (if):
  if given arguments:

    "STREQUAL" "serial"

  Unknown arguments specified

---
Failed   <<< microrosapp [0.49s, exited with code 1]
Aborted  <<< micro_ros_setup [0.91s]                           

Summary: 0 packages finished [1.12s]
  1 package failed: microrosapp
  1 package aborted: micro_ros_setup
  1 package had stderr output: microrosapp

```

pablogs9, 2021-03-16:
Hello @eden-desta!

Could you please provide some steps for replicating this?
Which `micro_ros_setup` commands did you run?

eden-desta, 2021-03-16:
@pablogs9 
This is how far i got:

```
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup

rosdep update
rosdep install --from-path src --ignore-src -y
colcon build
```

pablogs9, 2021-03-16:
I haven't been able to replicate it in a clean ROS 2 Foxy docker:

https://asciinema.org/a/PrMMMq45NJieh8PScYWGHLw8L

Have you tried to use a clean environment?

eden-desta, 2021-03-16:
I dont use a docker, how would i go about cleaning the environment?

I notice that my rosdep install only outputs one line. 
```
#All required rosdeps installed successfully
```

pablogs9, 2021-03-16:
Could you paste here the output of a `tree` command inside your micro-ROS workspace?

eden-desta, 2021-03-16:
```
.
├── buildSystem.sh
└── uros_apps
    └── apps
        └── custom_app
            ├── app-colcon.meta
            ├── CMakeLists.txt
            ├── host-udp.conf
            ├── inc
            │   ├── heartbeat.h
            │   └── sense.h
            ├── libraries
            │   └── BNO055_driver
            │       ├── bno055.c
            │       ├── bno055.h
            │       ├── bno055_support.c
            │       ├── LICENSE
            │       └── README.md
            ├── serial.conf
            ├── serial-usb.conf
            └── src
                ├── heartbeat.c
                ├── main.c
                └── sense.c

7 directories, 16 files
```

eden-desta, 2021-03-16:
With git clone:
```
.
├── buildSystem.sh
├── src
│   └── micro_ros_setup
│       ├── 3rd-party-licenses.txt
│       ├── CMakeLists.txt
│       ├── config
│       │   ├── agent_ros2_packages.txt
│       │   ├── agent_uros_packages.repos
│       │   ├── client_ros2_packages.txt
│       │   ├── client_uros_packages.repos
│       │   ├── freertos
│       │   │   ├── crazyflie21
│       │   │   │   ├── board.repos
│       │   │   │   ├── build.sh
│       │   │   │   ├── client-colcon.meta
│       │   │   │   ├── client_uros_packages.repos -> ../../client_uros_packages.repos
│       │   │   │   ├── configure.sh
│       │   │   │   ├── create.sh
│       │   │   │   ├── flash.sh
│       │   │   │   └── package.xml
│       │   │   ├── dev_ros2_packages.txt
│       │   │   ├── dev_uros_packages.repos
│       │   │   ├── esp32
│       │   │   │   ├── board.repos
│       │   │   │   ├── build.sh
│       │   │   │   ├── client-colcon.meta
│       │   │   │   ├── client_uros_packages.repos -> ../../client_uros_packages.repos
│       │   │   │   ├── configure.sh
│       │   │   │   ├── create.sh
│       │   │   │   ├── flash.sh
│       │   │   │   ├── package.xml
│       │   │   │   └── README.md
│       │   │   ├── list_apps.sh
│       │   │   ├── nucleo_f446re
│       │   │   │   ├── board.repos
│       │   │   │   ├── build.sh
│       │   │   │   ├── client-colcon.meta
│       │   │   │   ├── client_uros_packages.repos -> ../../client_uros_packages.repos
│       │   │   │   ├── configure.sh
│       │   │   │   ├── create.sh
│       │   │   │   ├── flash.sh
│       │   │   │   └── package.xml
│       │   │   ├── nucleo_f446ze
│       │   │   │   ├── board.repos
│       │   │   │   ├── build.sh
│       │   │   │   ├── client-colcon.meta
│       │   │   │   ├── client_uros_packages.repos -> ../../client_uros_packages.repos
│       │   │   │   ├── configure.sh
│       │   │   │   ├── create.sh
│       │   │   │   ├── flash.sh
│       │   │   │   └── package.xml
│       │   │   ├── nucleo_f746zg
│       │   │   │   ├── board.repos
│       │   │   │   ├── build.sh
│       │   │   │   ├── client-colcon.meta
│       │   │   │   ├── client_uros_packages.repos -> ../../client_uros_packages.repos
│       │   │   │   ├── configure.sh
│       │   │   │   ├── create.sh
│       │   │   │   ├── flash.sh
│       │   │   │   └── package.xml
│       │   │   ├── nucleo_f767zi
│       │   │   │   ├── board.repos
│       │   │   │   ├── build.sh
│       │   │   │   ├── client-colcon.meta
│       │   │   │   ├── client_uros_packages.repos -> ../../client_uros_packages.repos
│       │   │   │   ├── configure.sh
│       │   │   │   ├── create.sh
[... 87 lines cut ...]
└── uros_apps
    └── apps
        └── custom_app
            ├── app-colcon.meta
            ├── CMakeLists.txt
            ├── host-udp.conf
            ├── inc
            │   ├── heartbeat.h
            │   └── sense.h
            ├── libraries
            │   └── BNO055_driver
            │       ├── bno055.c
            │       ├── bno055.h
            │       ├── bno055_support.c
            │       ├── LICENSE
            │       └── README.md
            ├── serial.conf
            ├── serial-usb.conf
            └── src
                ├── heartbeat.c
                ├── main.c
                └── sense.c

29 directories, 147 files
```

pablogs9, 2021-03-16:
Ok, just create an empty file called `COLCON_IGNORE` inside your `uros_apps`. Colcon is trying to build your micro-ROS app...

Let me know if this works.

pablogs9, 2021-03-17:
I'm closing, please reopen if the issue has not been solved.

closed 2021-03-17 by pablogs9