micro-ROS/micro_ros_setup #293: Running entire uROS build on ARM PC

eden-desta, 2021-03-22:
Hey so this is a very specific case, and perfectly okay if it isnt something that we have the capability to support right now. 

But I am running a Linux system on an Arm chip (so the Mac M1 but this can translate over for a PI very easily). And on the step to create the firmware I am having a failure with a specific library: gcc-multilib. Did some digging and thats just not something that is ported over to ARM Linux which i found very surprising. 

Have you all seen this problem before? Or have a work around it? I imagine if I am dedicated i could potentially build it from source and see where it takes me but figured I would ask.

pablogs9, 2021-03-23:
Well, AFAIK there is a port for Raspbian, you can check the support scripts [here](https://github.com/micro-ROS/micro_ros_setup/tree/foxy/config/raspbian), and the apps [here](https://github.com/micro-ROS/raspbian_apps). At some point, we cross-compile both the Agent and some Client demos for a Raspberry Pi.

Is this useful for you? Maybe we do have `gcc-multilib` for Raspbian... I'm not sure. Another thought is: why do we need `gcc-multilib` which is a tool for cross-compiling if you are building in the host for the host. Are you aware of at which specific step is this library is installed? Maybe we can try to remove this dependency for `host`...

eden-desta, 2021-03-23:
Well kind of. My failure occurs on creating the firmware (using zephyr) in my specific case. Which I guess makes sense if I built it using Raspbian it would not occur. 
The gcc-multilib is attempted to be installed during the step you call the bash script to create the firmware, i can further identify at which point it is called as well. 

But thank you for the feedback I appreciate this, I will do some experimenting with the Raspbian side as well.

closed 2022-01-28 by pablogs9