micro-ROS/micro_ros_setup #295: ping_pong/int32_publisher,FreeRTOS,Olimex STM32-E407 stops sending UDP msgs

gahl-pv, 2021-03-24:
## ping_pong/int32_publisher,FreeRTOS,Olimex STM32-E407 stops sending UDP msgs

- Hardware description: Olimex STM32-E407
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
Using the [first app tutorial](https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/) and changing from serial to UDP:
```bash
git clone -b foxy https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
colcon build
source install/local_setup.bash
ros2 run micro_ros_setup create_firmware_ws.sh freertos olimex-stm32-e407
ros2 run micro_ros_setup configure_firmware.sh ping_pong -t udp -i 192.168.1.33 -p 8888
ros2 run micro_ros_setup build_firmware.sh
ros2 run micro_ros_setup flash_firmware.sh
# Agent
ros2 run micro_ros_setup create_agent_ws.sh
ros2 run micro_ros_setup build_agent.sh
source install/local_setup.bash
ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6
# Listener/Subscriber
ros2 topic echo /microROS/ping
```

#### Expected behavior
Seeing the messages being published regularly

#### Actual behavior
For `ping_pong` no message is showing up on the topic. Then trying with the simpler example app `int32_publisher`, one message shows up when subscribing to the topic:
```
data: 0
---
```
The output from the agent is:
```
[1616594008.984403] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1616594008.984643] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1616594009.721807] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 11 9D 96 7C 81 00 FC 01
[1616594009.721976] info     | Root.cpp           | create_client            | create                 | client_key: 0x119D967C, session_id: 0x81
[1616594009.722045] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x295540348, address: 192.168.1.14:448
[1616594009.722119] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1616594009.722608] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x119D967C, len: 112, data: 
0000: 81 80 00 00 01 05 68 00 00 0A 00 01 01 02 00 00 59 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 66 72 65 65 72 74 6F 73 5F 69 6E 74 33 32
0040: 5F 70 75 62 6C 69 73 68 65 72 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63
0060: 69 70 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 00 00
[1616594009.723727] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x119D967C, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1616594009.725413] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x119D967C, participant_id: 0x000(1)
[1616594009.725510] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1616594009.725533] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1616594009.725545] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1616594009.725810] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x119D967C, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1616594009.725905] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1616594009.726126] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x119D967C, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1616594009.726494] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x119D967C, len: 140, data: 
0000: 81 80 01 00 01 05 81 00 00 0B 00 02 02 02 00 00 73 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 72 74 2F 66 72 65 65 72 74 6F 73 5F 69 6E 74 33 32 5F 70 75 62 6C 69 73 68 65
0040: 72 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61 54 79 70 65 3E 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A
0060: 3A 64 64 73 5F 3A 3A 49 6E 74 33 32 5F 3C 2F 64 61 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E
[... 30 lines cut ...]
[1616594009.728644] debug    | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x119D967C, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[1616594009.728671] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 14, data: 
0000: 81 80 03 00 05 01 06 00 00 0D 00 05 00 00
[1616594009.728683] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1616594009.728731] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x119D967C, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 03 00 03 00 80
[1616594009.728770] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1616594009.729143] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x119D967C, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1616594009.735076] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x119D967C, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 03 00 03 00 80
[1616594009.735120] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1616594010.737241] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x119D967C, len: 16, data: 
0000: 81 80 04 00 07 01 08 00 00 0E 00 05 00 00 00 00
[1616594010.737343] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 4, data: 
0000: 00 00 00 00
[1616594010.737425] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
```

#### Additional Information
Removing the `timer_callback()` and publishing a message directly in the while loop instead, makes it keep sending messages for a long time, but eventually that stops as well.

pablogs9, 2021-03-24:
How much time does it take to stop publishing if you don't use the `timer_callback`?

gahl-pv, 2021-03-24:
From 340 seconds and up. It is different each time. Publishing one message every 0.1s and incrementing the number by 1 each new message, the last three trials stopped at numbers `3773`, `4995`, and `3423`. But it has also kept going above `8000` before I stopped it manually.

pablogs9, 2021-03-25:
I have tested this and increasing the stack size of the micro-ROS app solves the problem. I'm going to do a long term run to check if this not fails after some time.

Just change [this line](https://github.com/micro-ROS/freertos_apps/blob/74cd38c0a283e66741881d62b5f0967640927179/microros_olimex_e407_extensions/Src/main.c#L464) to:

```c
attributes.stack_size = 4*6000;
```

Let me know if this solves your issue.

pablogs9, 2021-03-25:
This maybe is related: https://github.com/micro-ROS/freertos_apps/issues/40

pablogs9, 2021-03-25:
I have updated the Olimex project with these changes: https://github.com/micro-ROS/freertos_apps/pull/66

Please retry and let us know if you have any other problem.

Thanks for the feedback!

gahl-pv, 2021-03-25:
Great, that seems to have fixed a lot. I have retried the ping_pong app with all the new changes and the messages are now being published. However, after a while they stop again. 
One time it appears to have happened when sending a fake_ping as shown in the tutorial:
```
stamp:
  sec: 491
  nanosec: 722000000
frame_id: '1568612021_594956306'
---
stamp:
  sec: 493
  nanosec: 726000000
frame_id: '150751668_594956306'
---
stamp:
  sec: 0
  nanosec: 0
frame_id: fake_ping
---
```
And another time it stopped by itself, although it did still work sending fake_ping. Then after a little while four new messages were published, and then it stopped again:
```
stamp:
  sec: 449
  nanosec: 912000000
frame_id: '1047833678_1210349306'
---
stamp:
  sec: 0
  nanosec: 0
frame_id: fake_ping
---
stamp:
  sec: 0
  nanosec: 0
frame_id: fake_ping
---
stamp:
  sec: 0
  nanosec: 0
frame_id: fake_ping
---
stamp:
  sec: 585
  nanosec: 20000000
frame_id: '823097781_1210349306'
---
stamp:
  sec: 586
  nanosec: 33000000
frame_id: '309129892_1210349306'
---
stamp:
  sec: 587
  nanosec: 906000000
frame_id: '386601848_1210349306'
---
stamp:
  sec: 589
  nanosec: 904000000
frame_id: '898606002_1210349306'
---
```

pablogs9, 2021-03-25:
Let me test that

pablogs9, 2021-03-29:
I have been replicating this and it seems to be stuck in some inner Ethernet layer, IMO this is not a micro-ROS problem but a configuration error in the LWIP layer...

Do you have any idea on how to debug this networking layer?

gahl-pv, 2021-03-29:
No, I've not had much experience with the networking stack, not quite sure where to start.

I was hoping to avoid getting too low-level. Do you reckon it would be easier to correct the configuration error in the LWIP layer, or is it quicker to get UDP messages working with one of the other two RTOSes instead?

vishnuvijay95, 2021-04-07:
> ## ping_pong/int32_publisher,FreeRTOS,Olimex STM32-E407 stops sending UDP msgs
> 
>     * Hardware description: Olimex STM32-E407
> 
>     * RTOS: FreeRTOS
> 
>     * Installation type: micro_ros_setup
> 
>     * Version or commit hash: foxy
> 
> 
> #### Steps to reproduce the issue
> 
> Using the [first app tutorial](https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/) and changing from serial to UDP:
> 
> ```shell
> git clone -b foxy https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
> colcon build
> source install/local_setup.bash
> ros2 run micro_ros_setup create_firmware_ws.sh freertos olimex-stm32-e407
> ros2 run micro_ros_setup configure_firmware.sh ping_pong -t udp -i 192.168.1.33 -p 8888
> ros2 run micro_ros_setup build_firmware.sh
> ros2 run micro_ros_setup flash_firmware.sh
> # Agent
> ros2 run micro_ros_setup create_agent_ws.sh
> ros2 run micro_ros_setup build_agent.sh
> source install/local_setup.bash
> ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6
> # Listener/Subscriber
> ros2 topic echo /microROS/ping
> ```
> 
> #### Expected behavior
> 
> Seeing the messages being published regularly
> #### Actual behavior
> 
> For `ping_pong` no message is showing up on the topic. Then trying with the simpler example app `int32_publisher`, one message shows up when subscribing to the topic:
> 
> ```
> data: 0
> ---
> ```
> 
> The output from the agent is:
> 
> ```
> [1616594008.984403] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
> [1616594008.984643] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
> [1616594009.721807] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
> 0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 11 9D 96 7C 81 00 FC 01
> [1616594009.721976] info     | Root.cpp           | create_client            | create                 | client_key: 0x119D967C, session_id: 0x81
> [1616594009.722045] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x295540348, address: 192.168.1.14:448
> [1616594009.722119] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 19, data: 
> 0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
> [1616594009.722608] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x119D967C, len: 112, data: 
> 0000: 81 80 00 00 01 05 68 00 00 0A 00 01 01 02 00 00 59 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
> 0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 66 72 65 65 72 74 6F 73 5F 69 6E 74 33 32
> 0040: 5F 70 75 62 6C 69 73 68 65 72 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63
> 0060: 69 70 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 00 00
> [1616594009.723727] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x119D967C, len: 13, data: 
> 0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
> [1616594009.725413] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x119D967C, participant_id: 0x000(1)
> [1616594009.725510] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 14, data: 
> 0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
> [1616594009.725533] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 13, data: 
> 0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
> [1616594009.725545] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 13, data: 
> 0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
> [1616594009.725810] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x119D967C, len: 13, data: 
[... 43 lines cut ...]
> [1616594009.728683] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 13, data: 
> 0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
> [1616594009.728731] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x119D967C, len: 13, data: 
> 0000: 81 00 00 00 0B 01 05 00 03 00 03 00 80
> [1616594009.728770] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 13, data: 
> 0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
> [1616594009.729143] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x119D967C, len: 13, data: 
> 0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
> [1616594009.735076] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x119D967C, len: 13, data: 
> 0000: 81 00 00 00 0B 01 05 00 03 00 03 00 80
> [1616594009.735120] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 13, data: 
> 0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
> [1616594010.737241] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x119D967C, len: 16, data: 
> 0000: 81 80 04 00 07 01 08 00 00 0E 00 05 00 00 00 00
[... cut ...]
> [1616594010.737425] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x119D967C, len: 13, data: 
> 0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
> ```
> 
> #### Additional Information
> 
> Removing the `timer_callback()` and publishing a message directly in the while loop instead, makes it keep sending messages for a long time, but eventually that stops as well.

since the Olimex board don't have a in-build wifi option, can you elaborate how you have done the hardware configuration. i am trying to replicate this in my project.

pablogs9, 2022-03-07:
@guilyx please open another issue with an explanation of your use case and reproducible code

ZhenshengLee, 2022-03-14:
@pablogs9 accoding to https://github.com/micro-ROS/freertos_apps/issues/96#issuecomment-1065019133
This can be closed.

closed 2022-03-14 by pablogs9