micro-ROS/micro_ros_setup #297: Subscription callback not working properly

vishnuvijay95, 2021-04-05:
## Issue template

    Hardware description: Olimex STM32-E407
    RTOS: FreeRTOS
    Installation type: micro_ros_setup
    Version or commit hash: foxy

#### Steps to reproduce the issue
1. Connecting the microROS agent using serial communication.
2. Subscribing to geometry_msgs/msg/Twist with the microROS subscriber node code i am posting here.
```
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include "FreeRTOS.h"
#include "task.h"
#include <rmw_uros/options.h>
#include <geometry_msgs/msg/twist.h>
#include <std_msgs/msg/int32.h>

#include <stdio.h>

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc);vTaskDelete(NULL);}}

#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

rcl_subscription_t cmd_vel_sub;
geometry_msgs__msg__Twist cmd_vel;

void subscription_callback(const void *msgin)
{
	const geometry_msgs__msg__Twist *msg = (const geometry_msgs__msg__Twist *)msgin;
	printf("Received: %f\n", msg->linear.x);
}

void appMain(void * arg)
{
  	rcl_allocator_t allocator = rcl_get_default_allocator();
	rclc_support_t support;

	// create init_options
	RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

	// create node
	rcl_node_t node=rcl_get_zero_initialized_node();
	RCCHECK(rclc_node_init_default(&node, "MicroROS_kobuki", "", &support));

	// create subscriber
	RCCHECK(rclc_subscription_init_best_effort(
		&cmd_vel_sub,
		&node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(geometry_msgs, msg, Twist),
		"cmd_vel"));

	// create executor
	rclc_executor_t executor;
	RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
	RCCHECK(rclc_executor_add_subscription(&executor, &cmd_vel_sub, &cmd_vel, &subscription_callback, ON_NEW_DATA));

	while(1){
	   
			rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
			usleep(100000);
	}

	// free resources
	RCCHECK(rcl_subscription_fini(&cmd_vel_sub, &node));
	RCCHECK(rcl_node_fini(&node));
	
	vTaskDelete(NULL);
}
```
3. Trying to print the received twist message through UART6
4. Publishing data to the topic using command
`ros2 topic pub /cmd_vel geometry_msgs/msg/Twist "{linear: {x: 2.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 0.0}}"`
[image]

#### Expected behavior
Getting the received twist message in the serial terminal.

#### Actual behavior
Data not getting displayed
[image]

#### Additional information

vishnuvijay95, 2021-04-05:
The microROS app is always in block state
[image]

pablogs9, 2021-04-06:
Could you share the micro-ROS agent output using the flag `-v6`?

pablogs9, 2021-04-06:
Well in your second screenshot I see that the `printf` is outputting some data have you resolved the issue of not seeing the received topic?

vishnuvijay95, 2021-04-06:
But its displaying some junk value, i am providing linear.x=2.0 its displaying 46

vishnuvijay95, 2021-04-06:
> Could you share the micro-ROS agent output using the flag `-v6`?

sure

vishnuvijay95, 2021-04-06:
> > Could you share the micro-ROS agent output using the flag `-v6`?
> 
> sure

the output of -v6 flag is as follows
```
[1617698144.291409] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1617698144.292092] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1617698208.650108] info     | Root.cpp           | create_client            | create                 | client_key: 0x041835D4, session_id: 0x81
[1617698208.650244] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x68695508, address: 0
[1617698208.650430] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1617698208.666049] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 104, data: 
0000: 81 80 00 00 01 05 5E 00 00 0A 00 01 01 02 00 00 50 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 4D 69 63 72 6F 52 4F 53 5F 6B 6F 62 75 6B
0040: 69 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70 61 6E 74 3E 3C 2F 64
0060: 64 73 3E 00 00 00 00 00
[1617698208.667100] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1617698208.672121] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1617698208.672826] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x041835D4, participant_id: 0x000(1)
[1617698208.672982] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1617698208.673003] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1617698208.673015] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1617698208.673025] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1617698208.676229] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1617698208.676437] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1617698208.678086] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1617698208.691120] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 128, data: 
0000: 81 80 01 00 01 05 75 00 00 0B 00 02 02 02 00 00 67 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 72 74 2F 63 6D 64 5F 76 65 6C 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61 54 79 70 65
0040: 3E 67 65 6F 6D 65 74 72 79 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 54 77 69 73 74
0060: 5F 3C 2F 64 61 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73 3E 00 00 01 00 00 00
[1617698208.691600] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x041835D4, topic_id: 0x000(2), participant_id: 0x000(1)
[1617698208.691740] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1617698208.691782] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1617698208.694116] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 01 00 80
[1617698208.694261] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1617698208.697173] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1617698208.700112] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 24, data: 
0000: 81 80 02 00 01 05 0F 00 00 0C 00 04 04 02 00 00 01 00 00 00 00 00 01 00
[1617698208.700439] debug    | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x041835D4, subscriber_id: 0x000(4), participant_id: 0x000(1)
[1617698208.700578] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 14, data: 
0000: 81 80 02 00 05 01 06 00 00 0C 00 04 00 00
[... cut ...]
[1617698271.018470] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000000, len: 48, data: 
0000: 00 00 00 00 00 00 00 40 00 00 00 00 00 00 00 40 00 00 00 00 00 00 00 40 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40
[1617698271.018584] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 60, data: 
0000: 81 01 28 00 09 01 34 00 00 0E 00 06 00 00 00 00 00 00 00 40 00 00 00 00 00 00 00 40 00 00 00 00
0020: 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40
```

vishnuvijay95, 2021-04-06:
The topic is getting subscribed and the call back function is working, but the transmitted value is not available
[image]
 at the output

pablogs9, 2021-04-06:
Hello this is not a micro-ROS problem, just that this FreeRTOS port does not provide a `printf` implementation for float numbers.

In fact just try to do:

```c
printf("Received: %f\n", 42.2);
```

and you will see the same output.

To check that the micro-ROS node is correctly receiving the data I have done:

```c
void subscription_callback(const void *msgin)
{
	const geometry_msgs__msg__Twist *msg = (const geometry_msgs__msg__Twist *)msgin;

	int int_value = msg->linear.x;
	printf("Received: %d\n", int_value);
}
```

closed 2021-04-06 by pablogs9

vishnuvijay95, 2021-04-07:
Thanks I verified, you are correct

vishnuvijay95, 2021-04-07:
It seems floating point arithmetic, working properly in the board, only `printf `has this problem, have you verified?

pablogs9, 2021-04-07:
No, but floating-point should not have problems, just the minimal implementation of `printf`

vishnuvijay95, 2021-04-07:
Ok i checked, the floating point operations have no limitations as off my usage, its working perfect.

vishnuvijay95, 2021-04-07:
i need to communicate through microROS agent through UDP. can i use a WiFi modem and connect to Olimex board through Ethernet? if so can you elaborate the steps.

pablogs9, 2021-04-07:
https://github.com/micro-ROS/micro_ros_setup#configuring-micro-ros-firmware

use configure with `-t udp`

vishnuvijay95, 2021-04-07:
That i have went through, i have to know whether i can use a external wifi modem for communication with the olimex board.

vishnuvijay95, 2021-04-07:
have you done that type of configuration?

vishnuvijay95, 2021-04-07:
Hardware description: Olimex STM32-E407
RTOS: FreeRTOS
Installation type: micro_ros_setup
Version or commit hash: foxy

i tried with following commands ro establish UDP communication with microROS agent

```
ros2 run micro_ros_setup configure_firmware.sh ping_pong -t udp -i 192.168.0.104 -p 8888
ros2 run micro_ros_setup build_firmware.sh
ros2 run micro_ros_setup flash_firmware.sh
# Agent
ros2 run micro_ros_setup create_agent_ws.sh
ros2 run micro_ros_setup build_agent.sh
source install/local_setup.bash
ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6
```
[image]
connection is not getting established