micro-ROS/micro_ros_setup #298: Problem with compiling uros apps for OLIMEX

masoudir, 2021-04-07:
## Issue

- Hardware description: 
- RTOS: 
- Installation type: 
- Version or commit hash: 

#### Steps to reproduce the issue
I have executed these commands as you suggest for olimex ping_pong app:

    ros2 run micro_ros_setup create_firmware_ws.sh freertos olimex-stm32-e407
    ros2 run micro_ros_setup configure_firmware.sh ping_pong --transport serial
    ros2 run micro_ros_setup build_firmware.sh

But it gives me error on "rclc_lifecycle" package.

How can I fix that?

Also I have tested this for other apps such as "int32_publisher", but nothing has changed!!!!

#### Expected behavior

#### Actual behavior

        /home/a/ws_test/uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/src/rclc_lifecycle/rclc_lifecycle.c:47:23: error: too few arguments to function 'rcl_lifecycle_state_machine_init'
           rcl_ret_t rcl_ret = rcl_lifecycle_state_machine_init(
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        In file included from /home/a/ws_test/uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/include/rclc_lifecycle/rclc_lifecycle.h:24,
                         from /home/a/ws_test/uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/src/rclc_lifecycle/rclc_lifecycle.c:17:
        /home/a/ws_test/uros_ws/firmware/mcu_ws/install/rcl_lifecycle/include/rcl_lifecycle/rcl_lifecycle.h:237:1: note: declared here
         rcl_lifecycle_state_machine_init(
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        /home/a/ws_test/uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/src/rclc_lifecycle/rclc_lifecycle.c: In function 'rcl_lifecycle_node_fini':
        /home/a/ws_test/uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/src/rclc_lifecycle/rclc_lifecycle.c:239:13: error: too few arguments to function 'rcl_lifecycle_state_machine_fini'
           rcl_ret = rcl_lifecycle_state_machine_fini(
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        In file included from /home/a/ws_test/uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/include/rclc_lifecycle/rclc_lifecycle.h:24,
                         from /home/a/ws_test/uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/src/rclc_lifecycle/rclc_lifecycle.c:17:
        /home/a/ws_test/uros_ws/firmware/mcu_ws/install/rcl_lifecycle/include/rcl_lifecycle/rcl_lifecycle.h:272:1: note: declared here
         rcl_lifecycle_state_machine_fini(
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        make[3]: *** [CMakeFiles/rclc_lifecycle.dir/build.make:66: CMakeFiles/rclc_lifecycle.dir/src/rclc_lifecycle/rclc_lifecycle.c.obj] Error 1
        make[2]: *** [CMakeFiles/Makefile2:108: CMakeFiles/rclc_lifecycle.dir/all] Error 2
        make[1]: *** [Makefile:144: all] Error 2
        ---
        Failed   <<< rclc_lifecycle [2.87s, exited with code 2]
        Aborted  <<< rcl_action [4.50s]                                                                                                
        Aborted  <<< visualization_msgs [16.0s]                                                                                     
        Aborted  <<< sensor_msgs [31.2s]                                         
        
        Summary: 53 packages finished [1min 44s]
          1 package failed: rclc_lifecycle
          3 packages aborted: rcl_action sensor_msgs visualization_msgs
          35 packages had stderr output: action_msgs actionlib_msgs builtin_interfaces composition_interfaces diagnostic_msgs example_interfaces geometry_msgs libyaml_vendor lifecycle_msgs micro_ros_msgs nav_msgs rcl rcl_action rcl_interfaces rcl_lifecycle rcl_logging_noop rclc rclc_lifecycle rcutils rmw rmw_implementation rmw_microxrcedds rosgraph_msgs rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_microxrcedds_c sensor_msgs shape_msgs statistics_msgs std_msgs std_srvs test_msgs trajectory_msgs unique_identifier_msgs visualization_msgs
          3 packages not processed
        make: *** [Makefile:334: colcon_compile] Error 2

#### Additional information

pablogs9, 2021-04-07:
Related: https://github.com/ros2/rclc/issues/65

Can you test this in a clean micro-ROS environment?

masoudir, 2021-04-07:
Thanks. That worked.

closed 2021-04-07 by masoudir