micro-ROS/micro_ros_setup #299: Unable to establish UDP connection with uros agent

vishnuvijay95, 2021-04-07:
Hardware description: Olimex STM32-E407
RTOS: FreeRTOS
Installation type: micro_ros_setup
Version or commit hash: foxy

i tried with following commands ro establish UDP communication with microROS agent

```
ros2 run micro_ros_setup configure_firmware.sh ping_pong -t udp -i 192.168.0.104 -p 8888
ros2 run micro_ros_setup build_firmware.sh
ros2 run micro_ros_setup flash_firmware.sh
# Agent
ros2 run micro_ros_setup create_agent_ws.sh
ros2 run micro_ros_setup build_agent.sh
source install/local_setup.bash
ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6
```
[image]
connection is not getting established

pablogs9, 2021-04-07:
Can you share the serial console of the Olimex (UART3)? It should provide some detailed information about the networking status.

vishnuvijay95, 2021-04-07:
Now connection established, i had rebuild the microROS_agent ws.
[image]

vishnuvijay95, 2021-04-07:
but the connection is not establishing on the go. i need to press on board  reset button 2 or more times. i had faced same issue with serial communication as well. why its happening?

vishnuvijay95, 2021-04-07:
but the topic is not echoing anything now.
[image]

vishnuvijay95, 2021-04-07:
Agent output
```
[1617806645.856370] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1617806645.856656] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1617806685.925286] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 15 B5 CC 50 81 00 FC 01
[1617806685.925428] info     | Root.cpp           | create_client            | create                 | client_key: 0x15B5CC50, session_id: 0x81
[1617806685.925477] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x364235856, address: 192.168.0.104:448
[1617806685.925564] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x15B5CC50, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1617806685.927872] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x15B5CC50, len: 100, data: 
0000: 81 80 00 00 01 05 5C 00 00 0A 00 01 01 02 00 00 4E 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 70 69 6E 67 70 6F 6E 67 5F 6E 6F 64 65 3C
0040: 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70 61 6E 74 3E 3C 2F 64 64 73
0060: 3E 00 00 00
[1617806685.928824] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x15B5CC50, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1617806685.932029] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x15B5CC50, participant_id: 0x000(1)
[1617806685.934531] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x15B5CC50, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1617806685.938191] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x15B5CC50, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1617806685.942817] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x15B5CC50, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1617806685.958918] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x15B5CC50, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1617806685.986947] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x15B5CC50, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1617806685.986990] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x15B5CC50, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1617806685.987010] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x15B5CC50, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1617806685.987027] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x15B5CC50, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1617806685.987042] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x15B5CC50, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1617806685.987057] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x15B5CC50, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1617806685.987074] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x15B5CC50, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1617806685.989304] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x15B5CC50, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1617806685.989451] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x15B5CC50, len: 128, data: 
0000: 81 80 01 00 01 05 77 00 00 0B 00 02 02 02 00 00 69 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 72 74 2F 6D 69 63 72 6F 52 4F 53 2F 70 69 6E 67 3C 2F 6E 61 6D 65 3E 3C 64 61
0040: 74 61 54 79 70 65 3E 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 48 65 61 64
0060: 65 72 5F 3C 2F 64 61 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73 3E 00 00 01 00
[1617806685.989708] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x15B5CC50, topic_id: 0x000(2), participant_id: 0x000(1)
[1617806685.989844] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x15B5CC50, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1617806685.989879] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x15B5CC50, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1617806685.990665] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x15B5CC50, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
[... cut ...]
[1617806686.303168] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x15B5CC50, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 09 00 09 00 80
[1617806686.303241] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x15B5CC50, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 0A 00 00 00 80
[1617806686.304572] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x15B5CC50, len: 13, data: 
```

Eembedder, 2021-04-11:
Hi, I have a similar problem. When I run a command:
`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0`
I see:
```
[1618172033.889203] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1618172033.889412] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1618172038.034231] info     | Root.cpp           | create_client            | create                 | client_key: 0x041835D4, session_id: 0x81
[1618172038.034376] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x68695508, address: 0
```
And only after power off/on on stm32 board I see:
```
[1618173038.120495] info     | Root.cpp           | create_client            | create                 | client_key: 0x041835D4, session_id: 0x81
[1618173038.120640] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x68695508, address: 0
```
So, I can't connect only after power cycle or reset stm32 board

pablogs9, 2021-04-12:
Please use `-v6` flag to paste here the agent output

Eembedder, 2021-04-12:
It is a part of the output. It is connects only after power off/on.
`~/microros_ws$ ros2 run micro_ros_agent micro_ros_agent serial -v 6 --dev /dev/ttyUSB0 -b 115200
[1618206424.783881] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1618206424.784094] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1618206442.029537] info     | Root.cpp           | create_client            | create                 | client_key: 0x041835D4, session_id: 0x81
[1618206442.029708] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x68695508, address: 0
[1618206442.029924] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1618206442.044153] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 100, data: 
0000: 81 80 00 00 01 05 5A 00 00 0A 00 01 01 02 00 00 4C 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 63 75 62 65 6D 78 5F 6E 6F 64 65 3C 2F 6E
0040: 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70 61 6E 74 3E 3C 2F 64 64 73 3E 00
0060: 00 00 00 00
[1618206442.047689] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1618206442.051579] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1618206442.055624] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1618206442.058536] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x041835D4, participant_id: 0x000(1)
[1618206442.059630] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1618206442.059708] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1618206442.059755] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1618206442.059798] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1618206442.059841] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1618206442.064598] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1618206442.064849] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1618206442.067540] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1618206442.080538] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 128, data: 
0000: 81 80 01 00 01 05 77 00 00 0B 00 02 02 02 00 00 69 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 72 74 2F 75 6C 74 72 61 73 6F 6E 69 63 2F 6F 6E 65 3C 2F 6E 61 6D 65 3E 3C 64
0040: 61 74 61 54 79 70 65 3E 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 49 6E 74
0060: 33 32 5F 3C 2F 64 61 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73 3E 00 00 01 00
[1618206442.081214] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x041835D4, topic_id: 0x000(2), participant_id: 0x000(1)
[1618206442.081377] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1618206442.081441] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1618206442.082512] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 01 00 80
[1618206442.082782] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1618206442.092524] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1618206442.095440] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 24, data: 
[... cut ...]
[1618206442.636333] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 16 00 17 00 80
[1618206442.636592] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 18 00 00 00 80
[1618206442.642248] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x041835D4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 18 00 00 00 80
`

pablogs9, 2021-04-12:
Well, if you see the code, the micro-ROS participant tries to start on the boot of the board. If it does not have an agent it will fail after a timeout, so it is normal that you neet a powerup cycle or reset.

Eembedder, 2021-04-12:
Thanks. Is it possible to establish connection after the agent start? I have stm32 powered up for a long time and I need to establish a connection immediately after the agent start.

pablogs9, 2021-04-12:
You can code a logic that checks if the agent is available (using the RMW ping functionality) and connect and reconnect on demand. This is related: https://github.com/micro-ROS/micro_ros_arduino/issues/40#issuecomment-753930601

closed 2022-01-28 by pablogs9