micro-ROS/micro_ros_setup #300: Can't initialize node to subscribe String topic

Needrom, 2021-04-08:
## Issue template

- Hardware description: https://github.com/Needrom/Microros-on-DIscovery-board
- RTOS: freertos
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
I add string subscriber to ```ping_pong``` app, and when initialize it, I get some error from the printf.

#### Expected behavior
I should finish initialization and subscribe the topic.

#### Actual behavior
```
[ERROR] [ld.ld] [rclc]: [rclc_subscription_init_best_effort] Error in rcl_subscription_init: error not set
```

#### Additional information
Code in app.c

```
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/header.h>
#include <std_msgs/msg/string.h>

#include <stdio.h>
#include <unistd.h>
#include <time.h>

#ifdef ESP_PLATFORM
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#endif

#define STRING_BUFFER_LEN 50

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc); vTaskDelete(NULL);}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

rcl_publisher_t ping_publisher;
rcl_publisher_t pong_publisher;
rcl_subscription_t ping_subscriber;
rcl_subscription_t pong_subscriber;
rcl_subscription_t string_subscriber;

std_msgs__msg__Header incoming_ping;
std_msgs__msg__Header outcoming_ping;
std_msgs__msg__Header incoming_pong;
std_msgs__msg__String test_data_String;

int device_id;
int seq_no;
int pong_count;

void ping_timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
	RCLC_UNUSED(last_call_time);

	if (timer != NULL) {
//		printf("%s : 1 \r\n", __func__);
		seq_no = rand();
		sprintf(outcoming_ping.frame_id.data, "%d_%d", seq_no, device_id);
		outcoming_ping.frame_id.size = strlen(outcoming_ping.frame_id.data);

		// Fill the message timestamp
		struct timespec ts;
		clock_gettime(CLOCK_REALTIME, &ts);
		outcoming_ping.stamp.sec = ts.tv_sec;
		outcoming_ping.stamp.nanosec = ts.tv_nsec;
		
//		printf("%s : 2 \r\n", __func__);
		// Reset the pong count and publish the ping message
		pong_count = 0;
		rcl_publish(&ping_publisher, (const void*)&outcoming_ping, NULL);
//		printf("Ping send seq %s\n", outcoming_ping.frame_id.data);
	}
}

void ping_subscription_callback(const void * msgin)
{
	const std_msgs__msg__Header * msg = (const std_msgs__msg__Header *)msgin;

	// Dont pong my own pings
	if(strcmp(outcoming_ping.frame_id.data, msg->frame_id.data) != 0){
		printf("Ping received with seq %s. Answering.\n", msg->frame_id.data);
		rcl_publish(&pong_publisher, (const void*)msg, NULL);
	}
}

void pong_subscription_callback(const void * msgin)
{
	const std_msgs__msg__Header * msg = (const std_msgs__msg__Header *)msgin;

	if(strcmp(outcoming_ping.frame_id.data, msg->frame_id.data) == 0) {
		pong_count++;
		printf("Pong for seq %s (%d)\n", msg->frame_id.data, pong_count);
	}
}

void string_subscription_callback(const void * msgin){
	const std_msgs__msg__String * msg = (const std_msgs__msg__Header *)msgin;
	if(msg == NULL){
		printf("Callback : msg NULL \r\n");
	} else {
		printf("Callback : I Heard : %s \r\n", msg->data.data);
	}
}

void appMain(void *argument)
{
	int test_printf_count = 0;
	rcl_allocator_t allocator = rcl_get_default_allocator();
	rclc_support_t support;

// create init_options
	RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

	rcl_node_t node;
	RCCHECK(rclc_node_init_default(&node, "pingpong_node", "", &support));

	RCCHECK(rclc_publisher_init_default(&ping_publisher, &node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Header), "/microROS/ping"));

// Create a best effort pong publisher
	RCCHECK(rclc_publisher_init_best_effort(&pong_publisher, &node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Header), "/microROS/pong"));

	RCCHECK(rclc_subscription_init_best_effort(&ping_subscriber, &node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Header), "/microROS/ping"));

// Create a best effort  pong subscriber
	RCCHECK(rclc_subscription_init_best_effort(&pong_subscriber, &node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Header), "/microROS/pong"));

	
// Create a best effort String subscriber
	RCCHECK(rclc_subscription_init_best_effort(&string_subscriber, &node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, String), "/microROS/String"));

// Create a 3 seconds ping timer timer,
	rcl_timer_t timer;
	RCCHECK(rclc_timer_init_default(&timer, &support, RCL_MS_TO_NS(1000), ping_timer_callback));

// Create executor
	rclc_executor_t executor;
	RCCHECK(rclc_executor_init(&executor, &support.context, 3, &allocator));
	RCCHECK(rclc_executor_add_timer(&executor, &timer));
	RCCHECK(rclc_executor_add_subscription(&executor, &ping_subscriber, &incoming_ping,
		&ping_subscription_callback, ON_NEW_DATA));
	RCCHECK(rclc_executor_add_subscription(&executor, &pong_subscriber, &incoming_pong,
		&pong_subscription_callback, ON_NEW_DATA));

	RCCHECK(rclc_executor_add_subscription(&executor, &string_subscriber, &test_data_String,
		&string_subscription_callback, ON_NEW_DATA));

// Create and allocate the pingpong messages
	char outcoming_ping_buffer[STRING_BUFFER_LEN];
	outcoming_ping.frame_id.data = outcoming_ping_buffer;
	outcoming_ping.frame_id.capacity = STRING_BUFFER_LEN;

	char incoming_ping_buffer[STRING_BUFFER_LEN];
	incoming_ping.frame_id.data = incoming_ping_buffer;
	incoming_ping.frame_id.capacity = STRING_BUFFER_LEN;

	char incoming_pong_buffer[STRING_BUFFER_LEN];
	incoming_pong.frame_id.data = incoming_pong_buffer;
	incoming_pong.frame_id.capacity = STRING_BUFFER_LEN;

	device_id = rand();
	
	rcl_ret_t ret = 0;
	while(1){
		ret = rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10));
//		printf("ret: %d \r\n", ret);
		usleep(10000);
	}

	// Free resources
	RCCHECK(rcl_publisher_fini(&ping_publisher, &node));
	RCCHECK(rcl_publisher_fini(&pong_publisher, &node));
	RCCHECK(rcl_subscription_fini(&ping_subscriber, &node));
	RCCHECK(rcl_subscription_fini(&pong_subscriber, &node));
	RCCHECK(rcl_node_fini(&node));
}

```

pablogs9, 2021-04-08:
Check this: https://github.com/Needrom/Microros-on-DIscovery-board/blob/425769fce73af3faf1d79fab23c588e2c5e52463/freertos_apps/apps/ping_pong/app-colcon.meta#L7

closed 2021-04-08 by pablogs9

Needrom, 2021-04-08:
OK nice

Needrom, 2021-04-09:
still get error when I set this in the code 
```
RCCHECK(rclc_executor_add_subscription(&executor, &string_subscriber, &test_data_String,
                &string_subscription_callback, ON_NEW_DATA));

```

error is here
```

[INFO] [ld.ld] []: Created a timer with period 1000 ms.

Failed status on line 139: 1. Aborting.

```

and the setting of app-colcon.meta is here
```
{
    "names": {
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=2",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=3",
                "-DRMW_UXRCE_MAX_SERVICES=0",
                "-DRMW_UXRCE_MAX_CLIENTS=0",
                "-DRMW_UXRCE_MAX_HISTORY=4",
            ]
        }
    }
}        
```
if parameter set a little more bigger, the bin file will overflow like this.
[image]

pablogs9, 2021-04-09:
Reduce the `DRMW_UXRCE_MAX_HISTORY` to 1 to save RAM.

Needrom, 2021-04-09:
reduce the param ```DRMW_UXRCE_MAX_HISTORY``` could work.

and a error still come with code 
```
RCCHECK(rclc_executor_add_subscription(&executor, &string_subscriber, &test_data_String,
                &string_subscription_callback, ON_NEW_DATA));
```

like this 
```

[INFO] [ld.ld] []: Created a timer with period 1000 ms.

Failed status on line 139: 1. Aborting.
```

pablogs9, 2021-04-09:
You are creating the executor with 3 slots for execution entities:

```c
RCCHECK(rclc_executor_init(&executor, &support.context, 3, &allocator));
```

and then you are adding 4 entities, the last one will raise an error. Please check the API documentation [here](https://github.com/ros2/rclc/tree/master/rclc/include/rclc).

Needrom, 2021-04-11:
yes, it's. Get it

Needrom, 2021-04-12:
new error when subscribe the string topic

```
printf is ready 
Init uxr serial 
endif----- 
freeRTOS allocator success 
before appMain 
**********************************
Task  State   Prio    Stack    Num
**********************************
defaultTask    	X	24	736	1
LinkThr        	R	16	220	6
IDLE           	R	0	105	2
tcpip_thread   	B	24	163	4
microROS_app   	B	25	4906	7
Tmr Svc        	B	2	214	3
EthIf          	B	48	35	5
**********************************
[INFO] [ld.ld] []: Created a timer with period 1000 ms.

[ERROR] [ld.ld] [rclc]: [rclc_take_new_data] Error in rcl_take: error not set

[ERROR] [ld.ld] [rclc]: Error number: 1
Callback : I Heard :  
[ERROR] [ld.ld] [rclc]: [rclc_take_new_data] Error in rcl_take: error not set

[ERROR] [ld.ld] [rclc]: Error number: 1
Callback : I Heard :  
[ERROR] [ld.ld] [rclc]: [rclc_take_new_data] Error in rcl_take: error not set

[ERROR] [ld.ld] [rclc]: Error number: 1
Callback : I Heard :  
[ERROR] [ld.ld] [rclc]: [rclc_take_new_data] Error in rcl_take: error not set

[ERROR] [ld.ld] [rclc]: Error number: 1
Callback : I Heard :  
[ERROR] [ld.ld] [rclc]: [rclc_take_new_data] Error in rcl_take: error not set

[ERROR] [ld.ld] [rclc]: Error number: 1
Callback : I Heard :  
[ERROR] [ld.ld] [rclc]: [rclc_take_new_data] Error in rcl_take: error not set

[ERROR] [ld.ld] [rclc]: Error number: 1
Callback : I Heard :  
```

app.c code here
```
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/header.h>
#include <std_msgs/msg/string.h>

#include <stdio.h>
#include <unistd.h>
#include <time.h>

#ifdef ESP_PLATFORM
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#endif

#define STRING_BUFFER_LEN 50

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc); vTaskDelete(NULL);}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}
[... 130 lines cut ...]
	incoming_pong.frame_id.data = incoming_pong_buffer;
	incoming_pong.frame_id.capacity = STRING_BUFFER_LEN;

	//Todo allocate string messages
//	rosidl_runtime_c__String incoming_string_buffer[STRING_BUFFER_LEN];
//	incoming_string.data = incoming_string_buffer;
//	incoming_string.capacity = STRING_BUFFER_LEN;

	device_id = rand();
	
	rcl_ret_t ret = 0;
	while(1){
		ret = rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10));
//		printf("ret: %d \r\n", ret);
		usleep(10000);
	}

	// Free resources
	RCCHECK(rcl_publisher_fini(&ping_publisher, &node));
	RCCHECK(rcl_publisher_fini(&pong_publisher, &node));
	RCCHECK(rcl_subscription_fini(&ping_subscriber, &node));
	RCCHECK(rcl_subscription_fini(&pong_subscriber, &node));
	RCCHECK(rcl_node_fini(&node));
}
```

pablogs9, 2021-04-12:
Why have you modified `#define STRING_BUFFER_LEN 100` to 50? maybe it does not have space for deserializing the string.

Does the example work as it is, with no modification like: https://github.com/micro-ROS/freertos_apps/blob/foxy/apps/ping_pong/app.c?

Needrom, 2021-04-12:
actually I did not initialize the buffer with ```STRING_BUFFER_LEN``` , because I not using a string publisher, just string subscriber. why would this happen.

pablogs9, 2021-04-12:
Does the provided example works?

Needrom, 2021-04-13:
the example work just with the ping subscribe, Not with the pong subscriber.

the message I send
```
xxxx@ubuntu:~/microros_ws$ ros2 topic pub --once /microROS/ping std_msgs/msg/Header '{frame_id: "fake_ping"}'
publisher: beginning loop
publishing #1: std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=0, nanosec=0), frame_id='fake_ping')

xxxx@ubuntu:~/microros_ws$ ros2 topic pub --once /microROS/pong std_msgs/msg/Header '{frame_id: "fake_pong"}'
publisher: beginning loop
publishing #1: std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=0, nanosec=0), frame_id='fake_pong')
```

the message I receive 
```
Ping received with seq fake_ping. Answering.
```

and When I publish string topic.
```
xxxx@ubuntu:~/microros_ws$ ros2 run cpp_pubsub talker
[INFO] [1618291568.651687499] [minimal_publisher]: Publishing: 'Hello, world! 0'
[INFO] [1618291569.151835735] [minimal_publisher]: Publishing: 'Hello, world! 1'
[INFO] [1618291569.652073584] [minimal_publisher]: Publishing: 'Hello, world! 2'
[INFO] [1618291570.151909929] [minimal_publisher]: Publishing: 'Hello, world! 3'
[INFO] [1618291570.652173507] [minimal_publisher]: Publishing: 'Hello, world! 4'
[INFO] [1618291571.151437613] [minimal_publisher]: Publishing: 'Hello, world! 5'
[INFO] [1618291571.651853781] [minimal_publisher]: Publishing: 'Hello, world! 6'
^C[INFO] [1618291572.012573930] [rclcpp]: signal_handler(signal_value=2)
xxxx@ubuntu:~/microros_ws$ 
```

the serial become this 
```
Ping received with seq fake_ping. Answering.
[ERROR] [ld.ld] [rclc]: [rclc_take_new_data] Error in rcl_take: error not set

[ERROR] [ld.ld] [rclc]: Error number: 1
Callback : I Heard :  
[ERROR] [ld.ld] [rclc]: [rclc_take_new_data] Error in rcl_take: error not set

[ERROR] [ld.ld] [rclc]: Error number: 1
Callback : I Heard :  
[ERROR] [ld.ld] [rclc]: [rclc_take_new_data] Error in rcl_take: error not set

[ERROR] [ld.ld] [rclc]: Error number: 1
Callback : I Heard :  
[ERROR] [ld.ld] [rclc]: [rclc_take_new_data] Error in rcl_take: error not set

[ERROR] [ld.ld] [rclc]: Error number: 1
Callback : I Heard :  
[ERROR] [ld.ld] [rclc]: [rclc_take_new_data] Error in rcl_take: error not set

[ERROR] [ld.ld] [rclc]: Error number: 1
Callback : I Heard :  
[ERROR] [ld.ld] [rclc]: [rclc_take_new_data] Error in rcl_take: error not set

[ERROR] [ld.ld] [rclc]: Error number: 1
Callback : I Heard :  
[ERROR] [ld.ld] [rclc]: [rclc_take_new_data] Error in rcl_take: error not set

[ERROR] [ld.ld] [rclc]: Error number: 1
Callback : I Heard :  

```

pablogs9, 2021-04-13:
Regarding the first part, how do you know that the node is not receiving the pongs? See that a message is only printed if the pong is for the node: https://github.com/micro-ROS/freertos_apps/blob/b7c36f41bd516e26af9d5daf9b21681ec2cec1c1/apps/ping_pong/app.c#L74

Regading the last part, in your callback the casting is weird:

```c
void string_subscription_callback(const void * msgin){
	const std_msgs__msg__String * msg = (const std_msgs__msg__Header *)msgin;  // WRONG CASTING HERE
	if(msg == NULL){
		printf("Callback : msg NULL \r\n");
	} else {
		printf("Callback : I Heard : %s \r\n", msg->data.data); // SEE BELOW
	}
}
```

The print is accessing to a inexisteng member of `rosidl_runtime_c__String`, `data.data` does not exist, check the header definition:

```c
typedef struct rosidl_runtime_c__String
{
  /// The pointer to the first character, the sequence ends with a null byte.
  char * data;
  /// The length of the string (excluding the null byte).
  size_t size;
  /// The capacity represents the number of allocated bytes (including the null byte).
  size_t capacity;
} rosidl_runtime_c__String;
```

Needrom, 2021-04-13:
But I copy the callback from here, it seem like a tutorial. [https://micro.ros.org/docs/tutorials/core/programming_rcl_rclc/](url)

[image]

pablogs9, 2021-04-13:
yes, but you are casting from `std_msgs__msg__Header *` to `std_msgs__msg__String *`

Needrom, 2021-04-13:
change to this 
```
void string_subscription_callback(const void * msgin)
{
  const std_msgs__msg__String * msg = (const std_msgs__msg__String *)msgin;
  if (msg == NULL) {
    printf("String Callback: msg NULL\n");
  } else {
    printf("String Callback: I heard: %s\n", msg->data.data);
  }
}
```

the error still comes out.
```
[ERROR] [ld.ld] [rclc]: [rclc_take_new_data] Error in rcl_take: error not set

[ERROR] [ld.ld] [rclc]: Error number: 1
String Callback: I heard: 
```

and I was confused that 
the code below, why you call the call_back function before ```rcl_take```, what's the meaning of the rcl_take. 
```
660           switch (handle->type) {                                                                           │
   │661             case SUBSCRIPTION:                                                                              │
   │662               if (wait_set->subscriptions[handle->index]) {                                                 │
   │663                 rmw_message_info_t messageInfo;                                                             │
   │664                 rc = rcl_take(                                                                              │
   │665                   handle->subscription, handle->data, &messageInfo,                                         │
   │666                   NULL);                                                                                    │
  >│667                 if (rc != RCL_RET_OK) {                                                                     │
   │668                   // rcl_take might return this error even with successfull rcl_wait                        │
   │669                   if (rc != RCL_RET_SUBSCRIPTION_TAKE_FAILED) {                                             │
   │670                     PRINT_RCLC_ERROR(rclc_take_new_data, rcl_take);                                         │
   │671                     RCUTILS_LOG_ERROR_NAMED(ROS_PACKAGE_NAME, "Error number: %d", rc);                      │
   │672                   }                                                                                         │
   │673                   return rc;                                                                                │
   │674                 }                                                                                           │
   │675               }                                                                                             │
   │676               break;                     
```

pablogs9, 2021-04-13:
Can you try [this example](https://github.com/micro-ROS/micro-ROS-demos/blob/foxy/rclc/string_subscriber/main.c) in your platform?

Needrom, 2021-04-13:
run it for no change and 
```
[ERROR] [ld.ld] [rclc]: [rclc_init] Error in rcl_init: error not set

Failed status on line 33: 11. Aborting.
```

pablogs9, 2021-04-13:
Let me replicate this

pablogs9, 2021-04-13:
Have you changed the `RCCHECK(rclc_support_init(&support, argc, argv, &allocator));` to `RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));`... I guess that in your platform you don't have argc/argv.

Needrom, 2021-04-13:
yes my platform don't have argc/argv, and string can subscribe now with the example

pablogs9, 2021-04-13:
Maybe `std_msgs__msg__String__init` is not initializing well the string message. Usually, in micro-ROS, the best approach is to assign memory to messages that have sequences of any type.

Needrom, 2021-04-13:
OK