micro-ROS/micro_ros_setup #301: build_agent.sh missing in dashing branch?

Rob-le-Houe, 2021-04-08:
- Hardware description: Olimex-STM32-E407
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: ros2 dashing, Ubuntu 18.04

When trying to follow the steps outlined here: https://github.com/micro-ROS/micro_ros_setup 
the build_agent.sh script is not present, I have added it from the foxy branch and added it to the CMakeLists and then run colcon build. When I try to run 'ros2 run micro_ros_setup build_agent.sh' it succeeds but when I try to launch the agent I get:   

`/home/rob/uros_ws/install/micro_ros_agent/lib/micro_ros_agent/micro_ros_agent: error while loading shared libraries: libfastrtps.so.1.8.3: cannot open shared object file: No such file or directory`

I had a native install of Micro-XRCE-DDS-Agent from using ros2arduino and when I try to run that it also gives:

`./MicroXRCEAgent: error while loading shared libraries: libfastrtps.so.1.8.3: cannot open shared object file: No such file or directory`

which hasn't been an issue until now. I am new to this kind of microcontroller and the aim is to use the olimex board and micro-ros to replace an arduino running ros2arduino. This is my first attempt at uploading code to the board using an STLink V2 and using a UART to USB to communicate. Using this kind of board is certainly not as straightforward as an arduino! I appear to be building and the flashing the firmware correctly but I feel like I am missing something with this agent...

I have also tried to get going with STMcubeIDE but have a whole other set of issues to solve before I can use that!
Any help would be greatly appreciated.
Rob

pablogs9, 2021-04-08:
micro-ROS is not fully supported in Dashing. Do you have any issues with upgrading to Foxy?

Rob-le-Houe, 2021-04-08:
Thanks for the speedy reply, I believe I would need to upgrade to Ubuntu 20.04 in order to use Foxy?

I have tried to use the Foxy docker container to build the firmware as a work around and it all seems okay but when I try to communicate I have to use the version on my local machine which is dashing.

pablogs9, 2021-04-08:
Yes, I think that a micro-ROS Foxy will communicate with ROS 2 Dashing.

Rob-le-Houe, 2021-04-08:
Thanks, I'll close this and keep going with the docker container. I'm currently able to see both /int32_publisher and /microROS/ping topics from the container (foxy) and outside of it (dashing) if I build the respective examples, however there are no messages being published.

closed 2021-04-08 by Rob-le-Houe