micro-ROS/micro_ros_setup #303: error when I add Joint_state topic to appMain

Needrom, 2021-04-19:
## Issue template

- Hardware description: https://github.com/Needrom/Microros-on-DIscovery-board
- RTOS: freertos
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
```
ros2 run micro_ros_setup configure_firmware.sh joint_states_subscriber -t serial -d 1
ros2 run micro_ros_setup build_firmware.sh 
```

and flash the bin with ```openocd``` and 
```
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0
```
and run moveit
```
ros2 launch run_move_group run_move_group.launch.py
```

#### Expected behavior
reveive joint_state topic in callback and print message ```I get joint_state topic```

#### Actual behavior
sequence in ros_message
[ERROR] [ld.ld] [rclc]: [rclc_take_new_data] Error in rcl_take: error not set

[ERROR] [ld.ld] [rclc]: Error number: 1
Failed status on line 79: 1. Aborting.

#### Additional information
and when I initialize the message like this 
```
        joint_states_msg.name.data = (char *) malloc(ARRAY_LEN * sizeof(char));
        joint_states_msg.name.size = 0;
        joint_states_msg.name.capacity = ARRAY_LEN;

```
 problem still usual.

```
sequence in ros_message
[ERROR] [ld.ld] [rclc]: [rclc_take_new_data] Error in rcl_take: error not set

[ERROR] [ld.ld] [rclc]: Error number: 1
```

should I initialize every member of the struct in here? I just initialize name of it.
```
typedef struct sensor_msgs__msg__JointState
{
  std_msgs__msg__Header header;
  rosidl_runtime_c__String__Sequence name;
  rosidl_runtime_c__double__Sequence position;
  rosidl_runtime_c__double__Sequence velocity;
  rosidl_runtime_c__double__Sequence effort;
} sensor_msgs__msg__JointState;

```

pablogs9, 2021-04-19:
You should init recursively every `__Sequence` member in your message.

Needrom, 2021-04-19:
> You should init recursively every `__Sequence` member in your message.

is there any api I could use to initialize a message.

pablogs9, 2021-04-19:
Not yet

Needrom, 2021-04-19:
How can't I init recursively? I just initialize it like this.
```
        joint_states_msg.name.data = (char *) malloc(ARRAY_LEN * sizeof(char));
        joint_states_msg.name.size = 0;
        joint_states_msg.name.capacity = ARRAY_LEN;

        joint_states_msg.position.data = (char *) malloc(ARRAY_LEN * sizeof(char));
        joint_states_msg.position.size= 0;
        joint_states_msg.position.capacity = ARRAY_LEN;

        joint_states_msg.velocity.size = (char *) malloc(ARRAY_LEN * sizeof(char));

        joint_states_msg.velocity.size = 0;
        joint_states_msg.velocity.size = ARRAY_LEN;

        joint_states_msg.effort.size = (char *) malloc(ARRAY_LEN * sizeof(char));

        joint_states_msg.effort.size = 0;
        joint_states_msg.effort.size = ARRAY_LEN;

```

and output still wrong
```
[ld.ld] [rclc]: [rclc_init] Error in rcl_init: error not set

Failed status on line 43: 1. Aborting.
```

but line 43 is here:
```
RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

```

pablogs9, 2021-04-19:
Do you have your client connected to an agent? It is failing the client-to-agent init session.

Needrom, 2021-04-19:
yeah, That was me reset the power and some how it is fail to connect to agent.

running with to subscriber on it ,if I don't publish the ```joint_states``` topic ,the string subscriber is all right.
but when I open a node to publish the ```joint_state``` , the system seem down and the string subscriber won't work any more.

pablogs9, 2021-04-19:
1. You are setting the same callback for both subscribers and they are not the same type:
```c
	RCCHECK(rclc_executor_add_subscription(&executor, &subscriber, &msg, &subscription_callback, ON_NEW_DATA));
	RCCHECK(rclc_executor_add_subscription(&executor, &joint_states_subscriber, &joint_states_msg, &subscription_callback, ON_NEW_DATA));
```

2. You are allocating `ARRAY_LEN*sizeof(char)` for the position, velocity, and effort when they are [floats64](http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/JointState.html). Maybe you should use `malloc(ARRAY_LEN * sizeof(double));`

Needrom, 2021-04-19:
change to this and already initialize double to position, velocity, and effort

```
RCCHECK(rclc_executor_add_subscription(&executor, &subscriber, &msg, &subscription_callback, ON_NEW_DATA));
        RCCHECK(rclc_executor_add_subscription(&executor, &joint_states_subscriber, &joint_states_msg, &subscription_joint_state_callback, ON_NEW_DATA));
```

```
        char joint_states_msg_buffer[ARRAY_LEN];
        joint_states_msg.header.frame_id.data = joint_states_msg_buffer;
        joint_states_msg.header.frame_id.capacity = ARRAY_LEN;

        joint_states_msg.name.data = (double *) malloc(ARRAY_LEN * sizeof(double));
        joint_states_msg.name.size = 0;
        joint_states_msg.name.capacity = ARRAY_LEN;

        joint_states_msg.position.data = (double *) malloc(ARRAY_LEN * sizeof(double));
        joint_states_msg.position.size= 0;
        joint_states_msg.position.capacity = ARRAY_LEN;

        joint_states_msg.velocity.size = (double *) malloc(ARRAY_LEN * sizeof(double));

        joint_states_msg.velocity.size = 0;
        joint_states_msg.velocity.size = ARRAY_LEN;

        joint_states_msg.effort.size = (char *) malloc(ARRAY_LEN * sizeof(char));

        joint_states_msg.effort.size = 0;
        joint_states_msg.effort.size = ARRAY_LEN;

```

pablogs9, 2021-04-19:
Are you setting the number of subscribers correctly? [here](https://github.com/Needrom/Microros-on-DIscovery-board/blob/a3a01faa5995d620d2d8629de6fd1a8df73659e0/freertos_apps/apps/int32_publisher/app-colcon.meta#L7)

Needrom, 2021-04-19:
yes here [https://github.com/Needrom/Microros-on-DIscovery-board/blob/main/freertos_apps/apps/joint_states_subscriber/app-colcon.meta](url) 

I set to 4

pablogs9, 2021-04-19:
I'm going to replicate your code.

Meanwhile, you are using FreeRTOS, are you sure that `malloc` is working properly? Usually, in FreeRTOS we use `pvPortMalloc`

pablogs9, 2021-04-19:
BTW:

```c
        joint_states_msg.name.data = (double *) malloc(ARRAY_LEN * sizeof(double));
```

name is an string so use `char`

Needrom, 2021-04-19:
delete malloc and change to this.
```
char joint_states_name_buffer[ARRAY_LEN];
joint_states_msg.name.data = joint_states_name_buffer;
```
still the same

Needrom, 2021-04-19:
> BTW:
> 
> ```c
>         joint_states_msg.name.data = (double *) malloc(ARRAY_LEN * sizeof(double));
> ```
> 
> name is an string so use `char`

my bad

pablogs9, 2021-04-19:
And check that part:

```
        joint_states_msg.velocity.size = (double *) malloc(ARRAY_LEN * sizeof(double));

        joint_states_msg.velocity.size = 0;
        joint_states_msg.velocity.size = ARRAY_LEN;

        joint_states_msg.effort.size = (char *) malloc(ARRAY_LEN * sizeof(char));

        joint_states_msg.effort.size = 0;
        joint_states_msg.effort.size = ARRAY_LEN;
```

you are always setting `size`...

pablogs9, 2021-04-19:
Maybe use something like:

```c
	joint_states_msg.name.data = (char *) malloc(ARRAY_LEN * sizeof(char));
	joint_states_msg.name.size = 0;
	joint_states_msg.name.capacity = ARRAY_LEN;

	joint_states_msg.position.data = (double *) malloc(ARRAY_LEN * sizeof(double));
	joint_states_msg.position.size= 0;
	joint_states_msg.position.capacity = ARRAY_LEN;
	
	joint_states_msg.velocity.data = (double *) malloc(ARRAY_LEN * sizeof(double));
	joint_states_msg.velocity.size = 0;
	joint_states_msg.velocity.capacity = ARRAY_LEN;
	
	joint_states_msg.effort.data = (double *) malloc(ARRAY_LEN * sizeof(double));
	joint_states_msg.effort.size = 0;
	joint_states_msg.effort.capacity = ARRAY_LEN;
```

pablogs9, 2021-04-19:
Also, take into account that you are allocating `ARRAY_LEN * sizeof(double)` = 200*8 = 1600 B four times, so 6400 B...

Are you sure that this is being allocated correctly?

Needrom, 2021-04-19:
what's the mean of 
```
  /// The number of valid items in data
  size_t size;

```

change to this 
```
char joint_states_name_buffer[ARRAY_LEN];
        joint_states_msg.name.data = joint_states_name_buffer;
        joint_states_msg.name.size = 4;
        joint_states_msg.name.capacity = ARRAY_LEN;

        double joint_states_position_buffer[JOINT_DOUBLE_LEN];
        joint_states_msg.position.data = joint_states_position_buffer;
        joint_states_msg.position.size= 4;
        joint_states_msg.position.capacity = JOINT_DOUBLE_LEN;

        double joint_states_velocity_buffer[JOINT_DOUBLE_LEN];
        joint_states_msg.velocity.size = joint_states_velocity_buffer;

        joint_states_msg.velocity.size = 4;
        joint_states_msg.velocity.size = JOINT_DOUBLE_LEN;

        double joint_states_effort_buffer[JOINT_DOUBLE_LEN];
        joint_states_msg.effort.size = joint_states_effort_buffer;

        joint_states_msg.effort.size = 4;
        joint_states_msg.effort.size = JOINT_DOUBLE_LEN;

```

and when I debug this elf.
when the ```joint_states``` topic comes out
the program will go into here.
```
   │92            */                                                                           │
   │93          void HardFault_Handler(void)                                                   │
   │94          {                                                                              │
   │95            /* USER CODE BEGIN HardFault_IRQn 0 */                                       │
   │96                                                                                         │
   │97            /* USER CODE END HardFault_IRQn 0 */                                         │
  >│98            while (1)                                                                    │
   │99            {                                                                            │
   │100             /* USER CODE BEGIN W1_HardFault_IRQn 0 */                                  │
   │101             /* USER CODE END W1_HardFault_IRQn 0 */                                    │
   │102           }                                                                            │
   │103         }                                                                              │
   │104                                                                                        │
   │105         /**          
```

pablogs9, 2021-04-19:
You are having a segfault or something similar...

In a type sequence, `capacity` is the available memory, `size` is the actual elements in the sequence and `data` should have memory allocated for storing up to `capacity` elements.

pablogs9, 2021-04-19:
Also:

```c
       double joint_states_velocity_buffer[JOINT_DOUBLE_LEN];
        joint_states_msg.velocity.size = joint_states_velocity_buffer;   <--- setting size to a buffer!!

        joint_states_msg.velocity.size = 4; <--- setting size to 4!!
        joint_states_msg.velocity.size = JOINT_DOUBLE_LEN; <--- setting size to a there number!!
```

Instead of:

```c
       double joint_states_velocity_buffer[JOINT_DOUBLE_LEN];
        joint_states_msg.velocity.data = joint_states_velocity_buffer;   <--- setting DATA to a buffer!!

        joint_states_msg.velocity.size = 0; <--- setting SIZE to 0!!
        joint_states_msg.velocity.capacity = JOINT_DOUBLE_LEN; <--- setting CAPACITY to a there number!!
```

Needrom, 2021-04-20:
> Also:
>
> ```c
>        double joint_states_velocity_buffer[JOINT_DOUBLE_LEN];
>         joint_states_msg.velocity.size = joint_states_velocity_buffer;   <--- setting size to a buffer!! [...]

yes, fix that now. 
size is 2 now, and capacity is 20 now.

but problem is still go on. I have comment all code in joint_states callback.
is there any  possible that the data copy is out of the array limit? Find no explain for now.
```
void subscription_joint_state_callback(const void *msgin){
//	const sensor_msgs__msg__JointState *msg = (const sensor_msgs__msg__JointState *)msgin;
//	printf("I get joint_state topic \r\n");
}
```
problem still go on.

pablogs9, 2021-04-20:
How long are the joint_states that are being published?

Needrom, 2021-04-20:
I check the ```joint_states``` topic rate
```
average rate: 96.131
	min: 0.009s max: 0.014s std dev: 0.00063s window: 98

```

pablogs9, 2021-04-20:
But how long are they? Which is the length of the `position` for example?

Needrom, 2021-04-20:
thinking for a long time, that was 7 
[image]

pablogs9, 2021-04-20:
so `JOINT_DOUBLE_LEN` should be greater than 7

Needrom, 2021-04-20:
> `JOINT_DOUBLE_LEN`

already test it with 20 and size is 7 now.
```
#define JOINT_DOUBLE_LEN 20
```

```
char joint_states_msg_buffer[ARRAY_LEN];
        joint_states_msg.header.frame_id.data = joint_states_msg_buffer;
        joint_states_msg.header.frame_id.size = 20;
        joint_states_msg.header.frame_id.capacity = ARRAY_LEN;

        char joint_states_name_buffer[ARRAY_LEN];
        joint_states_msg.name.data = joint_states_name_buffer;
        joint_states_msg.name.size = 7;
        joint_states_msg.name.capacity = ARRAY_LEN;

        double joint_states_position_buffer[JOINT_DOUBLE_LEN];
        joint_states_msg.position.data = joint_states_position_buffer;
        joint_states_msg.position.size= 7;
        joint_states_msg.position.capacity = JOINT_DOUBLE_LEN;

        double joint_states_velocity_buffer[JOINT_DOUBLE_LEN];
        joint_states_msg.velocity.data = joint_states_velocity_buffer;
        joint_states_msg.velocity.size = 7;
        joint_states_msg.velocity.capacity = JOINT_DOUBLE_LEN;

        double joint_states_effort_buffer[JOINT_DOUBLE_LEN];
        joint_states_msg.effort.data = joint_states_effort_buffer;
        joint_states_msg.effort.size = 7;
        joint_states_msg.effort.capacity = JOINT_DOUBLE_LEN;

```

pablogs9, 2021-04-20:
Other point, `joint_states_msg.name` is a **sequence** of strings:

```c
typedef struct sensor_msgs__msg__JointState
{
  std_msgs__msg__Header header;
  rosidl_runtime_c__String__Sequence name;
  rosidl_runtime_c__double__Sequence position;
  rosidl_runtime_c__double__Sequence velocity;
  rosidl_runtime_c__double__Sequence effort;
} sensor_msgs__msg__JointState;
```

So you need to do something like:

```c
rosidl_runtime_c__String string_buffer[JOINT_DOUBLE_LEN];
joint_states_msg.name.data = string_buffer;
joint_states_msg.name.size = 0;
joint_states_msg.name.capacity = JOINT_DOUBLE_LEN;

for(int i = 0; i < JOINT_DOUBLE_LEN; i++){
        joint_states_msg.name.data[i].data = (char*) malloc(ARRAY_LEN);
        joint_states_msg.name.data[i].size = 0;
        joint_states_msg.name.data[i].capacity = ARRAY_LEN;
}
```

Needrom, 2021-04-21:
yes,work for me now. amazing

closed 2021-04-21 by Needrom