micro-ROS/micro_ros_setup #305: microROS agent- client serial communication is not working in zephyr RTOS

Vishnu-Vijay95, 2021-04-19:
RTOS: Zephyr
MCU : olimex STM32E407

Tried the microROS zephyr RTOS int32_publisher program on olimex STM32E407 board. Tried agent client communication through usb-serial and its working fine. Tired the same with serial communication but the agent client communication is not establishing.
[image]

pablogs9, 2021-04-19:
Have you configured the serial port that you want to use [here](https://github.com/micro-ROS/zephyr_apps/blob/89e85c8cf29647d276203735a49a41ecbc4b8dd1/microros_extensions/microros_transports.h#L47)?

Vishnu-Vijay95, 2021-04-20:
Ok thanks @pablogs9

closed 2021-04-20 by Vishnu-Vijay95