micro-ROS/micro_ros_setup #308: Echo topic from micro-ros gives UnicodeDecodeError

anastasiapan, 2021-04-21:
## Issue template

- Hardware description: Nucleo F767
- RTOS: FreeRTOS
- Installation type: generic static lib
- Version or commit hash: foxy

#### Steps to reproduce the issue
- Micro ros agent:
`docker run -it --rm --net=host microros/micro-ros-agent:foxy udp4 --port 8888 -v6`
- Output of the agent: (some of it)
 ```
[1618998703.011296] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x50976793, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 76 00 00 00 80
[1618998703.110864] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x50976793, len: 508, data: 
0000: 81 80 76 00 0D 01 F4 01 07 01 7C 03 00 47 00 05 00 00 00 00 00 00 00 00 60 00 00 00 B9 77 F7 DE
0020: CD FB F3 68 87 36 68 6F 36 70 74 0F 77 FD 3B DA FE D8 77 57 60 41 22 26 3A 09 01 B4 EE 57 FC 79
0040: F3 CC 34 2B A1 F0 2E 90 28 52 E0 C3 FF FF 72 FD 3F EC A5 D9 C8 A2 03 90 46 0A 90 05 BC CD E5 5A
0060: DD 27 BF F7 86 43 90 CA C0 E1 14 B6 B3 F7 DC EC BA DE 2E D9 DF F0 E8 5D 2C 10 26 00 60 00 00 00
0080: B9 77 F7 DE CD FB F3 68 87 36 68 6F 36 70 74 0F 77 FD 3B DA FE D8 77 57 60 41 22 26 3A 09 01 B4
00A0: EE 57 FC 79 F3 CC 34 2B A1 F0 2E 90 28 52 E0 C3 FF FF 72 FD 3F EC A5 D9 C8 A2 03 90 46 0A 90 05
00C0: BC CD E5 5A DD 27 BF F7 86 43 90 CA C0 E1 14 B6 B3 F7 DC EC BA DE 2E D9 DF F0 E8 5D 2C 10 26 00
00E0: 00 00 00 00 00 80 4C 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0100: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0120: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0140: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0160: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0180: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
01A0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
01C0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
01E0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[1618998703.111022] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x50976793, len: 404, data: 
0000: 81 80 77 00 0D 03 8C 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00A0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00C0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00E0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0100: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0120: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0140: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0160: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0180: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[1618998703.111214] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x50976793, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 77 00 00 00 80
[1618998703.111256] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | cli
 ```

The ROS node is ok what I did is to just create a publisher (https://micro.ros.org/docs/tutorials/core/programming_rcl_rclc/) as instructed in the tutorials. I can even see the message values changing when I run with the debugger and if I type : `ros2 topic list` I can see the published topic. When I run `ros2 topic echo /my/odom/topic` I get this error:
 ``` 
Traceback (most recent call last):
  File "/opt/ros/foxy/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.9.8', 'console_scripts', 'ros2')()
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2topic/command/topic.py", line 41, in main
    return extension.main(args=args)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2topic/verb/echo.py", line 81, in main
    return main(args)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2topic/verb/echo.py", line 102, in main
    subscriber(
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2topic/verb/echo.py", line 117, in subscriber
    rclpy.spin(node)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/__init__.py", line 191, in spin
    executor.spin_once()
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 711, in spin_once
    raise handler.exception()
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/task.py", line 239, in __call__
    self._handler.send(None)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 419, in handler
    arg = take_from_wait_list(entity)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 344, in _take_subscription
    msg_info = _rclpy.rclpy_take(capsule, sub.msg_type, sub.raw)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb9 in position 0: invalid start byte
 ```
I have ros2 foxy on my machine and I have tried both with cyclone and fastrtps. I have also tried to echo it from a foxy container (https://hub.docker.com/_/Ros)

#### Expected behavior
I want to be able to echo the topic

#### Actual behavior
UnicodeError when I do 
`ros2 topic echo` or even `ros2 topic hz`. I can see the result with `ros2 topic info`:
 ```
Type: nav_msgs/msg/Odometry

Publisher count: 1

Node name: capra
Node namespace: /
Topic type: nav_msgs/msg/Odometry
Endpoint type: PUBLISHER
GID: 01.0f.3a.5f.1b.00.00.00.07.00.00.00.00.00.01.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_BEST_EFFORT
  Durability: RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL
  Lifespan: 9223372036854775807 nanoseconds
  Deadline: 9223372036854775807 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 9223372036854775807 nanoseconds

Subscription count: 0
 ```

pablogs9, 2021-04-21:
Can you share the code that you are using to create this publisher?

It can be a problem instantiating the msg structures in your micro-ROS client code.

anastasiapan, 2021-04-21:
Ok I have a class for that: 
```  
class UrosPublisher
{
public:
  UrosPublisher() : msg_{} {}

  void method1()
  {
    rcl_allocator_t allocator = rcl_get_default_allocator();
    rcl_ret_t rc_support = rclc_support_init(&support_, 0, NULL, &allocator);

    // Create node
    rcl_ret_t rc_node = rclc_node_init_default(&node_, "node", "namespace", &support_);

    // Create publisher
    rcl_ret_t rc_publisher = rclc_publisher_init_default(
      &publisher_, &node_, ROSIDL_GET_MSG_TYPE_SUPPORT(nav_msgs, msg, Odometry), "odometry");
  }

  void method2()
  {
    static float x = 0.0f;
    msg_.pose.pose.position.x = x++;
    msg_.pose.pose.position.y = x++;
    msg_.pose.pose.position.z = x++;
    rcl_ret_t rc_publish = rcl_publish(&publisher_, &msg_, NULL);
  }

private:
  nav_msgs__msg__Odometry msg_;
  rcl_publisher_t publisher_;
  rclc_support_t support_;
  rcl_node_t node_;
};
```
I am calling method1 once to initialize and then method2 to publish.

pablogs9, 2021-04-21:
You need to initialize `nav_msgs__msg__Odometry msg_`. By default in micro-ROS **sequence** messages does not have memory and are not initialized.  In your message you have a couple of `rosidl_runtime_c__String` (sequences) in `frame_id` and other inside `header`.

```c
typedef struct nav_msgs__msg__Odometry
{
  std_msgs__msg__Header header;
  rosidl_runtime_c__String child_frame_id;
  geometry_msgs__msg__PoseWithCovariance pose;
  geometry_msgs__msg__TwistWithCovariance twist;
} nav_msgs__msg__Odometry;
```

Related: https://github.com/micro-ROS/micro_ros_setup/issues/303#issuecomment-823182651

anastasiapan, 2021-04-22:
Thank you that was it!

closed 2021-04-22 by anastasiapan