micro-ROS/micro_ros_setup #310: micro ros service rclc - values not passed

anastasiapan, 2021-04-29:
## Issue template

- Hardware description: Nucleo F767
- RTOS: FreeRTOS
- Installation type: generic static lib
- Version or commit hash: foxy

#### Expected behavior
Ok so my issue is that I have created a service which just reads a 4 byte integer (int32). Whenever I call it I can see the service is called (from the debugger) but the values I get are garbage. And I do not get a response even with these values. I have followed [the demo service guide](https://github.com/micro-ROS/micro-ROS-demos/blob/foxy/rclc/addtwoints_server/main.c). I will post my source code, srv and msg format

#### Actual behavior
Call the service and pass the correct value

#### Additional information
##### Msg: (my_msg.msg)
`int32 my_attribute`

##### Srv (my_srv.srv):
```
my_pkg/my_msg write_value
---
my_pkg/my_msg  read_value
```

##### my_node.cpp
```
extern UrosService uros_srv;

class UrosService
{
public:
  UrosService() :req_{}, res_{} {}

  void initialise()
  {
    rcl_allocator_t allocator = rcl_get_default_allocator();
    rcl_ret_t rc_support = rclc_support_init(&support_, 0, NULL, &allocator);

    // Create node
    rcl_ret_t rc_node = rclc_node_init_default(&node_, "node", "namespace", &support_);

    // Create service
    rclc_service_init_default(
        &cservice_, &node_,
        ROSIDL_GET_SRV_TYPE_SUPPORT(cmy_pkgg, srv, my_srv), "/service_name");
      rclc_executor_init(
        &executor_, &support_.context, 1, rcl_get_default_allocator());
      rclc_executor_add_service(
        &executor_, &service_, &req_, &res_, [](const void * req, void * res) {
          uros_srv.req_ =
            *static_cast<my_pkg__srv__my_srv_Request *>(req);
          uros_srv.service_callback();
        });

  }

void UrosOdometryPublisher::service_callback()
{
  // This values are garbage and they do not get passed correctly
  int32_t my_variable = req_.write_value.my_attribute;
  res_.read_value.my_attribute = my_variable;
}

void main_loop()
{
  // This is run by the board's main loop
  rclc_executor_spin_some(&executor_, RCL_MS_TO_NS(25));
}

private:
  void service_callback();
  rclc_executor_t executor_;
  rcl_service_t service_;
  my_pkg__srv__my_srv_Request req_;
  my_pkg__srv__my_srv_Response res_;
};
```

##### Call service
```
ros2 service call /service_name my_pkg/srv/my_srv "{write_value: {my_attribute: 1000}}"
```

pablogs9, 2021-04-29:
Is this `ROSIDL_GET_SRV_TYPE_SUPPORT(cmy_pkgg, srv, my_srv)` correct? 
The extra `c` and `g`:  **c**my_pkg**g**?

pablogs9, 2021-04-29:
Also you have `cservice_` inside `    rclc_service_init_default` instead of `service_`

pablogs9, 2021-04-29:
Also, `rcl_allocator_t allocator` exists inside `initialise` function stack, once it returns `allocator` memory is not guaranteed to exists, maybe when you call `rclc_executor_spin_some`, this memory is not correct.

Make `rcl_allocator_t allocator` a private member of the class.

anastasiapan, 2021-04-29:
I am actually using a frertos allocator and it works fine. The other things are just copy-paste issues I am sorry but I can't show my source code so I created a dummy class. This is why these mistakes happened.

pablogs9, 2021-04-29:
Yes, but `executor_` is saving a pointer to `allocator` and it is being used in `rclc_executor_spin_some`. It can fail at any moment.

Regarding your problem, how does the following code behaves?

```cpp
      rclc_executor_add_service(
        &executor_, &service_, &req_, &res_, [](const void * req, void * res) {
               my_pkg__srv__my_srv_Request * req_in = (my_pkg__srv__my_srv_Request *) req;
               my_pkg__srv__my_srv_Request * res_in = (my_pkg__srv__my_srv_Request *) res;
               res_in->read_value.my_attribute = req_in->read_value.my_attribute 
        });
```

JanStaschulat, 2021-04-29:
In the executor_add_service you assign the request message `req` to `uros_srv.req_` in the callback `UrosOdometryPublisher::service_callback()` you are using a different variable `req_.`

anastasiapan, 2021-04-29:
Ok I have tried this and now I tried with the AddTwoInts example interface to make sure:

```
example_interfaces__srv__AddTwoInts_Request req;
example_interfaces__srv__AddTwoInts_Response res;

rcl_ret_t rc_support =
    rclc_support_init(&support_, 0, NULL, &board::ros::FreeRtosAllocator::get_allocator());

  // Create node
  rcl_ret_t rc_node = rclc_node_init_default(&node_, REMOTE_NODE_NAME, NAMESPACE_NAME, &support_);

  rclc_service_init_default(
    &calibration_service_, &node_, ROSIDL_GET_SRV_TYPE_SUPPORT(example_interfaces, srv, AddTwoInts),
    "/add");
  rclc_executor_init(
    &executor_, &support_.context, 1, &board::ros::FreeRtosAllocator::get_allocator());

  rclc_executor_add_service(
    &executor_, &calibration_service_, &req, &res, [](const void * req, void * res) {
    example_interfaces__srv__AddTwoInts_Request * req_in =
        (example_interfaces__srv__AddTwoInts_Request *)req;
      example_interfaces__srv__AddTwoInts_Response * res_in =
        (example_interfaces__srv__AddTwoInts_Response *)res;

      res_in->sum = req_in->a + req_in->b;
     
    });
```

This is how I get the allocator:
```
class FreeRtosAllocator
{
public:
  FreeRtosAllocator(const FreeRtosAllocator &) = delete;
  FreeRtosAllocator & operator=(const FreeRtosAllocator &) = delete;
  FreeRtosAllocator(FreeRtosAllocator &&) = delete;
  FreeRtosAllocator & operator=(FreeRtosAllocator &&) = delete;

  static rcl_allocator_t & get_allocator() { return instance().freertos_allocator_; }

private:
  FreeRtosAllocator() : freertos_allocator_{rcutils_get_zero_initialized_allocator()}
  {
    freertos_allocator_ = rcutils_get_zero_initialized_allocator();
    freertos_allocator_.allocate = freertos_allocate;
    freertos_allocator_.deallocate = freertos_deallocate;
    freertos_allocator_.reallocate = freertos_reallocate;
    freertos_allocator_.zero_allocate = freertos_zero_allocate;
    rcutils_set_default_allocator(&freertos_allocator_);
  }
  static FreeRtosAllocator & instance()
  {
    static FreeRtosAllocator instance;
    return instance;
  }
  rcl_allocator_t freertos_allocator_;
};
```
I still have the same problem now a is random but b is always 1.

pablogs9, 2021-04-30:
Is not possible for you to test the services as they are provided in C99?
This way we can detect if we are having platform support problems, transport problems, typesupport problems, or just C++ issues between the C99 micro-ROS API and your code?

JanStaschulat, 2021-04-30:
Indeed, I have never tested the services in C++ application.

bjv-capra, 2021-04-30:
We could try it out I believe, running an isolated task. However, I can say we have this exact setup running successfully with a subscriber (which would behave similarly to a service I suppose). Having no problems at all to run on our system, except the need to run the subscription with best-effort.
What we did was to try to change the subscription to service following the same pattern. The callback is correctly called, with no problem, however, the data is incorrect.

Here the code that we use to initialize a subscriber (successfully)

```
  rclc_support_init(&support_, 0, NULL, &board::ros::FreeRtosAllocator::get_allocator());

  // Create node
  rclc_node_init_default(&node_, REMOTE_NODE_NAME, NAMESPACE_NAME, &support_);

  // Create subscription
  rclc_subscription_init_best_effort(
    &subscriber_, &node_, ROSIDL_GET_MSG_TYPE_SUPPORT(geometry_msgs, msg, Twist),
    REMOTE_TOPIC_IN_NAME);

  rclc_executor_init(
    &executor_, &support_.context, 1, &board::ros::FreeRtosAllocator::get_allocator());

  rclc_executor_add_subscription(
    &executor_, &subscriber_, &msg_,
    [](const void * msgin) {
      // we're still connected. This is just WIP check
      uros_remote_provider.timeout_counter_ = 0;
      // Don't do anything unless this is active
      Cvalidate_void(uros_remote_provider.get_state() == capra::lifecycle::State::active);
      uros_remote_provider.msg_ = *static_cast<const geometry_msgs__msg__Twist *>(msgin);
      uros_remote_provider.status_led_.toggle();
    },
    ON_NEW_DATA);
```

anastasiapan, 2021-05-05:
@pablogs9 I will move for now to topics instead of services. I have not found a solution you can close the issue if you want though.

closed 2021-05-05 by pablogs9

conroy-cheers, 2021-10-03:
I've just run into what seems to be exactly the same issue, but using `micro_ros_arduino`.
Service calls never return, and the values passed into the service callback seem to be nonsense. Can replicate with the below minimal example:
```cpp
#include <Arduino.h>
#include <micro_ros_arduino.h>
#include <rcl/error_handling.h>
#include <rclc/executor.h>
#include <rclc/rclc.h>
#include <std_msgs/msg/int64.h>
#include <std_srvs/srv/set_bool.h>
#include <stdio.h>

rcl_node_t node;
rclc_support_t support;
rcl_allocator_t allocator;
rclc_executor_t executor;

rcl_service_t service;
rcl_wait_set_t wait_set;

std_srvs__srv__SetBool_Response res;
std_srvs__srv__SetBool_Request req;

static char res_message = '\0';

#define RCCHECK(fn)                                                            \
    {                                                                          \
        rcl_ret_t temp_rc = fn;                                                \
        if ((temp_rc != RCL_RET_OK)) {                                         \
            while (1) {                                                        \
            };                                                                 \
        }                                                                      \
    }
#define RCSOFTCHECK(fn)                                                        \
    {                                                                          \
        rcl_ret_t temp_rc = fn;                                                \
        if ((temp_rc != RCL_RET_OK)) {                                         \
        }                                                                      \
    }

void service_callback(const void *lreq, void *lres) {
    std_srvs__srv__SetBool_Request *req_in =
        (std_srvs__srv__SetBool_Request *)lreq;
    std_srvs__srv__SetBool_Response *res_in =
        (std_srvs__srv__SetBool_Response *)lres;

    if (req_in->data == true) {
        digitalWrite(LED_RED, HIGH);
    } else if (req_in->data == false) {
        digitalWrite(LED_RED, LOW);
    }

    res_in->success = true;
}

void setup() {
    pinMode(LED_RED, OUTPUT);

    set_microros_transports();
    delay(1000);

    allocator = rcl_get_default_allocator();

    // create init_options
    RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

    // create node
    RCCHECK(rclc_node_init_default(&node, "set_bool_rclc", "", &support));

    // create service
    RCCHECK(rclc_service_init_best_effort(
        &service, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(std_srvs, srv, SetBool),
        "/setbool"));

    // create executor
    RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
    RCCHECK(rclc_executor_add_service(&executor, &service, &req, &res,
                                      &service_callback));
}

void loop() {
    RCSOFTCHECK(rclc_executor_spin(&executor));
}
```

pablogs9, 2021-10-04:
@conroy-cheers please open an issue in the proper repo and mention this one. Also, detail which board and transport are you using.

JanStaschulat, 2021-10-04:
where is the client running? if you use ROS Rolling on Linux (Agent), make sure you use FastRTPS as DDS middleware:
`export RMW_IMPLEMENTATION=rmw_fastrtps_cpp`
https://github.com/ros2/rmw_fastrtps