micro-ROS/micro_ros_setup #313: Tracetools build errors when enabling multithread support

bjv-capra, 2021-05-05:
## Issue template

- Hardware description: STM32F7
- RTOS: FreeRTOS
- Installation type: Static library
- Version or commit hash: foxy

#### Steps to reproduce the issue
colcon.meta (notice how multithreading is enabled)
```
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF",
            ]
        },
        "rcl": {
            "cmake-args": [
                "-DBUILD_TESTING=OFF",
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
            ]
        }, 
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON",
                "-DconfigUSE_POSIX_ERRNO=1",
                "-DCMAKE_VERBOSE_MAKEFILE:BOOL=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON"
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=ON",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
                "-DUCLIENT_PROFILE_MULTITHREAD=ON"
                "-DPLATFORM_NAME_FREERTOS=ON"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=4",
                "-DRMW_UXRCE_MAX_PUBLISHERS=10",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=2",
                "-DRMW_UXRCE_TRANSPORT=udp",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=<some valid ip>",
                "-DRMW_UXRCE_DEFAULT_UDP_PORT=8888",
            ]
        }
    }
}
```

Toolchain
```
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

set(PLATFORM_NAME "FreeRTOS_Plus_TCP")
set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)
set(CMAKE_CROSSCOMPILING 1)
set(CMAKE_SYSTEM_NAME Generic)
#set(CMAKE_SYSTEM_VERSION 1)
set(CMAKE_C_COMPILER_WORKS 1)
set(CMAKE_CXX_COMPILER_WORKS 1)
#set(CMAKE_SYSTEM_PROCESSOR arm)
set(CMAKE_AR arm-none-eabi-ar)
message("-- TOOLCHAIN AR ${CMAKE_AR}")
set(CMAKE_C_COMPILER arm-none-eabi-gcc)
message("-- TOOLCHAIN C COMPILER ${CMAKE_C_COMPILER}")
set(CMAKE_CXX_COMPILER arm-none-eabi-g++)
message("-- TOOLCHAIN CXX COMPILER ${CMAKE_CXX_COMPILER}")
set(CMAKE_ASM_COMPILER arm-none-eabi-gcc)
message("-- TOOLCHAIN ASM COMPILER ${CMAKE_ASM_COMPILER}")
set(CMAKE_OBJCOPY arm-none-eabi-objcopy)
message("-- TOOLCHAIN OBJCOPY ${CMAKE_OBJCOPY}")
set(CMAKE_OBJDUMP arm-none-eabi-objdump)
message("-- TOOLCHAIN OBJDUMP ${CMAKE_OBJDUMP}")
set(CMAKE_SIZE arm-none-eabi-size)
message("-- TOOLCHAIN SIZE ${CMAKE_SIZE}")

#set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)

set(C_COMMON_FLAGS "${C_COMMON_FLAGS} -mcpu=cortex-m7 -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -DCLOCK_MONOTONIC=0 -D'__attribute__(x)=' -DconfigUSE_POSIX_ERRNO=1 -DipconfigZERO_COPY_TX_DRIVER=1 -DUSE_HAL_DRIVER -DSTM32F767xx")

# generates a separate ELF section for each function in the source file
set(C_COMMON_FLAGS "${C_COMMON_FLAGS} -ffunction-sections")

# Enables or disables the generation of one ELF section for each variable in the source file
set(C_COMMON_FLAGS "${C_COMMON_FLAGS} -fdata-sections")

set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} ${C_COMMON_FLAGS}")

# Enables or disables the generation of code needed to support C++ exceptions.
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fno-exceptions")

# Disable generation of information about every class with virtual functions for use by the C++ runtime
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fno-rtti")

# Do not emit the extra code to use the routines specified in the C++ ABI for thread-safe initialization of local statics.
# Reduce bin size
# TODO Do we have code that could have thread safety problem?
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fno-threadsafe-statics")

# Register destructors for objects with static storage duration with the __cxa_ atexit function rather than the atexit function
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fno-use-cxa-atexit")

# Add common C language flags
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${C_COMMON_FLAGS}")

set(CMAKE_BUILD_TYPE DEBUG)

include_directories(SYSTEM
${CMAKE_CURRENT_LIST_DIR}/code
${CMAKE_CURRENT_LIST_DIR}/f7-drivers/STM32F7xx_HAL_Driver/Inc
${CMAKE_CURRENT_LIST_DIR}/f7-drivers/CMSIS/Device/ST/STM32F7xx/Include
${CMAKE_CURRENT_LIST_DIR}/f7-drivers/CMSIS/Include
${CMAKE_CURRENT_LIST_DIR}/f7-middlewares/Third_Party/FreeRTOS/Source/include
${CMAKE_CURRENT_LIST_DIR}/f7-middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2
${CMAKE_CURRENT_LIST_DIR}/f7-middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM7/r0p1
# ==================
# = FreeRTOS+POSIX =
# ==================
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/FreeRTOS-Plus-POSIX/include
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/FreeRTOS-Plus-POSIX/include/portable
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/FreeRTOS-Plus-POSIX/include/portable/st
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/include
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/include/private
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/include/FreeRTOS_POSIX
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-posix/include/FreeRTOS_POSIX/sys
# ================
# = FreeRTOS+TCP =
# ================
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-tcp/include/
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-tcp/portable/Compiler/GCC/
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-tcp/portable/NetworkInterface/include/
${CMAKE_CURRENT_LIST_DIR}/freertos-plus-tcp/portable/NetworkInterface/STM32F7xx/
)
set(__BIG_ENDIAN__ 0)
```

#### Expected behavior
Builds with no problem

#### Actual behavior
Error 
```
tracetools/include/tracetools/utils.hpp:65:34: error: cannot use 'typeid' with '-fno-rtti'
   65 |   return _demangle_symbol(typeid(l).name());
```

#### Additional information

This is the meta file used to build prior to the change
```
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF",
            ]
        },
        "rcl": {
            "cmake-args": [
                "-DBUILD_TESTING=OFF",
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
            ]
        }, 
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON",
                "-DconfigUSE_POSIX_ERRNO=1",
                "-DCMAKE_VERBOSE_MAKEFILE:BOOL=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON",
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=ON",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=4",
                "-DRMW_UXRCE_MAX_PUBLISHERS=10",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=2",
                "-DRMW_UXRCE_TRANSPORT=udp",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=10.46.28.3",
                "-DRMW_UXRCE_DEFAULT_UDP_PORT=8888",
            ]
        }
    }
}
```

pablogs9, 2021-05-05:
Does this work with the second `colcon.meta`?

pablogs9, 2021-05-05:
Check the commas in the JSON:

```json
                "-DUCLIENT_PROFILE_MULTITHREAD=ON"   <--- here
                "-DPLATFORM_NAME_FREERTOS=ON"
```

If the JSON is not well-formatted the colcon.meta is not being used.

bjv-capra, 2021-05-05:
Damn, how did I miss that little bastard. Thanks :). Now it builds.

bjv-capra, 2021-05-05:
However, had to do `set(PLATFORM_NAME_FREERTOS ON)` on the toolchain rather than the meta (ofcourse)

closed 2021-05-05 by bjv-capra